adafruit/circuitpython #9378: Windows: CIRCUITPY is read-only after Windows goes to sleep and wakes up

dhalbert, 2024-06-27:
CircuitPython 9.0.5.
Noted by JungleJim in discord on a QT Py ESP32-S3 4/2.

CIRCUITPY reappears, but as read-only after the Windows 10 computer it is attached to goes to sleep and is woken up.

samblenny, 2024-07-29:
This could possibly be related to #9454 (ESP32-S3 USB serial output stops working). Both involve odd behavior with ESP32-S3 sleep modes when USB mass storage is enabled.

tannewt, 2024-08-29:
I suspect this is the same as #9454. The ESP usb code wasn't detecting a resume correctly. It is fixed in https://github.com/hathach/tinyusb/pull/2784. We'll update CP as soon as that is merged in.

tannewt, 2024-09-09:
Anyone have a windows 10 computer they can test this with?

ladyada, 2024-09-11:
@BlitzCityDIY do you mind testing this out 🙏 only if you have time (my laptop is win11)

BlitzCityDIY, 2024-09-11:
i tested with a qt py esp32-s3 running absolute newest and this seems to be fixed.

closed 2024-09-12 by tannewt