adafruit/circuitpython #9379: CP 9.1.0-beta.3 fails to install on ESP32-S2 Reverse TFT Feather

garrett-strudler, 2024-06-27:
Adafruit ESP32-S2 Reverse TFT Feather, no peripherals except a battery.

CP 9.0.5 (latest stable) installs & runs fine, but 9.1.0-beta.3 doesn't install correctly and just ends up with an empty, read-only CIRCUITPY drive.

I can re-install 9.0.5 and everything comes back to normal.

Mounted on a Mac Studio running macOS 14.5

I've downloaded the uf2 a couple times just to verify, also tried copying through Terminal (opposed to drag and drop), but same result every time.

dhalbert, 2024-06-27:
I tried this on the same board, and 9.1.0-beta.3 is working for me, so there's something different about my board or setup from yours.

1. Could you try the absolute latest from here? https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/adafruit_feather_esp32s2_reverse_tft/ . 

2. What is in INFO_UF2.TXT when the BOOT drive is showing?

3. If you do
```py
import storage
storage.erase_filesystem()
```
when 9.1.0-beta.3 is running, does that help any?

4. Do you have any non-Mac computer to try the upload on, even, say an RPi?

garrett-strudler, 2024-06-27:
The filesystem erase was the ticket, thanks for the suggestion!

Just for info, my INFO_UF2.TXT file read:

TinyUF2 Bootloader 0.10.2 - tinyusb (0.12.0-203-ga4cfd1c69)
Model: Adafruit Feather ESP32-S2 Reverse TFT
Board-ID: ESP32S2-FeatherRevTFT-revB
Date: Jun 24 2022

dhalbert, 2024-06-27:
My UF2 bootloader version is the same. I'm not sure why 9.1.0-beta.3 was pickier about what seemed like a corrupted filesystem, but in any case, glad it's working!

closed 2024-06-27 by dhalbert