adafruit/circuitpython #9383: dump processor stack trace on crashes to help debug

aguaviva, 2024-06-28:
Please dump a stack trace info (both python and the processor stack) to help pin down where the error happened

This is what I got but couldn't tell why:
```
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

bill88t, 2024-06-28:
Debug builds do that.
Build circuitpython with `DEBUG=1` and you'll get a dump when crashing.
Then (at least for `espressif`), you can then get it decoded with `tools/decode_backtrace.py`.

tannewt, 2024-06-28:
Hard faults can often be due to memory corruption so we try to do as little as possible. We already have related bugs as well #8704 and #7490.

closed 2024-06-28 by tannewt

aguaviva, 2024-06-29:
Then I suggest having an error message that is a bit more actionable and explains what the next step it, in this case something along the lines: "please build circuitpython with DEBUG=1 to get an extended debug trace and include it in the bug report"

Thanks for your great work!