adafruit/circuitpython #9384: Pi Pico-Filesystem (CIRCUITPY) mounted Read-only for versions > v9 on Debian 11.10

BusterRaid, 2024-06-29:
### CircuitPython version

```python
Version en-GB-8.1.0 is mounted read-write.
Version en-GB-8.2.10 is mounted read-write.
Version en_GB-9.0.0-beta is mounted read-only.
Version en_GB-9.0.5 is mounted read-only.
Version en_GB-20240626-main-PR9375-396aaef (latest at test) is mounted read-only.
```

### Code/REPL

```python
N/A.
```

### Behavior

When using CircuitPython >= v9 on Raspberry Pi Pico, the file system (CIRCUITPY), is mounted Read-Only when the PICO is connected to Raspberry Pi Desktop Debian 11.10 (VMware Hypervisor guest).

Versions 8.x of CircuitPython seem to mount the file system (CIRCUITPY) as read-write.

(This is is possibly due to the age of my system or my inexperiance with CircuitPython/Pico.)

### Description

N/A.

### Additional information

### Environment:

- Raspberry Pi Pico (not W):

No modifications.

- VM hypervisor/host:

Dell 490 Precision.
Intel 5000X (Greencreek) + 631x/6321 ESB2
Dual Intel Xeon 5160
Microsoft Windows 10 Professional (x64) Build 19045.4529 (22H2

- VM Guest:

Raspberry Pi Desktop VM.
Debian 11.10 Bullseye
Linux XRCERIUM 5.10.0-15-amd64 #1 SMP Debian 5.10.120-1 (2022-06-09) x86_64 GNU/Linux
/bin/ls: ELF 32-bit LSB pie executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, BuildID[sha1]=00695414aa5413c8667e62c2362d119cb233a504, for GNU/Linux 3.2.0, stripped

- Hypervisor:

VMware Player 12.5.9 most recent version that will run on chipset.

### Testing (method):

- VM guest suspended.
- Pico plugged in to host ready for firmware loading.
- "flash_nuke.uf2" loaded.
- Target firmware/CircuitPython loaded.
- Ejected from host.
- Connected to host.
- Test that "copy.py" is R/W via host - Always OK.
- VM guest opened from suspend.
- Pico connected _to_ guest _from_ host.
  - Note: Prior to connection to the VM, the icon in the VM tool bar looks like a web cam.
  - Once connected to the VM "Connect (Disconnect from host)" it appears as a speaker icon.
- "Mu" development environment opened.
- Attempt to change "code.py" made, via "Mu"
- Some information gathered from guest.
[... 315 lines cut ...]
Jun 29 15:15:33 XRCERIUM kernel: sd 3:0:0:0: [sdb] Attached SCSI removable disk
Jun 29 15:15:33 XRCERIUM pulseaudio[1095]: Failed to find a working profile.
Jun 29 15:15:33 XRCERIUM pulseaudio[1095]: Failed to load module "module-alsa-card" (argument: "device_id="1" name="usb-Raspberry_Pi_Pico_E66038B713777233-04" card_name="alsa_card.usb-Raspberry_Pi_Pico_E66038B713777233-04" >
Jun 29 15:15:34 XRCERIUM udisksd[489]: Mounted /dev/sdb1 at /media/pi/CIRCUITPY on behalf of uid 1000
lines 1-37/37 (END)

- USB data from HWinfo64 v7.62 (host) when Pico connected:

[Device Information]
Device Manufacturer:	Raspberry Pi
Product Name:	Pico
Serial Number:	E66038B713777233
USB Version Supported:	2.00
USB Device Speed:	USB 1.1 Full-speed
Driver Description:	USB Composite Device
Hardware ID:	USB\VID_239A&PID_80F4
 		
[Driver Information]
Driver Manufacturer:	(Standard USB Host Controller)
Driver Description:	USB Composite Device
Driver Provider:	Microsoft
Driver Version:	10.0.19041.4474
Driver Date:	21-Jun-2006
DeviceInstanceId	USB\VID_239A&PID_80F4\E66038B713777233
Location Paths	PCIROOT(0)#PCI(1D03)#USBROOT(0)#USB(2)

dhalbert, 2024-06-30:
What happens if you try the board directly on the RPi without using a VM?

Also try erasing and reformatting CIRCUITPY. 

```py
import storage 
storage.erase_filesystem()
```

BusterRaid, 2024-07-01:
Hi. 

Thank you for the reply.

On the "original" set up (Dell 460-Win 10 VM host running "Raspberry Pi Desktop - Bullseye, 11.x" as a VM and the Pico connected to the VM):
```
import storage 
storage.erase_filesystem()
```
.. did not resolve the issue. 

After running the command the Pico seems to be "ejected" from the VM and then reconnected.
However the Pico is (re)mounted as ReadOnly.

When directly connecting the Pico to a Raspberry Pi 3 Model B Plus Rev 1.3 running Raspberry Pi OS Bookworm 12.x, the Pico seems to be OK-it is mounted as ReadWrite.

At the moment I dont have access to a physical Pi which is running "Raspberry Pi OS - Bullseye, 11.x". I'll see if I can spin one up tomorrow.

The current version of the x86 desktop (which I use for the VM) is "stuck" on Bullseye - there is no Bookworm version.

As I do most of my tinkering on a VM, I'd like to get CircuitPython working on Bullseye.

dhalbert, 2024-07-01:
We have not seen problems like this with Bullseye on a physical RPi, so I suspect it is something about using the VM, perhaps something about the VM settings.

Can you see the serial connection from the VM to the board, and connect to the REPL? If not, that may be confusing the issue.

BusterRaid, 2024-07-02:
> Can you see the serial connection from the VM to the board, and connect to the REPL? If not, that may be confusing the issue.

I think I can (it's all a bit new to me) - I'm using Mu on the Pi VM and the option "serial" is available and seems to work - it seems to connect and I can run stuff on the Pico (e.g.: turning the onboard LED on/off).
This is the case for both Circuit Python v8.x and v9.x versions that i have tried.

_Some other tests I carried out:_

**- For Raspberry Pi Desktop, Bullseye (VM), CircuitPython en_GB-9.0.5:**

The Pico is mounted ReadOnly.
There is no change to this behaviour if, within the hypervisor, I set "USB compatibility" as 1.1, 2 or 3 between reboots of the VM.

**- For Raspberry Pi Desktop, Buster(10.13) (VM), CircuitPython en_GB-9.0.5:**

Mostly a vanilla install, with updates.
The Pico is mounted ReadOnly.

**- For Debian Buster(11.10) i386 (VM), CircuitPython en_GB-9.0.5:**

Mostly a vanilla install, with updates.
When using the hypervisor option "Connect (Disconnect from host)", the Pico is connected to the VM but not automagically mounted.
Mounting via the "CIRCUITPY" icon on the desktop causes the Pico to be mounted ReadOnly.

Issuing the mount command:

	`mount -t vfat /dev/sdb1 /media/user1/CIRCUITPY -o rw`

... elicits the response:

	`mount: /media/user1/CIRCUITPY: WARNING: source write-protected, mounted read-only.`

... suggesting the device is being considered as a ReadOnly USB device.

In summary:
**CircuitPython en_GB-8.2.10** 	- OK on every VM I tried. 		- Mounted as ReadWrite.
**CircuitPython en_GB-9.0.5** 	- Not OK on every VM I tried. 	- Mounted as ReadOnly.

If using "CircuitPython en_GB-9.0.5", the Pico is mounted on the HOST machine (the Dell Windows 10 physical) as ReadWrite.
Once the Pico is connected/mounted to a VM its file system is "set" to ReadOnly. 
Shutting down the VM (returning the Pico to the host), the Pico file system is remains ReadOnly.
Removing the Pico from the host and reinserting it restores the filesystem to ReadWrite.

I agree the issue is almost certainly in the hypervisor handling of the USB device but the behaviour seems different between CircuitPython versions v8.x and v9.x (Caveat: I have tested a few of the available v8.x and v9.x versions).

Thanks.

asmagill, 2024-07-15:
I am seeing similar behavior on a Raspberry Pi 5 running the latest bookworm pi os release (no VM).

I've seen it with multiple Pi Picos and with the PyPortal Titano most recently -- if I run the latest 8.x version of Circuitpython, the flash will be mountable read-write; if I install Circuitpython 9.x, about 4 out of 5 times when I plug the device into my Pi, it shows up in the logs with:

~~~
[  +0.022841] sd 5:0:0:0: [sdi] Write Protect is on
[  +0.000008] sd 5:0:0:0: [sdi] Mode Sense: 03 00 80 00
[  +0.019416]  sdi: sdi1
[  +0.000587] sd 5:0:0:0: [sdi] Attached SCSI removable disk
~~~

and, as you might expect, whenever I try to mount it, I'll get "WARNING: source write-protected, mounted read-only."

The odd thing is that if I run `dmesg -w` in a window on my Pi, and then plug and unplug the circuitpython device in, about 1 in 5 times, it will show "Write Protect mode is off"

I have also tried erasing the filesystem from a safe mode boot and it makes no difference -- getting a write enabled mount is a crapshoot -- the only reliable "fix" is to revert to the latest  CP8.

dhalbert, 2024-07-15:
@asmagill Could you post the string of consecutive relevant log entries from dmesg or /var/log/syslog (latter might have more or less info) when plugging in the board when it works and when it does not work, so we can compare the two sets of entries?

Do you see this problem even if code.py is not present or very simple? Do you have a `boot.py`?

I am using Ubuntu 24.04 which comes from the same code base, and don't see this problem. @jepler uses Debian and I think has not seen this. But the problems above are all on Raspberry Pi OS and we are running on x64.

jepler, 2024-07-15:
that's right, I haven't seen anything like that on debian 12

asmagill, 2024-07-16:
I'm not at home at moment, so it'll be a day or so to gather the logs requested.

As to to `boot.py/code.py`, mostly I've tested without these files (or the defaults as created after `storage.erase_filesystem()` -- no `boot.py` and a simple print statement in `code.py`).

I did have a thought... the system I do most of my work on has externally powered usb hubs, one to a USB3 port, the other to a USB2 port and these are what I plug devices into 90+% of the time... when I gather the requested logs in a day or so, I'll do so with another recently purchased Pi5 and plug directly into the Pi skipping any hubs. If it makes a difference, it still leaves the question as to what changed between 8 and 9, but sensitivity to various USB hubs is nothing new (my experience, at any rate) in the embedded world...

asmagill, 2024-07-16:
Ok, yeah, some preliminary tests suggest that the hub is in fact the issue... I remove that and the write-protects almost entirely go away.

While that makes it a much less critical issue, it still leaves open the question of what changed the sensitivity between 8.x and 9.x and is it something that can be addressed or just needs to be documented and worked around?

...before I post logs, should I open this as a new issue?

dhalbert, 2024-07-16:
> ...before I post logs, should I open this as a new issue

It's fine to open a new issue, and then close this as being superceded. That way the new thread will be more on point.

If you have a different hub, you might try that so see if it's that particular hub. I had a cheap USB-C to USB-A hub that disconnected frequently, causing all kinds of problems. Also maybe try the "bad" hub on some non-RPi computer.

BusterRaid, 2024-07-22:
With regard to my issue:
The pico was connected directly to the Dell 490 - I tried both front and rear USB ports.
I have also tried the Pico on a Dell 6400 Inspiron laptop - similar set up to the above Dell 490 (Win 10 Pro 22H2 x64, Raspberry Pi Desktop VM running under VMware 12.5.x Player).
In all cases the Pico mounts (in the Raspberry Pi Debian 11 VM) as ReadOnly.

dhalbert, 2024-07-23:
@hathach This is odd: MSC is mounting read-only in CircuitPython 9 when using a VM or when using certain hubs. Do this ring any bells for you related to some TinyUSB update?

djairjr, 2024-08-16:
I'm working on a version of Circuitpython for the Xiao ESP32S3 Sense and I had the same problem. In my case, I have a machine with Dual Boot (Ubuntu 22.03 and Windows 11). I carried out all the customization of the source files in my fork on Github and cloned the Adafruit repository. Then, on Linux, I moved the custom files and compiled the firmware.
When I start the machine in Linux, I have the CIRCUITPY drive mounted and read and write access. But when I boot into Windows, I don't have access to the CIRCUITPY drive.
I can open the files to view in Thonny, but I can't save them.

dhalbert, 2024-08-16:
@djairjr Let's discuss in #9528.

hathach, 2024-09-05:
@dhalbert sorry for late response. This does not ring any bell to me, I am not quite sure, but I remember to see this randomly (not very often). The flow would be around CPY is_writable_cb() usage
- when is_writable = false tinyusb will reject WRITE_10 command
- host will send TEST_UNIT_READY/REQUEST_SENSE for previous failed reason, currently is SCSI_SENSE_DATA_PROTECT
- normally host will attempt to re-write again later

not sure what went wrong, maybe we could use a not sense such as NOT READY or something to prevent host to give up on this (depending on drivers, host may give up retry). I am not sure, this sense code has been doing fine so far.

garberw, 2024-09-17:
I have a raspberry pi zero as my "desktop" with the pico w attached to the data usb port of the raspi zero.  I have run "mkdir /media/CIRCUITPY" on the raspi zero.  **Can someone please give me the mount parameters in the format for /etc/fstab.**
I currently use a line in fstab:
`UUID=********        /media/CIRCUITPY vfat    defaults,nosuid,nodev,nofail,sync,uid=1000,gid=1000,umask=027  0       0
`
I have been having problems with it mounting read-only too.  I swapped the usb cable and removed a hub as suggested above and that may have had something to do with it working now (not sure how long it will last).

dhalbert, 2024-09-17:
@garberw The automount that Ubuntu is doing for shows up as this:
```
/dev/sdb1 on /media/halbert/CIRCUITPY type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
```
Which board are you working with? We want to fix the read-only issue.

garberw, 2024-09-17:
I have a raspberry pi zero wh (not zero 2 w) as "desktop"  and a pico wh as microcontroller.  it's not the pico2.
I tried the above mount option and it did not improve things even with a simple straight through cable with no hubs or anything inbetween; 
Previously I tried using fsck on a different pico board of the same model.  that showed a corrupted boot sector on CIRCUITPY;
I also tried flashing the "nuke" .uf2 to reset the board and reinstall circuitpython;
the problem sometimes happens when I run my program which writes to the UART; that crashed it this time; have no idea what that has to do with anything;
best guess is that by unplugging the power to either the pico or the raspi zero without proper shutdown or syncing the pico I corrupted the filesystem on the pico;  should definitely use "sync" mount options always for pico;

garberw, 2024-09-17:
or maybe you have to mount it as type "-t msdos"

garberw, 2024-09-18:
I used the mount options 
`UUID=4F17-3FFE        /media/CIRCUITPY vfat    rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2        0       0
`
and _after rebooting_ and reflashing circuitpy .uf2 it was mounted read write.  I have not been able to crash it yet.
It may have something to do with the (wrong) mount options used when copying .mpy precompiled binaries to /media/CIRCUITPY/lib since the above mount options specify charset etc and may somehow translate the copied files;

garberw, 2024-09-18:
big hint

The original problem was that I installed raspbian in desktop mode but I switched to booting to console mode.
Unfortunately when I ssh in to raspbian on the zero it still tried to autostart pcmanfm when I plugged in the pico ...
but since the zero was headless (logged in over ssh) pcmanfm just hanged; so I turned off the autostart for pcmanfm;

Does the autostart for pcmanfm come from udev ????

the rapsi zero keeps telling me that e.g. after a reboot when I do `umount /media/CIRCUITPY` followed by `mount -a` that I need to run `systemctl daemon-reload` even though it has been run several times;

Could this be caused by a problem from modifying fstab while the disk is mounted?

Could this be from some udev rule that has been conflicting with fstab all along?  I don't think there is a udev rule for picos in raspbian; when I boot headless (before I added the fstab entry) the pico CIRCUITPY drive was not automounted;

garberw, 2024-09-18:
This is what you get when the pico is automounted **_on fedora_**:
`/dev/sda1 on /media/CIRCUITPY type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)`

garberw, 2024-09-19:
it seems to be mounting rw (read-write) now consistently; do not know how long this will last; only changes were
(1) update; full-upgrade;
(2) apt --reinstall install mount
(3) pipx install mpremote        (I think this installs micropython also !!!!)
something fixed it; hopefully that will help you;

garberw, 2024-09-21:
@dhalbert 
broken again.  I did
`apt --reinstall install mount
pix install mpremote
`
then it worked fine as reported in the previous post.  Was tested quite a bit.  Stayed mounted but had other stability problems.

then installed blinka and it failed immediately.  There's something in blinka that fouled it up.

garberw, 2024-09-21:
The problem that the CIRCUITPY disk on the pico keeps mounting read-only ... I fixed it myself.  run `fsck -CVlr /dev/sda1`  as follows:
`root@weathervt ####-> fsck -CVlr /dev/sda1
fsck from util-linux 2.38.1
[/usr/sbin/fsck.vfat (1) -- /media/CIRCUITPY] fsck.vfat /dev/sda1 
Locking disk by /run/fsck/sda.lock ... succeeded.
fsck.fat 4.2 (2021-01-31)
There is no label in boot sector, but there is volume label 'CIRCUITPY' stored in root directory
1) Copy volume label from root directory to boot sector
2) Remove volume label from root directory
[12?q]? 1

*** Filesystem was changed ***
The changes have not yet been written, you can still choose to leave the
filesystem unmodified:
1) Write changes
2) Leave filesystem unchanged
[12?q]? 1
/dev/sda1: 62 files, 271/980 clusters
/dev/sda1: status 1, rss 1888, real 51.965142, user 0.000000, sys 0.027163
Unlocking /run/fsck/sda.lock.
root@weathervt ####->
`
Now it mounts read-write.

dhalbert, 2024-09-21:
If this happens again, could you run `fsck -VN` just to report with what `fsck` thinks is wrong? I am thinking that the drive is being dismounted when it is an inconsistent state or the dirty bit is set. The latter should not make it be read-only, but there may be some defaults on your system that are causing this. I am not sure that blinka has anything to do with this. The volume label manipulation you mentioned above may also have corrected some other issue as a side-effect.

I will do some research about similar non-CIRCUITPY reports on Rsapberry Pi OS.

garberw, 2024-09-21:
`(env) root@weathervt ####>> fsck -VN /dev/sda1
fsck from util-linux 2.38.1
[/usr/sbin/fsck.vfat (1) -- /media/CIRCUITPY] fsck.vfat /dev/sda1 
(env) root@weathervt ####>>
`
This time my "solution" did not work.  Also did `fsck -CVlr` and did not work.
If this keeps up I can not use the pico as it (or something) is too unstable.  Works better if it is plugged into a fedora machine.  Mounts read-write more often.

dhalbert, 2024-09-22:
Do you have a regular x64 desktop as well? If so do you see the same kind of behavior? And do you have a second RP2040 board of some kind? If so does it also act up in the same way? I'm just trying to narrow the problem down. How about if you run regular Raspberry Pi OS on the Pico W?

garberw, 2024-09-22:
I have two computers 
(1) x86 running fedora
(2) raspberry pi zero W (not raspberry pi zero 2W) running raspberry pi os most recent version
And one microcontroller
(3) raspberry pi pico  running whatever it came with plus circuit python .uf2
You can't run raspberry pi os on the pico w as it is not a computer just a microcontroller.

What I want is for the pico to plug into the zero over usb and mount the pico CIRCUITPY on the zero using fstab on the zero or anyhow reliable and controllable.  This is what is so flaky.  The pico flash memory often mounts on the zero under /media/CIRCUITPY as read-only.  The zero sometimes shuts down when I plug the pico into it indicating that the zero power supply is  possibly not good enough.  What do you think ????  I usually leave the pico plugged into the zero and shutdown then unplug and re-plug in the zero to reboot the whole thing so the pico should get mounted by fstab when the zero boots.

When I sometimes plug the pico into the x86 fedora desktop directly it usually automounts (this is done using some automatic method not fstab on the fedora desktop) and this is quite reliable.  It is usually (perhaps always?) successfully mounted read write not read only.

I have four picos to play with and four zeros.  They are not expensive but I don't want to waste them.

dhalbert, 2024-09-22:
Sorry, I mean Pi Zero W, of course. Too many boards with "Pi" and "W" in their names. I will test on a Pi W and maybe some other Pi's.

garberw, 2024-09-22:
see update it is a pico WH **_with wireless_** and a zero _**with wireless**_
The zero power supply is 1.5 Amps on usb at 5 V and the whole system uses only 350 mA max at 5 V which is distributed in a complicated way.

dhalbert, 2024-09-22:
It seems to me very unusual that  Pi Zero W shuts down sometimes when you plug in the Pico W.

If you disconnect everything from your unusual power supply but the Pi Zero W does it happen?
If you use a more conventional wall wart power supply (say 1A or 2A), does that happen?
If you swap out the Pi Zero W for another Pi Zero W, does it also happen?

These are all things I would try in an effort to isolate the problem.

garberw, 2024-09-23:
The answer is no to all of those questions.  I suspect those are in fact due to my complicated power supply (solar weather station)

But I have a clue about the mounting read-only problem.
At one point I thought the pico had crashed but it was just stuck in the repl after a crash.  When I learned how to use the program "screen" I could differentiate between these crashes and real lockups.  
There is another problem "real lockups" where what I think is going on is you are disconnected from /dev/ttyACM0 (no screen; no thonny; no mpremote; no terminal; no `tail -f /dev/ttyACM0`) and the program crashes; at the end of the program is a command to do a soft reset;  The CIRCUITPY mount also disconnects after some timeout (then reconnects later); 

One fix is to put a 60 second pause at the end of the program before the reset so the CIRCUITPY does not keep disconnecting and reconnecting; this allows you to copy a known working program to overwrite `code.py` so it will stop cycling;

The `fsck` fix plus the update/upgrade probably was actually effective for fixing the problem with it being mounted read-only.
There were so many unplug-without-dismount power cut offs that it may have actually corrupted the pico.  but mount is working more predictably now.   (embarassed ! :-) blush)

dhalbert, 2024-09-23:
Glad to hear you have diagnosed it. Here are some ideas about how to make things more robust.

- Wrap the entire program in a try-except that catches all exceptions. This could be something like:
```py
try:
    do_everything()
except Exception:
    # do a reset, etc.
 ```
 You can put the main program in `do_everything()`. That makes the code a little neater. However, if you have a lot of globals, this might force you to restructure the program, so the alternative is just to wrap the whole main loop or whatever in try-except, so it will all need to be indented.

- call `microcontroller.reset()` (hard reset) or `supervisor.reload()` (soft restart) as necessary.
- If you might be going into "brownout" safe mode, add a `safemode.py` to recover from that. See https://learn.adafruit.com/circuitpython-safe-mode. This was designed exactly for the kind of scenario where a solar power supply is not charging the battery well (say, due to clouds), and the board keeps rebooting because it gets just enough voltage to come up and then shuts down when more power is consumed.
- It is possible that the RP2040 is coming up but the flash chip is not getting enough voltage to work well. We have seen this kind of thing before. Most flash chips require 2.6-2.7V to operate properly.
- Resets or power failures when a write is happening to the flash will often result in corruption.

I will close this issue for now, but open a new issue or reopen this one if you think there are CircuitPython-caused problems.

closed 2024-09-23 by dhalbert