adafruit/circuitpython #9390: saving +run in the web intereface adds extra wrong characters in REPL console that prevent `import <file>` from running

aguaviva, 2024-07-01:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.5 on 2024-07-01; Espressif ESP32 TTGO T8 v1.7 with ESP32
```

### Code/REPL

```python
just pressing `save+run` causes `Aimport test` to show up, the wrong `A` causes the action to fail
```

### Behavior

```
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Code done running.

Adafruit CircuitPython 9.0.5 on 2024-07-01; Espressif ESP32 TTGO T8 v1.7 with ESP32
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== Aimport test

Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```

dhalbert, 2024-07-01:
Is this in the bare web workflow interface or is it using https://code.circuitpython.org ?

aguaviva, 2024-07-01:
I access directly the web interface using circuitpython.local or cpy-xxxxxx.local, y also get an `R` sometimes, not sure where that is coming from. Not sure this is specific to the ESP32

```
Adafruit CircuitPython 9.0.5 on 2024-07-01; Espressif ESP32 TTGO T8 v1.7 with ESP32
>>> 
raw REPL; CTRL-B to exit
>R
```

I get the same issue on a esp32-cam, all very consistently.

dhalbert, 2024-07-01:
@makermelissa This looks like web-editor?

aguaviva, 2024-07-01:
all runs consistently fine using Thonny (no extra characters showin up)

aguaviva, 2024-07-13:
Should I file this bug in the web-editor repo? Also, am I the only one seeing these issues?

RetiredWizard, 2024-07-13:
I get the second results you indicated, when I click on the button, the editor appears to put the REPL into raw REPL mode, display an "R" and then disconnect/reconnect the web serial terminal. When reconnected, the REPL seems to be in the middle of a RAW paste operation, pressing ctrl-d closes the paste and then a ctrl-b exits the RAW REPL mode.

dhalbert, 2026-06-01:
I don't see this problem anymore in https://code.circuitpython.org, on a Feather ESP32 V2. I'm closing this for now, as I believe if we were still seeing this we would be getting multiple reports.

I was unable to get a Feather HUZZAH32 (the original version) to work at all, but I think that may be board-specific.