adafruit/circuitpython #9394: The IIC address of the GT911 touch chip cannot be detected when using the IIC of the raspberrypi Pico

Carlos-TGW, 2024-07-02:
### CircuitPython version

```python
adafruit-circuitpython-raspberry_pi_pico-en_GB-9.0.5.uf2
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
#
# SPDX-License-Identifier: MIT

# pylint: disable=broad-except, eval-used, unused-import

"""CircuitPython I2C Device Address Scan"""
import time
import board
import busio

# List of potential I2C busses
ALL_I2C = ("board.I2C()", "board.STEMMA_I2C()", "busio.I2C(board.GP21, board.GP20)")

# Determine which busses are valid
found_i2c = []
for name in ALL_I2C:
    try:
        print("Checking {}...".format(name), end="")
        bus = eval(name)
        bus.unlock()
        found_i2c.append((name, bus))
        print("ADDED.")
    except Exception as e:
        print("SKIPPED:", e)

# Scan valid busses
if len(found_i2c):
    print("-" * 40)
    print("I2C SCAN")
    print("-" * 40)
    while True:
        for bus_info in found_i2c:
            name = bus_info[0]
            bus = bus_info[1]

            while not bus.try_lock():
                pass

            print(
                name,
                "addresses found:",
                [hex(device_address) for device_address in bus.scan()],
            )

            bus.unlock()

        time.sleep(2)
else:
    print("No valid I2C bus found.")
```

### Behavior

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

code.py output:
Checking board.I2C()...SKIPPED: 'module' object has no attribute 'I2C'
Checking board.STEMMA_I2C()...ADDED.
Checking busio.I2C(board.GP21, board.GP20)...SKIPPED: I2C peripheral in use
----------------------------------------
I2C SCAN
----------------------------------------
board.STEMMA_I2C() addresses found: []
board.STEMMA_I2C() addresses found: []
board.STEMMA_I2C() addresses found: []
board.STEMMA_I2C() addresses found: []
board.STEMMA_I2C() addresses found: []

### Description

The IIC address of the GT911 touch chip cannot be detected when using the IIC of the raspberrypi Pico。
I only mount the GT911 touch screen in this IIC bus, but the address of IIC has not been detected. I want to use the driver of gt911.py of CircuitPython, but now even the IIC address of GT911 has not been detected。

Carlos-TGW, 2024-07-02:
It is normal to use other IIC devices, such as DHT20

RetiredWizard, 2024-07-02:
@Carlos-TGW I've only tested using an ESP chip, but I'd be curious if you have better luck using the [gt911_touch.py](https://github.com/RetiredWizard/PyDOS_virtkeyboard/blob/main/lib/gt911_touch.py) driver and [GT911Demo.py](https://github.com/RetiredWizard/PyDOS_virtkeyboard/blob/main/GT911demo.py). I don't think the Pico has a board.I2C and definitely doesn't have a board.TOUCH_RESET so you'll have to update the demo script with the appropriate objects/pins.

tannewt, 2024-07-02:
Please post a picture of your setup.

This is better asked on the Adafruit Support forum if you purchased from Adafruit: https://forums.adafruit.com/viewforum.php?f=60 Otherwise, ask the manufacturer of your devices for help.

RetiredWizard, 2024-07-02:
@Carlos-TGW if you open a forum post, please ping me so I can follow the progress :grin: The gt911 board I have uses what I think is a bit of a hack to initialize the touch screen. It ties the "touch interrupt" line high through a 5K resistor. You might try making that mod on your display and see if the issue your seeing is caused by the board not properly initializing on power-up.

RetiredWizard, 2024-07-03:
I just took another look at your code.py and I assuming you have the gt911 connected to GP21/GP20. I believe The "I2C peripheral in use" messages you are getting suggests that the GP21/GP20 pins can not be used for I2C. It looks to me like those pins are using the I2C0 port which is also used by the STEMMA I2C port. If you are trying to connect the touch screen to GP21/GP20, you probably will have to move it to a pair of pins assigned to I2C1. 

It's also possible (I don't know) that bitbangio would work with GP21/GP20 so if rewiring would be difficult at this stage, you might try importing bitbangio and trying an I2C device using that: ``i2c = bitbangio.I2C(board.GP21, board.GP20))``

RetiredWizard, 2024-07-03:
I did a little testing and it appears that if you don't substantiate the STEMMA_I2C object you won't see the peripheral conflict when trying to use GP21/GP20. So, if you're not actually using the STEMMA_I2C bus, try updating this line as follows and see if you see the touch screen I2C address:
`` ALL_I2C = ("board.I2C()", "busio.I2C(board.GP21, board.GP20)")``

Carlos-TGW, 2024-07-04:
> I did a little testing and it appears that if you don't substantiate the STEMMA_I2C object you won't see the peripheral conflict when trying to use GP21/GP20. So, if you're not actually using the [...]

Yes, I tried to use the pair of GP3 and GP2 pins to detect, but the address of the touch was still not detected. Whether it was GP21, GP20, or GP3 and GP2, I had added the pull-up resistor, but it had no effect

Carlos-TGW, 2024-07-04:
> Please post a picture of your setup.
> 
> This is better asked on the Adafruit Support forum if you purchased from Adafruit: https://forums.adafruit.com/viewforum.php?f=60 Otherwise, ask the manufacturer of your devices for help.

[image]
This is the hardware connection circuit

RetiredWizard, 2024-07-04:
Well, the pull ups and I2C conflicts are the only CircuitPython leads I can think of. Your connection circuit is very different than the one used by the board I have:
[image]
So I think Scott's suggestions are probably your best bet.

> This is better asked on the Adafruit Support forum if you purchased from Adafruit: https://forums.adafruit.com/viewforum.php?f=60 Otherwise, ask the manufacturer of your devices for help.

Carlos-TGW, 2024-07-04:
> Well, the pull ups and I2C conflicts are the only CircuitPython leads I can think of. Your connection circuit is very different than the one used by the board I have: [...]

This is the external circuit of the hardware. In fact, the supplier told me that there is a pull-up resistor added inside the touch screen GT911. When I use the Arduino platform, the touch screen is normal, but it does not work when I switch to the CircuitPython platform

bablokb, 2024-07-04:
In your schematic, you have a TP_EN pad. Do you have to change the state of this pad first? This is not part of the CP driver.

Carlos-TGW, 2024-07-05:
> In your schematic, you have a TP_EN pad. Do you have to change the state of this pad first? This is not part of the CP driver.

This is the initialization pin of the screen needs to change the timing to set the IIC address of the screen, but whether the address is changed or not, an address should appear during the IIC scan

RetiredWizard, 2024-07-05:
Since your hardware works with Arduino, it does sound like a CircuitPython support issue, you will probably get more people looking at the issue if you post to the forums or the [#help-with-circuitpython](https://discord.com/channels/327254708534116352/537365702651150357) discord channel. That being said, based on the test code.py and results you've posted the reason you're not finding the I2C address on the I2C pins in your circuit diagram is explained by ``Checking busio.I2C(board.GP21, board.GP20)...SKIPPED: I2C peripheral in use``. Your code.py only searched the board.STEMMA_I2C() port for the touch screen which isn't where you connected it.

To properly search for the test screen, I believe you would need to run your test code.py again but change the assignment of **ALL_I2C** to: 
```ALL_I2C = ("busio.I2C(board.GP21, board.GP20)")```

or simplify your test code to eliminate the loop:
```
# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
#
# SPDX-License-Identifier: MIT

# pylint: disable=broad-except, eval-used, unused-import

"""CircuitPython I2C Device Address Scan"""
import time
import board
import busio

bus = busio.I2C(board.GP21, board.GP20)

# Scan I2C bus
print("-" * 40)
print("I2C SCAN")
print("-" * 40)
while True:
    while not bus.try_lock():
        pass

    print(
        "addresses found:",
        [hex(device_address) for device_address in bus.scan()],
    )

    bus.unlock()
    time.sleep(2)
```

Carlos-TGW, 2024-07-08:
> import time
> import board
> import busio
>
> bus = busio.I2C(board.GP21, board.GP20)
>
> # Scan I2C bus
> print("-" * 40)
> print("I2C SCAN")
> print("-" * 40)
> while True:
>     while not bus.try_lock(): [...]

I used your code, but it didn't work. I tried to connect the SDA and SCL replacement pins of GT911 to GP3 and GP2, but it still didn't work.

RetiredWizard, 2024-07-08:
I think you've exhausted my limited I2C knowledge, the only other thing I might mention is that I'm always getting the SCL/SDA lines backwards, have you tried swapping the GP21/GP20 connections?

bablokb, 2024-07-08:
I still believe it has to do with the pin labelled `TP_EN`. The reply to my question about that pin is not satisfactory, because I don't think an enable pin for the touch driver has anything to do with screen initialization, I2C timing or address setting.

I would pull that pin low and see what is happening. If this does not help detecting the device, pull it high and check again.

@Carlos-TGW : please try. And also please post a link to the device on the supplier page. Otherwise, I don't see a chance to help you any further.

Carlos-TGW, 2024-07-09:
Strangely, I used a different size screen, the same GT911 IC, but it could be detected

Carlos-TGW, 2024-07-09:
> I still believe it has to do with the pin labelled `TP_EN`. The reply to my question about that pin is not satisfactory, because I don't think an enable pin for the touch driver has anything to do [...]

I tried the method you mentioned, but it did not work. The supplier of this screen seems to have no online sales channel, so I need to ask before sending it

Carlos-TGW, 2024-07-10:
The reason is that the screen manufacturer did not add a pull-up resistor to the screen. It is invalid to connect the external pull-up resistor, and the manufacturer must add a pull-up resistor inside. The problem has been solved

closed 2024-07-10 by Carlos-TGW

Carlos-TGW, 2024-07-24:
> > ... It is invalid to connect the external pull-up resistor, and the manufacturer must add a pull-up resistor inside. ...
> 
> Glad it is working! Could you explain what you meant by "invalid" above?

Adding an external pullup resistor to the motherboard does not make the touch screen work, but using a touch screen with an internal pullup resistor added by the supplier does.