adafruit/circuitpython #9406: wifi.radio.enabled causes error with debug mode on

tyeth, 2024-07-06:
### CircuitPython version

```python
latest (main) ~9.1beta5
```

### Code/REPL

```python
calling `wifi.radio.enabled` (before deep sleeping)
```

### Behavior

wifi:pm stop, total sleep time: Assertion '*fmt == '%' || !"unsupported fmt char"' failed, at file mpprint.c:586

### Description

Appears the debug message contains "%" which is not supported

jepler, 2024-07-06:
There's a hack somewhere that tries to print espidf messages via mp_printf so they are visible on the repl, not requiring a separate debug uart. I guess this isn't compatible with all formats used in espidf

tyeth, 2024-07-06:
> I think it may contain ... %<something>, where the <something> is not
> supported.

That was my thunk, thanks Dan. I did see a successful log example when
searching for the time related string.
Forgot to mention running on an itsy bitsy esp32

tannewt, 2024-07-08:
This is the printf connector: https://github.com/adafruit/circuitpython/blob/8fbbb598ffd72c93e01c5ed04618e2ce89134cf1/ports/espressif/supervisor/port.c#L489-L497

I do need to comment it out from time to time when debugging too.

jepler, 2024-07-08:
let's just back it out if it causes crashes.

PR #9572 (2024-08-29): espressif: Remove the debug printf adapter
this caused problems.

Closes #9406
commit 4267179b (2024-08-29): espressif: Remove the debug printf adapter

this caused problems.

Closes #9406
closed 2024-08-29 by dhalbert