adafruit/circuitpython #9413: Crashinh CircuitPython. KMK Python firmware, Specifying "KC.MO(1)" in code.py

ddam1k, 2024-07-08:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.5 on 2024-05-22; Maple Computing Elite-Pi with rp2040
Board ID:maple_elite_pi
```

### Code/REPL

```python
import board
import busio

from kmk.kmk_keyboard import KMKKeyboard
from kmk.extensions.media_keys import MediaKeys
from kmk.keys import KC
from kmk.scanners import DiodeOrientation
from kmk.modules.layers import Layers

keyboard = KMKKeyboard()

keyboard.col_pins = (board.D2,board.D3,board.D4,)
keyboard.row_pins = (board.D9,)
keyboard.diode_orientation = DiodeOrientation.COL2ROW

keyboard.modules = [Layers()]
keyboard.extensions = [MediaKeys()]

FN = KC.MO(1)

keyboard.keymap = [
    [KC.A, KC.B, KC.R,],
    [KC.C, KC.D, KC.F,],
]

if __name__ == '__main__':
    keyboard.go()
```

### Behavior

In code.py, (for latest KMK firmware) putting "FN = KC.MO(1)" crashes CircuitPython, enters Safe Mode and reports that a crash occurred.

REPL output:

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.0.5 on 2024-05-22; Maple Computing Elite-Pi with rp2040

### Description

KC.MO(1) crashes, also KC.LT(1). Putting in matrix also crashes. Revert back to 8.2.x can't, just because on my board latest 9.0.0 alpha 1. 
Also tested those versions: 9.0.0 stable, 9.1.0 beta 4, 9.0.5 stable. On all those versions crashes

dhalbert, 2024-07-08:
This was previously reported in #8425, which was closed only because the original author was not able to re-test.

ddam1k, 2024-07-08:
Yes, when it reaches this code line immediately crashes and I hear disconnected sound. After that connecting again in safe mode.

 Also when i put it in matrix, i see in terminal how previous buttons readed and initialized and when reach KC.MO again crashes.
In that way: [KC.A, KC.B, KC.MO(1),]

dhalbert, 2024-07-08:
Do you have a plain Pi Pico you could try this on, using, say, the pins I chose? You don't need to connect anything. I'm puzzled as to why I can't reproduce.

Are you using the kmk directory directly from the cloned repo?

I'm using this kmk commit:
```
ommit 35f2a692c38f98fcf594965b1b6b4933c1add1db (HEAD -> main, origin/main, origin/HEAD)
Author: xs5871 <60395129+xs5871@users.noreply.github.com>
Date:   Sat Jul 6 17:00:30 2024 +0000

    Fix inconsistent macro state when double tapping (#1002)
```

dhalbert, 2024-07-08:
I see this when it starts:
```
code.py output:
12846340 kmk.keyboard: Initialising KMKKeyboard
12846343 kmk.hid: use 6KRO
12846345 kmk.hid: use no pan
12846346 kmk.keyboard: hid=USBHID(REPORT_BYTES=9)
12846350 kmk.keyboard: matrix=['MatrixScanner']
12846353 kmk.keyboard: modules=['Layers']
12846355 kmk.keyboard: extensions=['MediaKeys']
12846377 kmk.keyboard: mem_info used:57088 free:114048
```

ddam1k, 2024-07-08:
Unfortunately i haven't Pi Pico, that's my first board, so also don't know what's a problem. I can tell for sure tomorrow which  commit i use, when will be at computer and test it.

Also forgot to add, that I use ElitePi, and cutted traces and soldered D10 and D11 pins, where on other's board by default is GND. I don't know if it will help, but maybe that what u need. Here's more info: https://docs.keeb.io/elite-pi-guide

jepler, 2024-07-08:
I tested on an Adafruit Feather RP2040 with USB Type A Host (just because I had it handy) and CircuitPython 9.0.5. Like Dan I also did not see a crash.

I made the following change to the reproducer to fit this board:
```
keyboard.col_pins = (board.D10,board.D11,board.D12,)
keyboard.row_pins = (board.D9,)
```

my kmk source files were in ".py" format and from commit [35f2a692c38f98fcf594965b1b6b4933c1add1db](https://github.com/KMKfw/kmk_firmware/tree/35f2a692c38f98fcf594965b1b6b4933c1add1db)

dhalbert, 2024-07-08:
I have a few things for you to try when you have a chance:

1. Try erasing and reformatting the filesystem:
```py
import storage
storage.erase_filesystem()
```
in case this problem is due to some corruption in CIRCUITPY.

2. Try this UF2. It is a maple elite build which allows more time for the crystal oscillator to start up. This is probably not the problem, but it's an easy thing to try. You'll need to unzip it: 
[maple_elite_pi-extra-startup-time-9.1.0-beta.4.uf2.zip](https://github.com/user-attachments/files/16134720/maple_elite_pi-extra-startup-time-9.1.0-beta.4.uf2.zip)

3. Try the regular Pi Pico 9.1.0-beta.4, from https://circuitpython.org/board/raspberry_pi_pico/. You'll need to change the pin names from "D..." to "GP..." but otherwise it should work the same.

jepler, 2024-07-08:
I also tried with the mpy files produced by "make compile" using mpy-cross `CircuitPython 9.1.0-beta.3-50-g9b85fe5766-dirty on 2024-07-08; mpy-cross emitting mpy v6.1` and didn't encounter a problem in this situation either. I got similar messages printed on the repl to what dan saw:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
337870 kmk.keyboard: Initialising KMKKeyboard
337873 kmk.hid: use 6KRO
337878 kmk.hid: use no pan
337878 kmk.keyboard: hid=USBHID(REPORT_BYTES=9)
337881 kmk.keyboard: matrix=['MatrixScanner']
337885 kmk.keyboard: modules=['Layers']
337886 kmk.keyboard: extensions=['MediaKeys']
337905 kmk.keyboard: mem_info used:61040 free:117648
```

jepler, 2024-07-08:
is there any content in your `boot.py`?

dhalbert, 2024-07-08:
And here is another UF2 to try, which specifies a different flash chip. It appears the flash chips chosen in the board def might be wrong, based on an inquiry I made in the keebio discord: https://discord.com/channels/425666562150629376/425668105872867329/1259990905579110591
(keebio discord invite: https://discord.com/invite/NrbY9pn)

[maple_elite_pi-correct-flash-chip-9.1.0-beta.4.uf2.zip](https://github.com/user-attachments/files/16134845/maple_elite_pi-correct-flash-chip-9.1.0-beta.4.uf2.zip)

dhalbert, 2024-07-08:
I tried the Maple Elite-Pi on a Pi Pico, which should emulate the problem, and could still not reproduce. Hmm.

ddam1k, 2024-07-09:
First of all thanks for your time and help. Tried to erase and reformat system, then installed uf2 with more time. For me it started to work. Not crashing anymore. Thanks

dhalbert, 2024-07-09:
@ddam1k Just to be clear, do you mean you tried https://github.com/user-attachments/files/16134720/maple_elite_pi-extra-startup-time-9.1.0-beta.4.uf2.zip? Could you also try the original UF2 from https://circuitpython.org/board/maple_elite_pi/ again? I would like to know if the change I made for https://github.com/user-attachments/files/16134720/maple_elite_pi-extra-startup-time-9.1.0-beta.4.uf2.zip is necessary. Thanks.

ddam1k, 2024-07-09:
Tested now both and both works perfectly. Now I know what helped me with formatting. I would also like to ask what could have influenced the cause of the error. Perhaps it's because sometimes I extracted usb unsafely, or my poor code could have caused an error that required formatting. Also, I encounter an error when connecting the display and encoder. Could this also be a reason? The error occurs if I don't press any keys for 5-10 seconds

```
232111 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
232162 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
232212 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
232262 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
```

dhalbert, 2024-07-09:
> Tested now both and both works perfectly.

OK, great. I think I will submit a PR to correct the flash chip, but otherwise it should be Ok as is.

>Now I know what helped me with formatting. I would also like to ask what could have influenced the cause of the error. Perhaps it's because sometimes I extracted usb unsafely, or my poor code could have caused an error that required formatting.

This is likely to be resetting or unplugging the board before a write has finished. It also might be because a .mpy file was corrupted, and since you loaded the files fresh, you got a good copy.

>Also, I encounter an error when connecting the display and encoder. Could this also be a reason? The error occurs if I don't press any keys for 5-10 seconds

This is something else, but I quite unfamiliar with KMK and don't know what that might be. There are multiple methods named `before_matrix_scan()`. Is there a way to turn on more verbose error reporting, such as file and line, or remove catching the exception so we can see the backtrace?

ddam1k, 2024-07-09:
The code stays almost same, just added from CircuitPython Libs those files: display_notifications, display_shapes, display_text and displayio_ssd1306.mpy.

```
import board
import busio

from kmk.kmk_keyboard import KMKKeyboard
from kmk.keys import KC
from kmk.scanners import DiodeOrientation
from kmk.modules.layers import Layers
from kmk.extensions.media_keys import MediaKeys
from kmk.extensions.display import Display, SSD1306, TextEntry, ImageEntry
from kmk.extensions.display.ssd1306 import SSD1306
from kmk.modules.encoder import EncoderHandler

keyboard = KMKKeyboard()

keyboard.col_pins = (board.D2,board.D3,board.D4,)
keyboard.row_pins = (board.D9,)
keyboard.diode_orientation = DiodeOrientation.COL2ROW

i2c_bus = busio.I2C(board.D1, board.D0)
driver = SSD1306(
    i2c=i2c_bus,
)

display = Display(
    # Mandatory:
    display= driver,
    # Optional:
    width=128, # screen size
    height=32, # screen size
    flip = False, # flips your display content
    flip_left = False, # flips your display content on left side split
    flip_right = False, # flips your display content on right side split
    brightness=0.8, # initial screen brightness level
    brightness_step=0.1, # used for brightness increase/decrease keycodes
    dim_time=20, # time in seconds to reduce screen brightness
    dim_target=0.1, # set level for brightness decrease
    off_time=60, # time in seconds to turn off screen
    powersave_dim_time=10, # time in seconds to reduce screen brightness
    powersave_dim_target=0.1, # set level for brightness decrease
    powersave_off_time=30, # time in seconds to turn off screen
)

encoder_handler = EncoderHandler()

keyboard.modules = [Layers(), encoder_handler]
keyboard.extensions = [MediaKeys(), display]

display.entries = [
    #ImageEntry(image="1.bmp", x=0, y=0),
    TextEntry(text="Longer text that", x=0, y=0,),
    TextEntry(text="has been divided", x=0, y=10,),
    TextEntry(text="for an example", x=0, y=20),
]

FN = KC.MO(1)

keyboard.keymap = [
    [KC.A, KC.B, FN,],
    [KC.C, KC.D, KC.F,],
]

encoder_handler.pins = ((board.D10_GND, board.D11_GND, None,),)
encoder_handler.map = [
    ((KC.UP, KC.DOWN,),),
]

if __name__ == '__main__':
    keyboard.go()
```

After of idling it just start to every millisecond write in REPL:

```
315119 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
315169 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
315219 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
315270 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
315332 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
315382 kmk.extensions.display: before_matrix_scan: TypeError: object with buffer protocol required
```

After pressing any key it stops. After 10-15 second again start and so on.

jepler, 2024-07-09:
Since you are now getting a Python exception it is more likely to indicate a bug in kmk or your keyboard. Please check with kmk people on how to debug this further.

dhalbert, 2024-07-09:
@ddam1k I looked at the debug printing in KMK and I don't see an easy way to get further information. I think this is unrelated to the original problem. You could try asking for support as described here: https://kmkfw.io/support/.  I'm going to close this now -- thank you for the testing you did to narrow this down.

closed 2024-07-09 by dhalbert