adafruit/circuitpython #9423: RUnMode.UF2 stopped working in 9.1.0-beta.4 and later, at least on nRF52840

dhalbert, 2024-07-11:
### CircuitPython version

```python
9.1.0-beta.4
```

### Code/REPL

```python
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.UF2)
microcontroller.reset()
```

### Behavior

Goes back to CircuitPython instead of bootloader

### Additional information

Works in 9.10-beta.3.

Need to look at PR's and see if it's nRF-specific.

jeffsf, 2024-07-11:
Discovered on `Adafruit CircuitPython 9.1.0 on 2024-07-10; nice!nano with nRF52840`

```
UF2 Bootloader 0.8.3 lib/nrfx (v2.0.0) lib/tinyusb (0.12.0-145-g9775e7691) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: nice!nano
Board-ID: nRF52840-nicenano
Date: Jan 29 2024
SoftDevice: S140 6.1.1
```

dhalbert, 2024-07-11:
I thought this might be due to #9160, but that was in 9.1.0-beta.1.

dhalbert, 2024-07-11:
I thought this might be due to https://github.com/adafruit/circuitpython/pull/9160, but that was in 9.1.0-beta.1.

EDIT: might be a combination of https://github.com/adafruit/circuitpython/pull/9160 and https://github.com/adafruit/circuitpython/pull/9344. https://github.com/adafruit/circuitpython/pull/9344 was added in beta.4

Does not fail on a Metro RP2040 on 9.1.0, so I think this is nordic-specific

dhalbert, 2024-08-25:
This works again, probably fixed by #9453.

dhalbert, 2024-08-31:
I just saw #9584, and I can't get it to work again. I thought I just tested on a Feather nRF52840, but it indeed goes back to CircuitPython. Saw same thing on Itsy nRF52840. On the chance it might be the bootloader, I tried with 0.9.3, 0.8.2., and 0.6.1, but they all go to CIRCUITPY.

dhalbert, 2024-08-31:
@jordan1110 watch this issue, which is the same as your #9584.

tannewt, 2024-09-12:
I suspect #9344 broke this. The current implementation only works with the SD (BLE) is running. Fix coming shortly.

PR #9618 (2024-09-12): Set UF2 boot value without SD
Fallback to setting GPREGRET without SD if it isn't active. This fixes resetting into UF2 when BLE isn't active. Fixes #9423
commit 8bd5ce50 (2024-09-12): Set UF2 boot value without SD

Fallback to setting GPREGRET without SD if it isn't active. This
fixes resetting into UF2 when BLE isn't active. Fixes #9423
closed 2024-09-13 by dhalbert