adafruit/circuitpython #9426: CircuitPython crashes if nothing has been drawn on the Seeed Studio Round Display before deinit() functions are called

nbercher, 2024-07-12:
### CircuitPython version

```python
Adafruit CircuitPython 9.0.5 on 2024-05-22; Seeed Studio XIAO ESP32C3 with ESP32-C3FN4
```

### Code/REPL

```python
import board
import busio
import displayio
import gc9a01

# Round Display constants:
ROUND_ROUND_WIDTH  = 240
ROUND_ROUND_HEIGHT = 240

# Board pins:
ROUND_DISPLAY_TFT_CS   = board.D1  # LCD_CS
ROUND_DISPLAY_TFT_DC   = board.D3  # LCD_DC
ROUND_DISPLAY_TFT_BL   = board.D6  # Backlight
ROUND_DISPLAY_TFT_CLK  = board.D8  # Aka board.SCK
ROUND_DISPLAY_SD_MISO  = board.D9  # Aka board.MISO, for SPI read/write devices (eg, SDCard)
ROUND_DISPLAY_TFT_MOSI = board.D10 # Aka board.MOSI, for SPI read only devices (eg, Display)

def display_init():
    spi_bus = busio.SPI(
        clock=ROUND_DISPLAY_TFT_CLK,
        MOSI=ROUND_DISPLAY_TFT_MOSI,
    )
    display_conn = displayio.FourWire(
        spi_bus,
        command=ROUND_DISPLAY_TFT_DC,
        chip_select=ROUND_DISPLAY_TFT_CS,
        reset=ROUND_DISPLAY_TFT_BL,
    )
    display = gc9a01.GC9A01(
        display_conn, rotation=0,
        width=ROUND_ROUND_WIDTH, height=ROUND_ROUND_HEIGHT,
    )
    return spi_bus, display_conn, display

def display_deinit(spi_bus, display_conn, display):
    display_conn.reset() # Don't know yet how to release board.D3 used by display_conn!
    spi_bus.deinit()

## Testing sequence:
# Init:
spi_bus, display_conn, display = display_init()

# Deinit makes CircuitPython to crash:
display_deinit(spi_bus, display_conn, display)
```

### Behavior

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xd0
load:0x403cc710,len:0x7c8
load:0x403ce710,len:0x2518
entry 0x403cc710

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Crikey!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.

### Description

CircuitPython crashes while we call `deinit()` on `displayio.FourWire` object.

Alternatvely, if we draw something on the Display **before** calling `deinit()`, then CircuitPython would only crash at the next soft reset:

```
## Alternate testing sequence:
import time
import terminalio
from adafruit_display_text import label

# Init:
spi_bus, display_conn, display = display_init()

# Draw something
text_area = label.Label(terminalio.FONT, text="Goodbye\n World!", width=128, height=64)
text_area.scale = 4
text_area.x, text_area.y = 40, 80
display.root_group = text_area
time.sleep(.1)

# Deinit runs OK here, but CircuitPython would crash on the next soft reset (Ctrl-D):
display_deinit(spi_bus, display_conn, display)
```

### Additional information

Setup using:
 - Seeed Studio Round Display plugged to the board:
   - Driver `gc9a01.mpy` from circuitpython-community-bundle-9.x-mpy-20240617
   - Switches: 1:ON 2:ON
   - No SD card inserted
   - No RTC battery in use
   - No 3.7v battery in use

tannewt, 2024-07-15:
You aren't deiniting the display so I suspect we need to have it ensure the display bus hasn't been deinit.

nbercher, 2024-07-16:
The crash disappear if I use `displayio.release_displays()`. I assume this what you called deiniting the display.

I let you close the issue if you think it is enough.

Thanks!

tannewt, 2024-07-16:
We can leave it open. It should be easy to guard against this and fixing crashes is always a good idea.

tannewt, 2024-08-21:
Does this crash in 9.1.1? It doesn't for me.

PR #9568 (2024-08-28): Avoid crashing when display components are deinitialized
Now, try_lock (SPI & I2C) & begin_transaction (display bus core) will check that the related objects are still valid first; if they are not, the lock/begin transaction will fail by returning false, rather than "other things" such as raising a Python exception where it is not permitted, accessing invalid memory, etc.

I tested by running a version of the #9426 reproducer on a Raspberry Pi Pico. It was able to complete & soft reset multiple times with no error. I also made sure that basic display functionality still worked on a Feather ESP [... cut]
commit 19e5cf3d (2024-08-28): Avoid crashing when display components are deinitialized

Now, try_lock (SPI & I2C) & begin_transaction (display bus core) will check
that the related objects are still valid first; if they are not,
the lock/begin transaction will fail by returning false, rather than
"other things" such as raising a Python exception where it is not
permitted, accessing invalid memory, etc.

Closes #8278 and Closes #9426
closed 2024-08-28 by tannewt