adafruit/circuitpython #9442: nordic: restarting BLE program causes USB disconnect/reconnect or hard reset (not sure which)

dhalbert, 2024-07-20:


tannewt, 2024-07-22:
Does the board come back up in safe mode? Or is it just USB that is reset?

dhalbert, 2024-07-22:
It does not come back in safe mode. It looks like this:
```
[15:58:35.446] tio v2.7
[15:58:35.446] Press ctrl-t q to quit
[15:58:35.446] Warning: Could not open tty device (No such file or directory)
[15:58:35.446] Waiting for tty device..
[15:58:43.551] Connected
    [***typed ctrl-C***]
Traceback (most recent call last):
  File "code.py", line 17, in <module>
KeyboardInterrupt: 

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.1.0 on 2024-07-10; Adafruit ItsyBitsy nRF52840 Express with nRF52840
>>> 
[15:58:50.265] Disconnected
[15:58:51.266] Warning: Could not open tty device (No such file or directory)
[15:58:51.266] Waiting for tty device..
[15:59:09.489] Connected
    [***typed ctrl-C***]
Traceback (most recent call last):
  File "code.py", line 17, in <module>
KeyboardInterrupt: 

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.1.0 on 2024-07-10; Adafruit ItsyBitsy nRF52840 Express with nRF52840
>>> 
```

tannewt, 2024-07-23:
Weird! You could use UART debugging to try and sort it out.

dhalbert, 2024-07-23:
I tried this on a CLUE, and could not reproduce. I uploaded its bootloader to 0.9.3, still could not reproduce. Then I did `_bleio.adapter.erase_bonding()`, and now I can reproduce. So something about having totally empty bonding is causing the problem.

The display on the CLUE shows that it is doing a hard restart.

dhalbert, 2024-07-24:
Fixed by #9453.

PR #9453 (2024-07-24): bleio_user_reset(): check if BLE running before doing any BLE ops
- Fixes #9442 

`bleio_user_reset()` was trying to do SoftDevice operations after the `Adapter` was disabled, causing `NRF_ERROR_SOFTDEVICE_NOT_ENABLED`, which tried to raise an exception after the VM was shut down. Check before trying to do these operations.
closed 2024-07-24 by dhalbert