adafruit/circuitpython #9443: Screen not working on Adafruit boards with included display - PyGamer, PyPortal

snkYmkrct, 2024-07-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0 on 2024-07-10; Adafruit PyPortal Titano with samd51j20
```

### Code/REPL

```python
any
```

### Behavior

none

### Additional information

Based on someone having an issue in Discord with their PyBadge board, I also tested on my PyGamer, and PyPortal Titano, and they all have the same behaviour. 
The screen is completely black as soon as CP 9.1.0 is installed. 
There is no visible splash screen on reset, and any code using the display seems to be working correctly (no error messages in serial), but doesn't show anything. 

All the tested devices work correctly with 9.0.5

FoamyGuy, 2024-07-22:
I just tested a few of the recent releases and found that `9.1.0-beta.3.uf2` is the most recent beta that allows the display to function properly on the PyPortal Titano at least.   beta.4 has the same issue with display remaining dark as 9.1.0 release.

FoamyGuy, 2024-07-22:
I've narrowed this down further. On the Pyportal Titano the last build that runs successfully is:

[PR9344-f66313c.uf2](https://adafruit-circuit-python.s3.amazonaws.com/bin/pyportal_titano/en_US/adafruit-circuitpython-pyportal_titano-en_US-20240618-main-PR9344-f66313c.uf2)

And the first build where the display stays dark is the next one:

[PR9343-cbd5131.uf2](https://adafruit-circuit-python.s3.amazonaws.com/bin/pyportal_titano/en_US/adafruit-circuitpython-pyportal_titano-en_US-20240621-main-PR9343-cbd5131.uf2)

snkYmkrct, 2024-07-22:
I tested, and the changes in #9343 are the ones breaking the display. 

Specifically, removing the 2 while loops,  `while (tc->COUNT16.SYNCBUSY.bit.CC1 != 0) {  }`  and  `while (tcc->SYNCBUSY.reg != 0) {  }`  
Based on the datasheet, these delays seem to be necessary before you can do the other register operations.

timchinowsky, 2024-07-22:
Sounds good, will need to dig into it.

dhalbert, 2024-07-23:
Fixed by #9447

closed 2024-07-23 by dhalbert

timchinowsky, 2024-07-24:
@snkYmkrct @FoamyGuy thanks for your testing and diagnosis!  I am trying to reproduce what you are seeing and unfortunately so far everything is working fine for me on a PyPortal Titano with 9.1.0.  Can you send me the libraries and application code that you are using?

snkYmkrct, 2024-07-24:
The problem was originally identified because someone on Discord was trying to run this  project [Blinka Jump](https://learn.adafruit.com/blinka-jump-pybadge-game) on the PyBadge, and the screen was all black. Same when I tested on the PyGamer board. 
But the code is irrelevant, the display goes black on all boards (including the Titano) as soon as the CP version 9.1.0 is installed. There is no splash screen, and it's not showing the serial. 

> everything is working fine for me on a PyPortal Titano with 9.1.0. 

Are you sure you are not testing with the latest release, 9.1.1 ? That one contains the fix / reverse for the problem.

dhalbert, 2024-07-24:
I looked at the display initialization on both boards. They are mostly the same but the PWM frequency differs:

pyportal:
```c
        50000); // backlight pwm frequency
```
pyportal titano:
```
        500); // backlight_pwm_frequency
```

timchinowsky, 2024-07-24:
@snkYmkrct see video:  https://github.com/user-attachments/assets/75904714-98ef-48c6-a600-75375f696a63

@dhalbert thanks, I'm thinking there's something timing-related going on.  In the `__init__.py` it ramps the PWM up and down, and the behavior of something like that will change a little bit when you change from blocking to non-blocking.  But so far it is working just fine for me, at both 50000 Hz and 500 Hz.  Using libraries in `.mpy` format except for `adafruit_pyportal` which is in `.py` so I can dink with it.  Was not expecting things to just work...

dhalbert, 2024-07-24:
I believe you might be able to run the regular pyportal UF2 on the Titano, and you'll just see a smaller display.

timchinowsky, 2024-07-24:
What is the expected behavior with no `code.py`?  Where does the display init take place in that case?

dhalbert, 2024-07-24:
It's done before the VM runs, so it can display the startup text and REPL output.

timchinowsky, 2024-07-24:
OK, I see that, so it's pre-`py`

timchinowsky, 2024-07-24:
I'm seeing that inserting a delay before the call to `common_hal_busdisplay_busdisplay_set_brightness` fixes it.  Line is here:

 https://github.com/adafruit/circuitpython/blob/bd834182c43e46606124ce63a604afd91b6568cf/shared-module/busdisplay/BusDisplay.c#L121

2 ms doesn't do it, but 3 ms does, which makes sense for the 500 Hz PWM frequency.

The question seems to be, what happens if you try to update the PWM duty cycle faster than the PWM frequency?
To me the most logical thing would be that the duty cycle of each PWM period would be the last value set before the PWM period started, but I think what samd51 actually does is, once you set it once, it locks you out from setting it again until the next PWM period.  So, if you init a PWM, which sets the duty cycle to some initial value, then immediately try to change it, it fails.

The 9.0.5 code blocks until the lock is released, which solves the problem, in exchange for forcing you to wait.

The thing that got me started with this is that the blocking behavior of the atmel-samd port is different from others like rp2040, where calls to set the duty cycle are instantaneous.  What I don't know is whether the rp2040 PWM would cause similar problems (failing to respond to rapid changes in duty cycle) in similar applications, but just hasn't been asked to do that yet, or if it's fundamentally different.

timchinowsky, 2024-07-25:
Here are some experiments where `code.py` sets the duty cycle of 500 Hz PWM to 10%, then right away sets it to 90%.  The top (green) signal is programmed to go high when duty is set to 10%, and go low when it's set to 90%.  The bottom (yellow) signal is the PWM.

On samd51 running 9.05, both 10% and 90% are always seen, because the code blocks until it is ready to accept another duty cycle value:

[image]

On samd51 running 9.10, mostly only 10% is seen, because the 90% setting is locked out.  (This is why it breaks the backlight.) The first setting wins:

[image]

On rp2040 by contrast, mostly only 90% is seen, because the 10% is overwritten before a new cycle starts.  The last setting wins:

[image]
 
It would be nice to have consistent behavior across ports.  If I had my druthers, I'd want them all to act like rp2040, but hardware differences might make that impossible.

Video:
  
[pwm_comparison.webm](https://github.com/user-attachments/assets/42bf84a8-41d7-41e1-96cd-b0e3df6c933a)

tannewt, 2024-07-25:
The TCC docs say:
> As long as the respective Buffer Valid Status flag (PATTBUFV, PERBUFV
or CCBUFVx) are set to '1', the related SYNCBUSY bits are set (SYNCBUSY.PATT, SYNCBUSY.PER or
SYNCBUSY.CCx), a write to the respective PATT/PATTBUF, PER/PERBUF or CCx/CCBUFx registers will
generate a PAC error,

Try to clear the status bit for buffer valid before re-writing it.

snkYmkrct, 2024-07-25:
> It would be nice to have consistent behavior across ports. 

This delay only happens for the samd51, and removing it was the thing breaking the backlight. 
```
        #ifdef SAM_D5X_E5X
        Tc *tc = tc_insts[t->index];
        while (tc->COUNT16.SYNCBUSY.bit.CC1 != 0) {
        }
        tc->COUNT16.CCBUF[1].reg = adjusted_duty;
        #endif
```
The rest of your changes seemed to work great.

timchinowsky, 2024-07-25:
@tannewt trying various stuff like
```
        tcc->CTRLBCLR.bit.LUPD = 1;
        tcc->STATUS.vec.CCBUFV ^= (1 << channel);
        tcc->CCBUF[channel].reg = adjusted_duty;
```
not seeing any change to 9.1.0 behavior.  But I have only a vague sense of how these registers work...

tannewt, 2024-07-25:
```tcc->STATUS.vec.CCBUFV = (1 << channel);` is mainly what I was thinking of. I was assuming the PAC error would cause the second value to not be saved. Unfortunately, it'll take some trial and error. I don't see any related errata.

PR #9447 (2024-07-22): revert code changes from PR#9343
- Fixes #9443.

Reverts `PWMOut.c` code changes from #9343 to restore screen backlight functionality on SAMD51 boartds.

Tested on a PyPortal (SAMD51) and Hallowing M0 (SAMD21).

I could not do an automated revert because of further changes in later PR's having to do with the tests and/or documentation. So those commits remain.

Interestingly, #9443 did not affect Hallowing M0 (SAMD21).