adafruit/circuitpython #9448: Fix and restore PR: #9343 Fix delays and rounding in samd PWM

dhalbert, 2024-07-22:
The code changes in #9343 were reverted by #9447, because they broke screen backlighting on SAMD51 boards like PyPortal (and maybe other PWM things as well), as reported in #9443.

The intention of #9343 is good, but it needs to be fixed.

Note that #9343 did not affect backlighting on Hallowing M0, so I'm guessing the #9343 changes were OK for SAMD21 but not SAMD51.

@timchinowsky I hope you are able to take a look at this. Thanks.

timchinowsky, 2024-07-22:
Yes, thanks for reverting, I am planning to investigate with PyPortal.

snkYmkrct, 2024-07-23:
> Note that https://github.com/adafruit/circuitpython/pull/9343 did not affect backlighting on Hallowing M0, so I'm guessing the https://github.com/adafruit/circuitpython/pull/9343 changes were OK for SAMD21 but not SAMD51.

After further testing, it looks like the backlight breaking change was only the removal of the first while loop (` while (tc->COUNT16.SYNCBUSY.bit.CC1 != 0) {}` ), the second loop had no effect on the display.
The first while loop is in a SAMD51 specific section of the code, so this behavior makes sense then.