adafruit/circuitpython #9451: Remove `sys.print_traceback()`

tannewt, 2024-07-23:
It isn't in CPython and we've had the `traceback` module since at least 8.2.x.

eightycc, 2025-03-18:
I'm unable to locate `sys.print_traceback()`. There is a `sys.print_exception()`, but it is part of the upstream Micropython `python-stdlib`, which is what I think the doc change in #9427 is referring to. Since it's upstream I don't think we can deprecate or remove it?

dhalbert, 2025-03-18:
You're right, there is no `print_traceback`.

I looked for `MP_QSTR_print_exception` and it's in `py/modsys.c` (for `sys`) and in `shared-bindings/traceback/__init__.c` (for `traceback`). The latter _is_ in CPython, so we wouldn't remove that.

`sys.print_exception()` is present in `tests/basics/sys_tracebacklimit.py` and in `frozen/Adafruit_CircuitPython_asyncio/asyncio/stream.py`, so we'd need to fix these up.

We can defer this for now. I am usually the one to update asyncio, usually when I do a merge from upstream.

We don't use `micropython-lib` despite the fact that it's in `lib`. Or maybe it' used to help run some tests -- I'm not sure.

tannewt, 2025-03-19:
It is in the current build:
```
Adafruit CircuitPython 9.2.4-330-gea45b0e2cb-dirty on 2025-03-18; Adafruit Fruit Jam with rp2350b
>>> import sys
>>> dir(sys)
['__class__', '__name__', '__dict__', 'argv', 'byteorder', 'exit', 'implementation', 'maxsize', 'modules', 'path', 'platform', 'print_exception', 'stderr', 'stdin', 'stdout', 'version', 'version_info']
```

I suspect we'll need to modify modsys.

PR #10332 (2025-05-12): Remove non-standard sys.print_exception()
It already warned about it.

Fixes #9451
commit 3435f473 (2025-05-12): Remove non-standard sys.print_exception()

It already warned about it.

Fixes #9451
closed 2025-05-12 by tannewt