adafruit/circuitpython #9460: Watchdog timer expired running polling loop

mMerlin, 2024-07-25:
I have a script to explore working with SSE. It «seems to» work. I can connect and disconnect browser sessions. However, if I then just leave it running with no connections, and come back later (maybe an hour), the app is frozen. ctrl+c in the serial window aborts and gives:
```text
192.168.20.105 -- "GET /sse" 360 -- "200 OK" 101 -- 126ms
192.168.20.105 -- "GET /favicon.ico" 390 -- "404 Not Found" 90 -- 18ms
client ('192.168.20.105', 8332) went away: code 32
client ('192.168.20.108', 47266) went away: code 32
client ('192.168.20.108', 11392) went away: code 32

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit-Qualia-S3-RGB666 with ESP32S3
>>> 
>>> 
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.
```
The first part of that is the final log messages while it was running. This has occurred several times with different versions of the code. Including (I think) the httpserver_sse.py example in the repo that I started from. I don't know if it it total run time, or total idle time that triggers this.

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit-Qualia-S3-RGB666 with ESP32S3
Board ID:adafruit_qualia_s3_rgb666

Here is code that 'almost' matches what produced the above log. Almost, because the sequence has been: run the script, do some testing, then edit the offline version. In this case, the edit was cleanup to remove some debug messages, tweak the remaining, then start working on a more full html file. Got called away for awhile, when returned, found it locked up. again. Some of those lockup resulted in CIRCUITPY becoming read only. I had to do the file system erase to get it back.
```python
# SPDX-FileCopyrightText: 2024 H Phil Duby
# SPDX-License-Identifier: MIT

"""
HTTP Server with SSE request handling for ambient environment data
"""

from time import monotonic_ns
try:
    # not used at runtime, useful for IDE and documentation
    from typing import NoReturn, Dict, Tuple
except ImportError:
    pass
import microcontroller  #type:ignore
import socketpool  #type:ignore
import wifi  #type:ignore
from adafruit_httpserver import FileResponse, Request, Route, Server, SSEResponse, GET

class AmbientHTTPServer:
    """Serve ambient environment information to clients"""
    def __init__(self):
        print('Initializing server...')  # TRACE
        pool = socketpool.SocketPool(wifi.radio)
        self.server = Server(pool, debug=True)
        self.sse_response: Dict[Tuple[str, int], SSEResponse] = {}
        self.next_event_time_ns = monotonic_ns()
        self.setup_routes()

    def client(self, request: Request) -> FileResponse:
        """Return HTML page that initiates SSE communication"""
        print(f'Serving HTML to {request.client_address}')  # DEBUG
        return FileResponse(request, 'index.html', '/client')

    def sse_client(self, request: Request) -> SSEResponse:
        """Establish SSE communication with client"""
        if request.client_address in self.sse_response:
            raise NameError(f'Client {request.client_address} already connected')
        print(f'Connecting client {request.client_address}')  # DEBUG
        response = SSEResponse(request)
        self.sse_response[request.client_address] = response
        print(f'SSE response created for {request.client_address}')  # DEBUG
        return response

    def setup_routes(self) -> None:
        """Create web server routes for application"""
        self.server.add_routes([
            Route('/client', GET, self.client),
            Route('/sse', GET, self.sse_client)
        ])

    def send_event_to_clients(self) -> None:
        """Send fresh data to all connected SSE clients"""
        if self.sse_response and self.next_event_time_ns < monotonic_ns():
            cpu_temp = round(microcontroller.cpu.temperature, 2)
            for client_id, client in list(self.sse_response.items()):
                # print(f'Sending event {cpu_temp} to client {client_id} for {client._request}')  # DEBUG pylint:disable=protected-access
                try:
                    client.send_event(str(cpu_temp))  # send data to client
                except BrokenPipeError as be:
                    print(f'Client {client_id} disconnected: code {be}')
                    del self.sse_response[client_id]
            self.next_event_time_ns = monotonic_ns() + 1000000000  # 1 second

    def start(self) -> NoReturn:
        """Start the application web server"""
        self.server.start(str(wifi.radio.ipv4_address))
        # Except: RuntimeError: Cannot start server on None:5000
        # Except: gaierror: (-2, 'Name or service not known')
        while True:
            self.server.poll()  # watch for new connection or page requests
            self.send_event_to_clients()

if __name__ == "__main__":
    instance = AmbientHTTPServer()
    print('Server instantiated')  # TRACE
    instance.start()
```
When it locks up, the only thing it is doing is polling, because self.sse_response is empty.

The index.html file referenced is:
```html
<html lang="en">
    <head>
        <title>Server-Sent Events Client</title>
    </head>
    <body>
        <p>CPU temperature: <strong>-</strong>&deg;C</p>
        <script>
            const eventSource = new EventSource('/sse');
            const cpuTemp = document.querySelector('strong');

            eventSource.onmessage = event => cpuTemp.textContent = event.data;
            eventSource.onerror = error => cpuTemp.textContent = error;
        </script>
    </body>
</html>
```

I'm not setup to create a debug version of circuitpython, but with a little guidance, I probably could add some logging (with timestamps) to a local copy of HTTPServer.Server

mMerlin, 2024-07-25:
Can I help tracking it down? I don't know what it will take to reliably reproduce the failure. I'm a programmer, but not really set up to do debug builds. Main environment is Fedora. I can spin up a VM if other environment (not Windows or Mac) is better for building.

dhalbert, 2024-07-25:
The "internal watchdog timer" that was triggered happens because something was taking far too long, blocking other operations (for example, a tight infinite loop). It reflects a bug, not just a timeout that's too short.

If you can reduce your program to the smallest example that still shows the problem, that would be great. We'll have to let it sit for a while to trigger the error, it seems like.

mMerlin, 2024-07-28:
I did some more testing. I have sample code that reliably triggers a non-responsive httpserver, and most of the time safe mode on restart. My initial pointer to .poll() is back in the mix. I tried a couple of variations. When I use server.server_forever(…), I do not see the failures. Change that to start plus while True and server.poll() triggers the problem. Idle time is also part of the mix. If I do a fresh request every 5 minutes, the error does not trigger … until I leave it idle for about 15 minutes.
```python
# SPDX-FileCopyrightText: 2024 H Phil Duby
# SPDX-License-Identifier: MIT

import sys
from time import monotonic_ns
import socketpool
import wifi
from adafruit_httpserver import Server, Request, Response

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, "/static", debug=True)

@server.route("/")
def base(request: Request):
    request_ref = monotonic_ns()
    print(f'{request_ref}: request for page {TAG}')  # LOG
    content = f"""<!DOCTYPE html>
<html><head><title>Watchdog {TAG}</title></head>
<body><p>{TAG} {request_ref}: <span id="datetime"></span></p>
<script>var date = new Date();document.getElementById("datetime").innerHTML = date;
</script></body></html>
"""
    return Response(request, content, content_type="text/html")

TAG = 'simple debug 002'
print(f'{monotonic_ns()}: Starting page {TAG}')  # LOG
print(f'{sys.implementation.name} {sys.implementation.version}\n{sys.implementation._machine}')
server.start(str(wifi.radio.ipv4_address))
while True:
    server.poll()
```
Smaller code fails, but this keeps enough instrumentation/logging to match events (when it is still working).
log and browser capture using serve_forever
```text
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
264137084966: Starting page simple debug 001
circuitpython (9, 1, 1)
Adafruit-Qualia-S3-RGB666 with ESP32S3
Started development server on http://192.168.20.230:5000
277870208741: request for page simple debug 001
192.168.20.108 -- "GET /" 397 -- "200 OK" 340 -- 19ms
192.168.20.108 -- "GET /favicon.ico" 373 -- "404 Not Found" 131 -- 29ms
336848815920: request for page simple debug 001
192.168.20.108 -- "GET /" 448 -- "200 OK" 340 -- 269ms
1519142303476: request for page simple debug 001
192.168.20.108 -- "GET /" 397 -- "200 OK" 341 -- 21ms
2810475250253: request for page simple debug 001
192.168.20.108 -- "GET /" 397 -- "200 OK" 341 -- 25ms
2846935180672: request for page simple debug 001
192.168.20.108 -- "GET /" 397 -- "200 OK" 341 -- 20ms
2937651519777: request for page simple debug 001
192.168.20.108 -- "GET /" 397 -- "200 OK" 341 -- 306ms
Traceback (most recent call last):
  File "main.py", line 28, in <module>
  File "adafruit_httpserver/server.py", line 190, in serve_forever
KeyboardInterrupt:

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit-Qualia-S3-RGB666 with ESP32S3
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
[... 137 lines cut ...]
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

```text
simple debug 002 4025758972170: Sat Jul 27 2024 19:52:33 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 4373888305672: Sat Jul 27 2024 19:58:21 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 4678340362556: Sat Jul 27 2024 20:03:25 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 4972010681165: Sat Jul 27 2024 20:08:19 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 5277275360120: Sat Jul 27 2024 20:13:24 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 5571675323493: Sat Jul 27 2024 20:18:18 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 5895380767834: Sat Jul 27 2024 20:23:42 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 6184399719242: Sat Jul 27 2024 20:28:31 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 6488390472423: Sat Jul 27 2024 20:33:35 GMT-0600 (Mountain Daylight Saving Time)
simple debug 002 7697915405278: Sat Jul 27 2024 20:53:45 GMT-0600 (Mountain Daylight Saving Time)
22:20 Unable to connect
```

mMerlin, 2024-07-29:
It appears that this is not going to affect the project that it was initially encountered in. The problem was seen when doing development and testing of separate sections. With everything (so far) merged together, there is no 'idle time' that is needed to trigger the failure. The project shows the sensor data both on a local display, and on web pages. It seems that the processing needed for the display portion is enough to prevent the watchdog timeout seen in server.poll(). Even with the web server totally idle (no active connections).

Which means that some sort of heartbeat may be a work around. That also could mean that figuring out just what that heartbeat needs could point to where the watchdog is being triggered.

michalpokusa, 2024-07-29:
Possibly related https://github.com/adafruit/circuitpython/issues/9428

mMerlin, 2024-07-29:
Reading over the code provided in that, this does look like exactly the same case: CP 9.1.1, ESP32-S3, server.poll() that is idle 'long enough', watchdog timer. Smaller code sample here, which could be shrunk further if logging not useful.

michalpokusa, 2024-07-29:
If the problem can be prevented by adding some code to `.poll()` that heartbeats or restart the watchdog timer i can make a PR for `adafruit_httpserver`, although I would like to know whether it is possible and if it is something that should be done there, because maybe that is not something that the library should do.

mMerlin, 2024-07-29:
Here is a bit smaller sketch that can produce the same symptoms. Note that the first cycle had the server stop responding, but on control C abort and control D reload, it restarted successfully. That restart also did not stop responding after a 16 minute idle period, then a 22 minute idle period. It did not respond after a following 27 minute idle period, and the following forced abort and restart showed the watchdog timer expired problem. For most of my previous testing 15 minutes idle time was sufficient to trigger the watchdog timer.
```python
import socketpool
import wifi
from adafruit_httpserver import Server, Request, Response

pool = socketpool.SocketPool(wifi.radio)
server = Server(pool, "/static", debug=True)

@server.route("/")
def base(request: Request):
    content = """<!DOCTYPE html>
<html><head><title>Watchdog check</title></head>
<body><p><span id="datetime"></span></p>
<script>document.getElementById("datetime").innerHTML = (new Date()).toISOString();
</script></body></html>
"""
    return Response(request, content, content_type="text/html")

server.start(str(wifi.radio.ipv4_address))
while True:
    server.poll()
```

```text
from the web browser:
2024-07-29T08:26:57.426Z
2024-07-29T08:27:36.756Z
08:44 Unable to connect
2024-07-29T08:45:30.908Z
2024-07-29T09:06:41.214Z
2024-07-29T09:06:41.214Z
2024-07-29T09:28:24.754Z
09:55 Unable to connect

serial output:
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Started development server on http://192.168.20.230:5000
192.168.20.108 -- "GET /" 363 -- "200 OK" 298 -- 31ms
192.168.20.108 -- "GET /" 363 -- "200 OK" 298 -- 16ms
Traceback (most recent call last):
  File "main.py", line 20, in <module>
  File "adafruit_httpserver/server.py", line 432, in poll
KeyboardInterrupt:

Code done running.

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit-Qualia-S3-RGB666 with ESP32S3
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
main.py output:
Started development server on http://192.168.20.230:5000
192.168.20.108 -- "GET /" 363 -- "200 OK" 298 -- 16ms
192.168.20.108 -- "GET /client" 412 -- "404 Not Found" 126 -- 141ms
192.168.20.108 -- "GET /" 363 -- "200 OK" 298 -- 31ms
192.168.20.108 -- "GET /" 363 -- "200 OK" 298 -- 16ms
192.168.20.108 -- "GET /" 363 -- "200 OK" 298 -- 47ms

Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit-Qualia-S3-RGB666 with ESP32S3
>>>
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

dave4jr, 2024-08-20:
@mMerlin I had the same issue and I noticed that when I commented this out:
`pool = socketpool.SocketPool(wifi.radio)`
Then it stopped doing that. I'm running CP 9.1.1 on an ESP32-S3 Feather

dave4jr, 2024-08-21:
@mMerlin I'm experiencing the same issue. ESP32-S3 Feather TFT REV, CP 9.1.1.. Basically just connecting to wifi in beginning of script and reading the voltage on A4 inside of a while loop. I go into safemode from the same watchdog timer expiration error after about 20min.

jepler, 2024-08-29:
I reproduced this **once** on an Adafruit Feather ESP32-S3 TFT and 9.2.0-alpha.2351-7-g19e5cf3d8f using a very simple test program that eliminates `adafruit_httpserver`:
```py
import errno
import wifi
import socketpool

def wrap_accept(s):
    try:
        return s.accept()
    except OSError as e:
        if e.errno == errno.EAGAIN:
            return None, ("", 0)
        raise

socket = socketpool.SocketPool(wifi.radio)
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.listen(10)
s.setblocking(False)
while True:
    s1, peerinfo = wrap_accept(s)
    if s1 is not None:
        print(peerinfo)
        s1.close()
```
I ran this test without even being connected to wifi, and it's also a slightly dubious program because it did not `bind` a specific port. I paste this into the repl using paste mode. I saw this safe-mode-reset after less than 5 minutes on the first run, but now I haven't seen it again for ~10 minutes:
```
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import errno
=== import wifi
=== import socketpool
=== 
=== def wrap_accept(s):
===     try:
===         return s.accept()
===     except OSError as e:
===         if e.errno == errno.EAGAIN:
===             return None, ("", 0)
===         raise
=== 
=== socket = socketpool.SocketPool(wifi.radio)
=== s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
=== s.listen(10)
=== s.setblocking(False)
=== while True:
===     s1, peerinfo = wrap_accept(s)
===     if s1 is not None:
===         print(peerinfo)
===         s1.close()
=== 

[tio 10:32:38] Disconnected
[tio 10:32:42] Connected

Adafruit CircuitPython 9.2.0-alpha.2351-7-g19e5cf3d8f on 2024-08-28; Adafruit Feather ESP32-S3 TFT with ESP32S3
>>> 
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
Internal watchdog timer expired.
Press reset to exit safe mode.
```

I'm a little conflicted about posting this info since I only saw a WDT reset once -- there may be some other hidden variable. but I think it's good info that MAYBE this simple script is actually a reproducer.

jepler, 2024-08-29:
I did eventually get a 2nd crash with just about the same code. That run took almost 1 hour to crash.

<details>

```
[11:44:01] Traceback (most recent call last):
[11:44:01]   File "code.py", line 1
[11:44:01] SyntaxError: invalid syntax
[11:44:01] 
[11:44:01] Code done running.

[11:44:01] Press any key to enter the REPL. Use CTRL-D to reload.
[11:44:02] 
[11:44:02] Adafruit CircuitPython 9.2.0-alpha.2351-7-g19e5cf3d8f on 2024-08-28; Adafruit Feather ESP32-S3 TFT with ESP32S3
[11:44:02] >>> 
[11:44:02] >>> 
[11:44:02] paste mode; Ctrl-C to cancel, Ctrl-D to finish
[11:44:02] === import errno
[11:44:02] === import wifi
[11:44:02] === import socketpool
[11:44:02] === 
[11:44:02] === def wrap_accept(s):
[11:44:02] ===     try:
[11:44:02] ===         return s.accept()
[11:44:02] ===     except OSError as e:
[11:44:02] ===         if e.errno == errno.EAGAIN:
[11:44:02] ===             return None, ("", 0)
[11:44:02] ===         raise
[11:44:02] === 
[11:44:02] === socket = socketpool.SocketPool(wifi.radio)
[11:44:02] === with socket.socket(socket.AF_INET, socket.SOCK_STREAM) as s:
[11:44:02] ===     s.listen(10)
[11:44:02] ===     s.setblocking(False)
[11:44:02] ===     while True:
[11:44:02] ===         s1, peerinfo = wrap_accept(s)
[11:44:02] ===         if s1 is not None:
[11:44:02] ===             print(peerinfo)
[11:44:02] ===             s1.close()
[11:44:02] === 
[11:44:03] 
[tio 12:37:03] Disconnected
[tio 12:37:06] Connected
[12:37:06] soft reboot

Auto-reload is off.
[12:37:11] Running in safe mode! Not running saved code.

[12:37:11] You are in safe mode because:
[12:37:11] CircuitPython core code crashed hard. Whoops!
[12:37:11] Hard fault: memory access or instruction error.
[12:37:11] Please file an issue with your program at github.com/adafruit/circuitpython/issues.
[12:37:11] Press reset to exit safe mode.

[12:37:11] Press any key to enter the REPL. Use CTRL-D to reload.
```

</details>

jepler, 2024-08-30:
I didn't find any further useful information during today's debugging session, and I'm not planning to work further on this issue right now.

tannewt, 2024-09-12:
Moved to 9.x.x because we need a clearer reproducer to debug.