adafruit/circuitpython #9464: Excessive power drain in deep sleep on ESP32-S3

dy0x, 2024-07-26:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.1 on 2024-07-23; ProS3 with ESP32S3
```

### Code/REPL

```python
import alarm
import board
import time
from microcontroller import watchdog as w
from watchdog import WatchDogMode

# Watchdog and Supervisor
w.timeout = 300
w.mode = WatchDogMode.RESET

print("test")
time.sleep(1)

w.feed()
    
wakeup_period = time.monotonic() + int(30)
time_alarm = alarm.time.TimeAlarm(monotonic_time=wakeup_period)
alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```

### Behavior

ESP32-S3 consumes ~28mA in deep sleep on 9.1.0/9.1.1 but only ~26µA on 9.0.5

```
Auto-reload is off.
code.py output:
Wi-Fi: off | BLE:Off | code.py | 9.1.1test
Wi-Fi: off | BLE:Off | Done | 9.1.1
Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
Pretending to deep sleep until alarm, CTRL-C or file write.
```

### Description

In terms of functionality, everything runs as expected. However, I noticed after updating from 9.0.5 to 9.1.1, my code consumes substantially more power in deep sleep. In CircuitPython 9.1.0 and 9.1.1, the provided code runs on ESP32-S3 in deep sleep at [~28mA](https://github.com/user-attachments/assets/99896c5a-f80c-453b-9e39-62f604b0a4ab). Whereas if I run this code using CircuitPython 9.0.5, it deep sleeps at [~26µA](https://github.com/user-attachments/assets/1fe7964f-b440-42a3-8e43-3fd42e3cd3f2).

### Additional information

Things I have tried:
- Different hardware/power configurations including untouched
- Multiple brand new ESP32 ProS3 boards
- Main code and provided simplified code
- Different library versions
- Forcing BLE to be disabled with the supervisor

UnexpectedMaker, 2024-07-26:
Maybe something is keeping LDO2 enabled? That shouldn't cause 28mA but might cause a few mA.  Can you confirm there is no voltage on the 3V3_2 pin when it's sleeping?

dy0x, 2024-07-28:
Yeah, seems like there is a voltage on 3V3_2 during sleep.

UnexpectedMaker, 2024-07-28:
And if you pull IO17 LOW before entering deep sleep, does it keep it low (LDO2 off)?

dy0x, 2024-07-28:
No, it only stays low until deep sleep starts, and then it goes back to ~3.3V

UnexpectedMaker, 2024-07-28:
Ok, that shouldn't really be possible at a HW level as the LDO enable is controlled by a logical AND and requires both Io17 and VDD_SPI to be high, and VDD_SPI is shut down during deep sleep.

Just to be sure - you're not testing with power from USB/5V right?  It's being powered from a battery?

dy0x, 2024-07-28:
Yep, I'm testing on two separate boards. One with alkaline batteries, the other connected to a power profiler on 3V3_1.

UnexpectedMaker, 2024-07-29:
I have no idea what has changed under the hood in CP for 9.1, so I'll need to make some enquiries.
I know there's a way in the board configs to tell CP not to touch certain IO, but that should not be needed as it was working correctly before 9.1.

@dhalbert Can you think of anything off the top of your head (apart from possible IDF level changes) that could have changed the behaviour on the IO when going into deep sleep? Maybe some other bug fix that might have caused some knock-on effect?

dhalbert, 2024-07-29:
We updated to ESP-IDF 5.2.2. Also #9324 was merged.

UnexpectedMaker, 2024-07-29:
> We updated to ESP-IDF 5.2.2. Also #9324 was merged.

Ok, so the only way to fix this now for my boards is to add something like this to `board.c` ?
```
void board_deinit(void) {
  // Turn off LDO2
  gpio_set_direction(17, GPIO_MODE_DEF_OUTPUT);
  gpio_set_level(17, false);
}
```

Didn't there used to be a way to tell CP not to touch certain IO when going to sleep?

dhalbert, 2024-07-29:
I haven't yet, but I think looking carefully at the code in #9324 would be good.

Yes, we had "never reset" certain pins, and "never ever reset" certain pins. Some of that logic has been simplified. A code comparison between 9.0.5 and 9.1.1 would be helpful, or a bisect.

tannewt, 2024-07-29:
> Didn't there used to be a way to tell CP not to touch certain IO when going to sleep?

[`alarm.exit_and_deep_sleep_until_alarms()`](exit_and_deep_sleep_until_alarms) has a `preserve_dios` kwarg that you can give DigitalInOuts to be preserved.

dhalbert, 2024-10-02:
I think this is the same as #9626. There were some changes in the BLE workflow logic. I'm closing this in favor of #9626.

PR #9679 (2024-10-04): Correct `ble_serial_connected()` when BLE workflow was never started
- Fixes #9626.
- Fixes #9464 (already closed in favor of #9626).

`ble_serial_connected()` was checking `_tx_packet_buffer.conn_handle` as an indicator of whether BLE was connected. But that value was not initialized when BLE hadn't started up at all, and so `ble_serial_connected()` was returning `true` incorrectly in that case. The check should have also included a check for deinited. I added that and refactored it into a common-hal routine.

I asked the issue authors to test. It fixed the problem for at least one of them: h [... cut]
closed 2024-10-02 by dhalbert