adafruit/circuitpython #9465: "board" module still gets incorrect docs

jepler, 2024-07-26:
### CircuitPython version

```python
9.1.x, main
```

### Code/REPL

```python
# readthedocs problem
```

### Behavior

https://docs.circuitpython.org/en/latest/shared-bindings/board/index.html is empty, when it should have content

### Description

During readthedocs, the empty `__init__.py` is used for generating the documentation of this module, rather than the `__init__.pyi` file:

```
[AutoAPI] Reading files... [ 47%] /home/docs/checkouts/readthedocs.org/user_builds/circuitpython/checkouts/9.1.x/circuitpython-stubs/board/__init__.py
```

### Additional information

I thought I had fixed this in #9431 but apparently my fix was incorrect.

jepler, 2024-07-26:
It's still wrong in main branch now that it's been merged up: https://docs.circuitpython.org/en/latest/shared-bindings/board/index.html

I am perplexed but don't have any time to work on it now.

jepler, 2024-07-26:
or maybe I needed to shift-reload. can someone else take a look and close if appropriate? I'm at my wits' end.

FoamyGuy, 2024-07-29:
It appears to me that this is resolved now. I see the content inside the board module docs page on the currently live docs. I also saw an actions run that was previously failing due to this that was able to pass now.

[image]

PR #9466 (2024-07-26): Only use pyi files for autoapi doc generation
Closes #9465

Please check the RTD html build when reviewing this bug! Last time we thought we fixed this, I think only local builds were checked (or something? not sure why the previous PR passed muster)

Will want to merge 9.1.x up to main after this. This is required for fixing the builds of various bundle libraries, fixing e.g., the failure seen in https://github.com/adafruit/Adafruit_CircuitPython_EMC2101/pull/34
commit a0558428 (2024-07-26): Only use pyi files for autoapi doc generation

Closes #9465
closed 2024-07-29 by FoamyGuy