adafruit/circuitpython #9472: Feather ESP32-S3 TFT deletes user added contents of CP drive

BlitzCityDIY, 2024-07-29:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit Feather ESP32-S3 TFT with ESP32S3
```

### Code/REPL

```python
import os
import ssl
import time
import wifi
import board
import displayio
import terminalio
import socketpool
import adafruit_requests
import adafruit_ntp

displayio.release_displays()
display = board.DISPLAY

# The time of the thing!
EVENT_YEAR = 2024
EVENT_MONTH = 8
EVENT_DAY = 16
EVENT_HOUR = 0
EVENT_MINUTE = 0
# we'll make a python-friendly structure
event_time = time.struct_time((EVENT_YEAR, EVENT_MONTH, EVENT_DAY,
                               EVENT_HOUR, EVENT_MINUTE, 0,  # we don't track seconds
                               -1, -1, False))  # we dont know day of week/year or DST

wifi.radio.connect(os.getenv("CIRCUITPY_WIFI_SSID"), os.getenv("CIRCUITPY_WIFI_PASSWORD"))
pool = socketpool.SocketPool(wifi.radio)
splash = displayio.Group()

ntp = adafruit_ntp.NTP(pool, tz_offset=0, cache_seconds=3600)
now = ntp.datetime
refresh_time = None

while True:
    # only query the online time once per hour (and on first run)
    if (not refresh_time) or (time.monotonic() - refresh_time) > 3600:
        try:
            print("Getting time from internet!")
            now = ntp.datetime
            refresh_time = time.monotonic()
        except RuntimeError as e:
            print("Some error occured, retrying! -", e)
            continue
    print(now)
    now = time.localtime()
    remaining = time.mktime(event_time) - time.mktime(now)
    print("Time remaining (s):", remaining)

    secs_remaining = remaining % 60
    remaining //= 60
    mins_remaining = remaining % 60
    remaining //= 60
    hours_remaining = remaining % 24
    remaining //= 24
    days_remaining = remaining
    print("%d days, %d hours, %d minutes and %s seconds" %
          (days_remaining, hours_remaining, mins_remaining, secs_remaining))
    # update every 10 seconds
    time.sleep(10)
```

### Behavior

I've replicated this four times now but I can't tell what exactly is causing it because it's a bit random. The code runs fine initially, then if I update the code by saving, it seems like the board freezes up and the REPL becomes unresponsive. If I hit reset on the board, "Hello World" prints on the display. The code I was working on along with any installed libraries, the contents of settings.toml and any additional files are gone.

### Description

Initially this was occurring when I was adding bitmap or font files. As I tried to get the program smaller that's when I was able to reproduce by saving the code.

BlitzCityDIY, 2024-07-29:
i switched over to an earlier version (`Adafruit CircuitPython 9.0.4 on 2024-04-16; Adafruit Feather ESP32-S3 TFT with ESP32S3`) and the issue doesn't seem to be present

RetiredWizard, 2024-07-30:
How were you running the code? In a code.py started on board reset, or through an IDE like Thonny or the Web workflow editor?

I've seen something similar when the flash file system was created but damaged. The host OS would mount the CIRCUITPY drive and appear to write/update files but nothing was actually written. I think the Host OS would hold a cache so it would appear from the host side that the changes were made but if the micro controller attempted to run or access a file the changes wouldn't be there.

If you were running your referenced code from a code.py after a reset then you're seeing something different  but if you were running the code from an IDE, then I believe the code would be run via the REPL and the files may never have made it to the micro controller flash.

tannewt, 2024-08-21:
It is possible this was caused by #9545 because the display release will deinit the SPI bus incorrectly. Let's see if the next release has this problem.

dhalbert, 2024-09-17:
@BlitzCityDIY Did you say that you are still seeing this?  Not sure if it can be replicated reliably.

BlitzCityDIY, 2024-09-17:
i just tested with 9.1.4 and could not recreate it on the s3 tft. previously i could when iterating on code so i think this is good to close

dhalbert, 2024-09-17:
Great, thanks!

closed 2024-09-17 by dhalbert