adafruit/circuitpython #9473: RP2040 Prop Maker becomes non-writable

dolson56, 2024-07-29:
### CircuitPython version

```python
CP9.1.1
Boot_out.txt:
Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit Feather RP2040 Prop-Maker with rp2040
Board ID:adafruit_feather_rp2040_prop_maker
UID:DF625857C7736034
```

### Code/REPL

```python
any code. the issue is in saving to the micro
# SPDX-FileCopyrightText: 2020 Jerry Needell for Adafruit Industries
# SPDX-License-Identifier: MIT

# Example to send a packet periodically between addressed nodes

import adafruit_ds3231
from time import mktime

import time
import board
import busio
import digitalio
import adafruit_rfm69
import displayio
import rotaryio
import terminalio
from adafruit_display_text import label
import adafruit_displayio_ssd1306
from i2cdisplaybus import I2CDisplayBus
import microcontroller
from math import atan2, degrees
import adafruit_lis3mdl
from audiocore import WaveFile

# for I2S audio with external I2S DAC board
import audiobusio

# I2S audio on PropMaker Feather RP2040
power = digitalio.DigitalInOut(board.EXTERNAL_POWER)
power.switch_to_output(value=True)
audio = audiobusio.I2SOut(board.I2S_BIT_CLOCK, board.I2S_WORD_SELECT, board.I2S_DATA)

wave_file = open("LeFreak.wav", "rb")
wave = WaveFile(wave_file)
playing = False

oled_reset = board.D4
i2c = board.I2C()
ds3231 = adafruit_ds3231.DS3231(i2c)
months = ("Jan", "Feb", "Mar", "Apr", "May", "Jun", "Jul", "Aug", "Sept", "Oct", "Nov", "Dec")
currentTime  = ds3231.datetime

def which_day(t):
    # Lookup table for names of days (nicer printing).
    days = ("Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday", "Sunday")
    this_day = days[int(t.tm_wday)]
#    print("function which_day thinks the weekday is {}".format(this_day))
    count = "first"
    if int(t.tm_mday)<8:
        count = "first"
    elif int(t.tm_mday) < 15:
        count = "second"
    elif int(t.tm_mday) < 22:
        count = "third"
    elif int(t.tm_mday) < 29:
        count = "fourth"
[... 320 lines cut ...]

        if AlarmButton.value == False:
            discoBall = True
            AlarmButtonLight.value = True
            audio.play(wave)
            time.sleep(5)
            AlarmButtonLight.value = False
            audio.stop()
    time.sleep(0.01)
```

### Behavior

Any code can be used to reproduce. The issue is the micro becomes un-writable. Furious, random unplugs and restarts of Mu can sometimes briefly recover the write-permission. No discernable pattern yet. It also helps if you curse out loud while doing this.

### Description

Over time, the micro becomes more and more likly to appear as non-writable. You can't save to it. You see this when developing with Mu, however, you can verify the issue with windows file explorer. Also shows up as non-writable. I have tried to reinstall CP as recommended on CP.org(link) and this has fixed the issue for a time, but the issue re-appears. I have also tried storage.erase_filesystem(), this had no better success than the re-install.  
This project involves two systems linked by RFM69 radios. So all troubleshooting includes 2  RP2040's. I have learned late in this troubleshooting that a USB device 'eject' is a good idea for each unplug.  I don't have extra RP2040's to try this out with - that is, doing eject each time from time zero to prevent this issue from coming back in the first place.

### Additional information

Windows 10 PC.
I have a really big doc to describe the project, however I don't think the issue is related to anything but the USB plug and un-plugs.  I'll attach the write all the same.
[Anti Parking Ticket System Description.docx](https://github.com/user-attachments/files/16418251/Anti.Parking.Ticket.System.Description.docx)

tannewt, 2024-07-30:
Does this still occur if you eject the device before unplugging?

There isn't much we can do from the CircuitPython side if you unplug without eject. Windows caches writes to the file system and unplugging prevents those from completing. Eject will write out those changes.

dhalbert, 2024-08-28:
Closing for now since we haven't heard back.

closed 2024-08-28 by dhalbert