adafruit/circuitpython #9477: Qualia S3 display is glitchy with 9.1, not glitchy with 9.0.5

BlitzCityDIY, 2024-07-31:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit-Qualia-S3-RGB666 with ESP32S3
```

### Code/REPL

```python
import time
import board
import displayio
from adafruit_qualia.graphics import Graphics, Displays

graphics = Graphics(Displays.ROUND40, default_bg=None, auto_refresh=False)

bitmap = displayio.OnDiskBitmap("/img.bmp")

# Create a TileGrid to hold the bitmap
tile_grid = displayio.TileGrid(bitmap, pixel_shader=bitmap.pixel_shader)

# Center the image
tile_grid.x -= (bitmap.width - graphics.display.width) // 2
tile_grid.y -= (bitmap.height - graphics.display.height) // 2

# Create a Group to hold the TileGrid
group = displayio.Group()

# Add the TileGrid to the Group
group.append(tile_grid)

# Add the Group to the Display
graphics.display.root_group = group

graphics.display.auto_refresh = True

# Loop forever so you can enjoy your image
while True:
    pass
```

### Behavior

I'm using the 4" display (https://www.adafruit.com/product/5793) with the Qualia S3. I'm getting some tearing/glitching on the display, usually vertically. The glitching is also present when the REPL is on the display. I installed 9.0.5 and the glitching did not occur.

mMerlin, 2024-07-31:
I am seeing the same issue with the SQUARE40_480X480 display and qualia board. Also 9.1.1. Have not (re)checked 9.0.5, but do not remember noticing problems when I was using it.
One side of the display twists up, then falls back down. One my display, that is the right side, but I have rotation set to 90, if that makes a difference. Whole display can shift up a couple of pixels too, but that is a lot less often.

mMerlin, 2024-08-04:
I reverted to 9.0.5 and confirm that no display glitches are occurring.

Before rolling back, I tested rotation changes. The physical side of the display that glitches the most stays constant with rotation changes. That is the side where the ribbon cable attaches.

tannewt, 2024-08-15:
I think #9526 also fixes this. It may be an IDF bug.

tyeth, 2024-08-16:
@tannewt  Just retested my qualia project on #9526
Still seeing the glitch described here, and the weird scanlines in #9498

tannewt, 2024-08-21:
After merging #9545 into main, this looks better to me. On restart it does glitch a little but it doesn't during the `while True:`.

Please retest after that fix propagates.

mMerlin, 2024-08-22:
Not sure how to tell if #9545 is in new 9.1.2 (uf2 downloaded half an hour ago), but still get the display glitches with that version.

BlitzCityDIY, 2024-08-22:
same, i just tested 9.1.2 the alpha 2351 and am still seeing the same glitches on the display

dhalbert, 2024-08-22:
#9545 and its equivalent are in both 9.1.2 and 9.2.0-alpha.2351.

FoamyGuy, 2024-08-22:
I have come across this as well on the Qualia S3. 

I tested both 9.1.2 and 9.2.0-alpha.2351 and both do have the glitchy screen pixel artifacts. 

I tested releases back to 9.0.5 and found it to be the most recent one that has a clean display for this project: https://learn.adafruit.com/1d-chomper-tabletop-arcade-game/overview

BlitzCityDIY, 2024-08-23:
that has been my experience as well, 9.0.x has no glitching

FoamyGuy, 2024-08-23:
In-case it can be useful, I put together a small reproducer that shows the glitchy qualia screen.
```
import time

import board
from adafruit_qualia import Qualia
from adafruit_qualia.graphics import Displays
import adafruit_imageload
import displayio

# Create the Qualia object for the 240x960 display PID 5799
qualia = Qualia(Displays.BAR240X960)

# OR create the Qualia object for the 320x960 display PID 5805
# qualia = Qualia(Displays.BAR320X960)

display = qualia.display

# Landscape orientation, mounted with ribbon cable coming off the left side of display.
display.rotation = 270

# Load the sprite sheet (bitmap)
sprite_sheet, palette = adafruit_imageload.load("1d_chomper_spritesheet.bmp",
                                                bitmap=displayio.Bitmap,
                                                palette=displayio.Palette)

# Create a sprite (tilegrid)
sprite_tg = displayio.TileGrid(sprite_sheet, pixel_shader=palette,
                               width=1,
                               height=1,
                               tile_width=15,
                               tile_height=15)

sprite_tg.y = 15

# Create a sprite (tilegrid)
second_sprite_tg = displayio.TileGrid(sprite_sheet, pixel_shader=palette,
                               width=1,
                               height=1,
                               tile_width=15,
                               tile_height=15)

second_sprite_tg.y = 15
second_sprite_tg.x = 120

background_tg = displayio.TileGrid(sprite_sheet, pixel_shader=palette,
                                   width=20, height=3,
                                   tile_width=15, tile_height=15)

for i in range(20):
    background_tg[i, 0] = 13
    background_tg[i, 1] = 4
    background_tg[i, 2] = 13

palette.make_transparent(0)

main_group = displayio.Group(scale=3)

animation_sprite_indexes = [0, 5]

sprite_tg[0] = animation_sprite_indexes[0]
second_sprite_tg[0] = animation_sprite_indexes[0]

main_group.append(background_tg)
main_group.append(sprite_tg)
main_group.append(second_sprite_tg)

display.root_group = main_group

cur_index = 0
last_animation_time = 0
animation_interval = 0.2

move_interval = 0.01
last_move_time = 0
# main loop
while True:
    now = time.monotonic()

    # if now > last_animation_time + animation_interval:
    #     sprite_tg[0] = animation_sprite_indexes[cur_index]
    #     second_sprite_tg[0] = animation_sprite_indexes[cur_index]
    #     cur_index = (cur_index + 1) % len(animation_sprite_indexes)
    #     last_animation_time = now

    if now > last_move_time + move_interval:
        sprite_tg.x = (sprite_tg.x + 1) % (display.width // 3)
        second_sprite_tg.x = (second_sprite_tg.x + 1) % (display.width // 3)
        last_move_time = now

        try:
            background_tg[sprite_tg.x // 15, 1] = 4
            background_tg[second_sprite_tg.x // 15, 1] = 17
        except IndexError:
            pass
```
It uses this BMP file: https://github.com/adafruit/Adafruit_Learning_System_Guides/raw/main/Qualia/Qualia_S3_1D_Chomper_Game/1d_chomper_spritesheet.bmp

Here is a recording running on CircuitPython 9.2.0 that shows the glitchy-ness on the left half of the screen, the blue bars are kind of shifting up and down over time. 

[glitchy_screen_9_2_0.webm](https://github.com/user-attachments/assets/c76b0597-38eb-4004-b6bb-8067cc9d23ae)

Here is a recording running on CircuitPython 9.0.5 that shows the same code running without the same glitchy-ness occurring

[nonglitchy_screen_9_0_5.webm](https://github.com/user-attachments/assets/7274fb50-8832-4b53-aa03-77d511e9b42d)

Worth noting that with these recordings and this reproducer script in general the glitchy-ness is more muted and les drastic than it is with the real 1D Chomper Game code from the learn guide. Here is a recording of the game code running on 9.2.0

[glitchy_chompergame_9_2_0.webm](https://github.com/user-attachments/assets/13af5959-be36-4c3d-b4bd-0412948ea5a0)

The same game code running on 9.0.5 renders smoothly without that pixel shifting on the blue bars. 

I'm not entire sure why the game has worse visual artifacts than the reproducer, perhaps it's worth trying to add some labels to the reproducer to see if that has an effect. I did test both with and without the "chomping" sprite animation, and it doesn't seem to make any difference, it's currently commented out in the main loop.

tannewt, 2024-08-27:
I think the core problem is that we've dropped the PSRAM speed from 120 MHz to 80 MHz. The IDF included new warnings about PSRAM stability at 120MHz.

I'll bump it up to 120MHz for this board only.

jepler, 2024-08-27:
Thanks @tannewt I had not figured out that change occurred

closed 2024-08-28 by jepler

mMerlin, 2024-08-30:
Verified that 9.1.3 works. No display content tearing.

PR #9564 (2024-08-28): Bump Qualia to 120 MHz flash and PSRAM speed
This allows us to push more pixels and reduce tearing.

Flash is bumped to 120 MHz too because there is no tuning to run PSRAM at 120 but flash at 80. It should be stable at 120 anyway.

Fixes #9477
commit f4a6460e (2024-08-28): Bump Qualia to 120 MHz flash and PSRAM speed

This allows us to push more pixels and reduce tearing.

Fixes #9477
commit a763343e (2024-08-28): Bump Qualia to 120 MHz flash and PSRAM speed

This allows us to push more pixels and reduce tearing.

Fixes #9477