adafruit/circuitpython #9485: Strange keypad behaviour with CP 9.1.0-beta.2

gbechtinger, 2024-08-07:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.0-beta.2 on 2024-05-15; Waveshare RP2040-Zero with rp2040
```

### Code/REPL

```python
import usb_hid
import keypad
import storage
import board
import supervisor
import analogio
from time import sleep

def range_map(x, in_min, in_max, out_min, out_max):
    return (x - in_min) * (out_max - out_min) // (in_max - in_min) + out_min
    

# This is only one example of a gamepad descriptor, and may not suit your needs.
GAMEPAD_REPORT_DESCRIPTOR = bytes((
    0x05, 0x01,  # Usage Page (Generic Desktop Ctrls)
    0x09, 0x05,  # Usage (Game Pad)
    0xA1, 0x01,  # Collection (Application)
    0x85, 0x04,  #   Report ID (4)
    0x05, 0x09,  #   Usage Page (Button)
    0x19, 0x01,  #   Usage Minimum (Button 1)
    0x29, 0x80,  #   Usage Maximum (Button 32, 0x20 / 0x60 for 96 / 0xA0 for 160)
    0x15, 0x00,  #   Logical Minimum (0)
    0x25, 0x01,  #   Logical Maximum (1)
    0x75, 0x01,  #   Report Size (1)
    0x95, 0x80,  #   Report Count (32)
    0x81, 0x02,  #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x05, 0x01,  #   Usage Page (Generic Desktop Ctrls)
    0x15, 0x00,  #   Logical Minimum (0)
    0x26, 0xFF, 0x0F,  #   Logical Maximum (4095)
    0x09, 0x37,  #   Usage (dial)
    0x75, 0x10,  #   Report Size (16)
    0x95, 0x01,  #   Report Count (1)
    0x81, 0x02,  #   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,        # End Collection
))

supervisor.set_usb_identification(manufacturer='me', product='cool device', vid=0x2E8A, pid=0x101F)
usb_hid.set_interface_name("Button Box 01")

gamepad = usb_hid.Device(
    report_descriptor=GAMEPAD_REPORT_DESCRIPTOR,
    usage_page=0x01,           # Generic Desktop Control
    usage=0x05,                # Gamepad
    report_ids=(4,),           # Descriptor uses report ID 4.
    in_report_lengths=(18,),    # This gamepad sends 6 bytes in its report.
    out_report_lengths=(0,),   # It does not receive any reports.
)

usb_hid.enable(
    (usb_hid.Device.KEYBOARD,
     usb_hid.Device.MOUSE,
     gamepad)
)

kpTemp = keypad.KeyMatrix(
    row_pins=(board.GP0, board.GP1, board.GP2),
    column_pins=(board.GP3, board.GP4, board.GP5,board.GP6),
)
    
print("Disabling mass storage on boot")
storage.disable_usb_drive()

kpTemp.reset()
sleep(0.1)
event = kpTemp.events.get()

if event:
    if event.pressed:
        if (event.key_number == 0):
            print("Enabling mass storage on boot")
            storage.enable_usb_drive()
```

### Behavior

this code should enable usb drive if the button connected to row 0 and colum 0 is pressed on boot but its not working as expected. It works fine with Adafruit CircuitPython 8.0.0-beta.0-13-gaa5f892a1-dirty on 2022-08-26; Waveshare RP2040-Zero with rp2040 but not with Adafruit CircuitPython 9.1.0-beta.2 on 2024-05-15; Waveshare RP2040-Zero with rp2040.

To detect the press in boot.py I need to change the code to that:

if event:
    if event.released:
        if (event.key_number == 1):
            print("Enabling mass storage on boot")
            storage.enable_usb_drive()
            
   So I need to detect the release of button 1 to check the press of button 0. If I read events three times, it shows 3 releases of button 0, 1 and 2 even if no button is pressed.

dhalbert, 2024-08-07:
Could you please re-verify with 9.1.1? I don't know if there's anything that would change this, but that helps us to make sure it hasn't already been fixed.

dhalbert, 2025-03-23:
I think that #9819 may have fixed this. I'm closing for now. Please reopen if it is not fixed.

closed 2025-03-23 by dhalbert