adafruit/circuitpython #9486: FunHouse flashes Yellow lights then nothing, won't boot after installing circuitpython v9.1.1

tyeth, 2024-08-07:
### CircuitPython version

```python
circuitpython v9.1.1 (en_GB UF2) and adafruit-circuitpython-adafruit_funhouse-en_GB-20240806-main-PR9479-7f92099.uf2
```

### Code/REPL

```python
none
```

### Behavior

 it gives me Yellow flashing dotstars at circuitpython boot then nothing. 
No circuitpython drive shows up and nothing on display. 

### Additional information

I've tried two funhouses, both erased first then bootloader 0.18.2 then UF2 of 9.1.1 en_GB 

TodBot in discord reported seeing the same yesterday, and found it works with v9.0.5 - They are betting that this change could be related https://github.com/adafruit/circuitpython/commit/2ddae3d96c56b94bc15ac0673fc4023720e37370 although I can't see how the rev-tft feather could not be affected while the funhouse is unless the problem is in the board specific library (pyportal-like library)

todbot, 2024-08-07:
From marching through the UF2s on S3, 
* v9.1.0-beta0 works 
* v9.1.0-beta1 does not work  
 
And by "not work" I mean the board doesn't even come up: doesn't show up as a USB device.  So probably not just the display change from 2ddae3d. I'm looking now through the [changes between 9.1.0-beta0 and 9.1.0-beta1](https://github.com/adafruit/circuitpython/compare/9.1.0-beta.0...9.1.0-beta.1) but wooo that's a lot of files

dhalbert, 2024-08-07:
I'd start by seeing if there are any board-specific changes.

todbot, 2024-08-07:
There did not appear to be any board-specific changes between 9.1.0-beta0 & beta1

tannewt, 2024-08-14:
Did you both load CircuitPython via esptool? Or the uf2s from circuitpython.org/downloads?

todbot, 2024-08-14:
> Did you both load CircuitPython via esptool? Or the uf2s from circuitpython.org/downloads?

I was using the UF2s, after I upgraded the TinyUF2 bootloader to latest (18.2 at the time)

tyeth, 2024-08-15:
UF2s here too.

tannewt, 2024-08-15:
@tyeth Are you on 18.2 as well?

tyeth, 2024-08-15:
Yeah looks like it is 0.18.2

tannewt, 2024-08-20:
@todbot and @tyeth do you have wifi credentials in settings.toml? Dan repro'd this using a manual `import wifi` after USB comes up instead of settings.toml.

I've spent all day debugging this with little progress. The USB interrupt fires, enqueues the event but the tinyusb task never runs to pull from the queue. This happens after `esp_wifi_start()` is called.

todbot, 2024-08-21:
> @todbot and @tyeth do you have wifi credentials in settings.toml? 

In my test two weeks ago, I did not have a settings.toml file. Nor did I test WiFi, just if the board came up on USB.  I just now created a settings.toml on the FunHouse with that firmware, giving a proper `CIRCUITPY_WIFI_SSID` &`CIRCUITPY_WIFI_PASSWORD` and the device appeared to immediately reset and disappear off USB. The FunHouse display does come up and says "Wi-Fi: No Ip" but is otherwise a brick.

PR #9526 (2024-08-16): Update to IDF 5.3
The Funhouse problem was due to a locking change for SPI. The Funhouse is one of a few boards with DotStar status LEDs that use SPI and have this bug.

I thought it might be IDF related so I updated it first. :-)

This was meant to fix #9486, but it doesn't however, it has other useful fixes.
commit cedd6779 (2024-08-16): Fix Funhouse and update to IDF 5.3

The Funhouse problem was due to a locking change for SPI. The
Funhouse is one of a few boards with DotStar status LEDs that use
SPI and have this bug.

I thought it might be IDF related so I updated it first. :-)

Fixes #9486
PR #9545 (2024-08-21): Fix FreeRTOS weirdness on Funhouse
Funhouse has DotStars attached to SPI that exercises this bug. Most boards are NeoPixel and avoid this.

Fixes #9486
commit 07452b21 (2024-08-21): Fix FreeRTOS weirdness on Funhouse

Funhouse has DotStars attached to SPI that exercises this bug.
Most boards are NeoPixel and avoid this.

Fixes #9486
closed 2024-08-21 by dhalbert