adafruit/circuitpython #9491: Low power support for RP2350

tannewt, 2024-08-08:
The SDK doesn't have an official API for it yet. So, we can wait or DIY it.

tannewt, 2024-09-10:
Reference code is linked to (and discussed) in this issue: https://github.com/raspberrypi/pico-examples/issues/530

dmcomm, 2025-05-07:
I'm trying to run an RP2040 program that uses `sleep_memory` to pass data between resets and between boot.py and main program, but doesn't actually use any of the low-power features. Not sure if this is a common situation or not.

bablokb, 2025-05-20:
@eightycc: since you self-assigned this, please have a look at my PR. It works fine on all of my systems. And in contrast to the discussion, it does not "change semantics" (except for a single academic border case which is not relevant for any practical real-life scenario).

It *does* change the documentation bringing it in line with reality. The docs currently claims things that the code does not stand up to and never will ("It does not shut down WiFi, BLE, or other communications, or ongoing activities such as audio playback").

eightycc, 2025-05-20:
Thank you, @bablokb! I'll be looking closely at your work on #9965 and #10023.

bablokb, 2025-06-04:
Some more notes:

- powman resets directly after wakeup, so you cannot set the wakeup cause
- On the RP2350, the AON-timer uses the powman-timer internally.
- as a user, I would want the choice between power-savings and timer accuracy. If I need an accurate timer, I use an external RTC and GPIO-wakeup anyhow. LPOSC accuracy can be tested and taken into account for. I have specimen with 1%, 8% and 25% differences. These are not constants and depend e.g. on temperature, but there are use-cases where that is good enough. As an example, I have sensors that need about 30s warmup time. There is no need to accurately "light-sleep" the RP2350 at high currents during this time.
- I have experimental code that uses powman-states for deep-sleep. Enter and exit works fine for timers, but I have problems with the reliability of GPIOs. With a timer, I have something like 600µA, with a GPIO I have 150µA. This is all very preliminary.

eightycc, 2025-06-04:
> powman resets directly after wakeup, so you cannot set the wakeup cause

The wake-up cause can be determined by examining POWMAN register LAST_SWCORE_PWRUP after reset. My implementation leaves a special value in SCRATCH0 to indicate this case, although CHIP_RESET with HAD_SWCORE_PD set likely tells the same story.

> On the RP2350, the AON-timer uses the powman-timer internally.

As far as I can tell, the AON timer _is_ the POWMAN timer.

> I have experimental code that uses powman-states for deep-sleep.

Would you be willing to share your code? I'm doing quite a bit of stumbling around in the dark with P1 states. Which P1 state are you entering for those impressive power numbers?

The LPOSC is crazy inaccurate and as you note extremely temperature sensitive. I've seen noted on the RPi discussion board that any attempt to calibrate it for use in low-power states is problematic due to the change in die temperature between a P0 and a P1 state.

bablokb, 2025-06-05:
> As far as I can tell, the AON timer _is_ the POWMAN timer.

As far as I understand the SDK, this is only true for the RP2350.

> Would you be willing to share your code? I'm doing quite a bit of stumbling around in the dark with P1 states. Which P1 state are you entering for those impressive power numbers?

It is still too experimental but I will push my powman-tree once it is cleaned up and a bit more tested. But I mainly follow the examples from https://github.com/peterharperuk/pico-examples/tree/powman/powman. This code uses 

```
// Power states
    powman_power_state P1_7 = POWMAN_POWER_STATE_NONE;

    powman_power_state P0_3 = POWMAN_POWER_STATE_NONE;
    P0_3 = powman_power_state_with_domain_on(P0_3, POWMAN_POWER_DOMAIN_SWITCHED_CORE);
    P0_3 = powman_power_state_with_domain_on(P0_3, POWMAN_POWER_DOMAIN_XIP_CACHE);

    off_state = P1_7;
    on_state = P0_3;
```

> ... is problematic due to the change in die temperature between a P0 and a P1 state.

I am not that pessimistic, since once I am in P0 state, I am in P0 state. In all of my tests, the inaccuracy was rather deterministic for a given specimen. Nevertheless, I would not vote for the LPOSC as the default, but I would strongly support to have the option to use it. (if it is worth it - I haven't measured the dormant-state using XOSC yet).

I am currently fighting with GPIO wakeup. It is really strange, only about one out of three times the pico will enter the off-state. It seems like there is a glitch that triggers wake up already a fraction of a second after the device powered down.  And I have my GPIO hardwired to 3V3 so I cannot really understand why.

eightycc, 2025-06-05:
> I am currently fighting with GPIO wakeup. It is really strange, only about one out of three times the pico will enter the off-state. It seems like there is a glitch that triggers wake up already a [...]

Some of the power manager registers do not reset on a power manager triggered reset. Calling `powman_disable_all_wakeups()` before setting up new alarm conditions for entry to a low-power state will ensure you don't exit again for a previously triggered alarm condition. It's not in the example code, which is why I mention it.

bablokb, 2025-06-06:
I pushed my working tree: https://github.com/bablokb/circuitpython/tree/rp2xxx_powman

Note that this is based on 9.2.8.

I am also adding an image from ppk2 that demonstrates the behavior I described above. In this test, the system needed four attempts to enter the power-off state, but that count is random.

[image]

bablokb, 2025-08-04:
`LAST_SWCORE_PWRUP` does not seem reliable even without any debugprobe attached. It always reads `2` (which is pwrup1 according to the docs), but I am only testing with a single GPIO and I am using pwrup0 which should turn up as `1`.

adeldigital, 2025-10-24:
@bablokb Thanks for sharing the PPK2 screenshots, they are very useful. Could you confirm if I understood correctly, you got 154µA@5V during sleep with SRAM retention? Is that in the dormant mode?

dwrrehman, 2025-10-24:
> [@bablokb](https://github.com/bablokb) Oh, okay. So 1.9mA is the absolute lowest power sleep state the RP2350 can reach with SRAM retention? Shoot, I need my system way lower than that. Looks like I need to look for a different microcontroller. Thanks

Actually ~210uA is the lowest the RP2350 can go with 256kB of SRAM retained (SRAM power domain 1, specifically), specifically using low power mode "P1.6" as i mentioned above. Like Bablokb mentioned, this does involve a CPU reset (as it is being power-gated in P1.x modes), however SRAM contents are not reset on a CPU reset of this kind, IIRC.

(note, this is using low power mode P1.6, ie not the RP2350's "Sleep" mode, nor it's "Dormant" mode. Sleep and Dormant modes are distinct modes with somewhat low power consumption. but these do not power-gate anything in the chip, they simply gate the clocks to the cpu, or other simple power saving methods like that).

dwrrehman, 2025-10-24:
> There has been some work done on this, e.g. here: https://github.com/mamba2410/rp2350-powman-sleep, see also https://forums.raspberrypi.com/viewtopic.php?t=384176#p2297511
> 
> But this does not sound as if it were really usable ("I managed to get this working, mostly!").
> 
> Anyhow, before investigating this, I will first try to make the system enter sleep correctly for gpio-wakeups.

Thanks for linking the page! This is fascinating work, I was not aware this already existed! Hopefully eventually a polished solution/API for a low power mechanism like this comes together at some point. I know it sure would be quite useful in many battery-powered RP2350 applications!

Best of luck in your gpio-wakeup testing! feel free to reach out for any help or questions, i'd be happy to help in any way i can.

bablokb, 2025-10-24:
You can store it in nvram and then you could even turn of power.

Be aware that all of this discussed here is very experimental, we don't even have official support for the alarm-module for the RP2350. Your out-of-the-box experience will be better with an ESP32-S3 or a Nordic mcu.

dwrrehman, 2025-10-24:
> [@dwrrehman](https://github.com/dwrrehman) 188 µA sounds much better. In my case I don’t need the CPU to continue where it left off. I just need to keep an API key in SRAM retained. If I don’t need the CPU to resume its previous state, could allowing it to start from reset (or something along those lines) reduce the power consumption even further?

It seems allowing the CPU to start from reset will give you equivalent low power consumption, 188uA, (low power mode P1.5/P1.6, where 1 bank of SRAM retention is enabled) as adding this (currently hypothetical) mechanism to let the CPU pick up where it left off in execution, from my understanding. 

However, as bablokb mentioned, for your use case its entirely possible you could be able to store the API key in some form of non-volatile memory (NVRAM), such as an FRAM chip, MRAM chip, BBSRAM, etc. Technically using the Flash could work, however only if this data does not need to be written often. Once the data is written to some non-volatile memory, you can simply unpower the RP2350 entirely using an external circuit instead of using the sleep modes.

bablokb, 2025-10-27:
@Skorpionm Thanks for sharing! Do you also have the complete program somewhere?

dwrrehman, 2025-11-03:
Hi, @bablokb,

Thanks for the question! :) I'll see if I can illuminate whats happening a bit more, if you'd like!

Below I've included the implementation of `powman_enable_gpio_wakeup` for our reference.

```c
254 | void powman_enable_gpio_wakeup(uint gpio_wakeup_num, uint32_t gpio, bool edge, bool high) {
255 |     invalid_params_if(HARDWARE_POWMAN, gpio_wakeup_num >= count_of(powman_hw->pwrup));
256 | 
257 |     // Need to make sure pad is input enabled
258 |     gpio_set_input_enabled(gpio, true);
259 | 
260 |     // Set up gpio hardware for what we want
261 |     uint32_t pwrup = (edge ? POWMAN_PWRUP0_MODE_VALUE_EDGE : POWMAN_PWRUP0_MODE_VALUE_LEVEL) << POWMAN_PWRUP0_MODE_LSB;
262 |     pwrup |= (high ? POWMAN_PWRUP0_DIRECTION_BITS : 0);
263 |     pwrup |= gpio << POWMAN_PWRUP0_SOURCE_LSB;
264 |     powman_write(&powman_hw->pwrup[gpio_wakeup_num], pwrup);
265 | 
266 |     // Clear the status bit in case an edge is already latched
267 |     powman_clear_bits(&powman_hw->pwrup[gpio_wakeup_num], POWMAN_PWRUP0_STATUS_BITS);
268 | 
269 |     // Important to enable it separately to allow the gpio to change
270 |     powman_set_bits(&powman_hw->pwrup[gpio_wakeup_num], POWMAN_PWRUP0_ENABLE_BITS);
271 | }
```

Of important note here is that on line `267` we use the value `POWMAN_PWRUP0_STATUS_BITS`, which has the value `0x00000200`:

```c
#define POWMAN_PWRUP0_STATUS_BITS   _u(0x00000200)
```

which equates to having a `1` bit set at bit position `9`, which the datasheet says is the `STATUS` bit/field. Digging deeper, we see that `powman_clear_bits` has the following definition:

```c
124 | static inline void powman_clear_bits(volatile uint32_t *reg, uint32_t bits) {
125 |     invalid_params_if(HARDWARE_POWMAN, bits >> 16);
126 |     hw_clear_bits(reg, POWMAN_PASSWORD_BITS | bits);
127 | }
```
...And diving even further, to see what actual underlying C code will be used, we see:

```c

112 | #define hw_clear_alias_untyped(addr) ((void *)(REG_ALIAS_CLR_BITS + hw_alias_check_addr(addr)))
...	...
145 | __force_inline static void hw_clear_bits(io_rw_32 *addr, uint32_t mask) {
146 |     *(io_rw_32 *) hw_clear_alias_untyped((volatile void *) addr) = mask;
147 | }
```

...Which uses the following definition, found in `addressmap.h`:

```c
21 | #define REG_ALIAS_CLR_BITS (_u(0x3) << _u(12))
```

it thus seems that `REG_ALIAS_CLR_BITS` effectively has the value `0x3000`, and we are effectively OR'ing it with the `PWRUP0` register's address, to obtain the actual address we write to, and the actual value we write is the bits _set_ to `1`. Looks like what is happening here, is that the SDK opts to (confusingly) use the register address aliases, which are outlined in the RP2350 datasheet in section `2.1.3.`, titled `Atomic register access`. if `PWRUP0` has address `A`, then we are writing to:

```c
*(A | 0x3000) = 0x00000200;
```

However, this bit we are writing to here, is of the register type "`WC`", meaning "Cleared-when-writing-a-1". This is because "`WC`" appears next to the bitfield name in `PWRUP0`'s register documentation, and `Appendix A: Register field types` of the RP2350 datasheet defines `WC` as the following:

```
WC: 
  - Write-Clear
  - Writing a 1 to a bit in a WC field will write that bit to 0
  - Writing a 0 to a bit in a WC field does nothing
  - Read operation returns the register value
```

And although I can't find this exact page which talks about the combination of these two semantics, i think its clear (pardon the pun) that using the "Clear-on-write" register address aliases on a `WC` bitfield yields the same (somewhat redundant) semantics, of writes of a 1 cause the field to clear to be a 0. This register bitfield's value is always set to `1` by hardware (software can only clear it), and given that fact, reads work as you would expect, I think. Connecting this to what I mentioned earlier, it seems that you can only enable a GPIO wakeup once this bit is clear (ie would read as a 0, if read). 

Finally, I have been thinking about what might be the cause for your spurious wakeup events being triggered directly after going to sleep, and I cannot quite think of what might be triggering that. On the bright side, i have a hunch that it might be due to interactions of some other piece of code with the POWMAN system, for example, generated C code / CPU instructions involved in controlling something else on the RP2350 other than POWMAN's low power mode interface itself, and this seperate system is somehow sending events, IRQs, or something else which is causing the CPU to wake up. 

I hypothesize this mainly because when I use my RISC-V assembler and manually program the direct register writes I want to happen in sequence, and flash the UF2 generated by the assembler to my Pico 2 boards, the program works flawlessly with no spurious wakeups whatsoever (at least, none that i have been able to detect, anyways). This leads me to conclude that if there is no other conflicting interactions happening on the system, and one has the right sequence of register writes to the POWMAN subsystem, low power modes, and GPIO wakeups / ALARM wakeups work flawlessly. It's possible I could be wrong about this hypothesis, however. 

Anyways, I hope this has helped slightly in some way, and made things a tiny bit more clear. Let me know if anything didn't quite make sense, and I'd be happy to explain it in further detail, and/or look deeper at the code you are using to perform these low power modes. :)

bablokb, 2025-11-03:
Thanks, this is very helpful and I really appreciate that you  take the time to explain these details.

> On the bright side, i have a hunch that it might be due to interactions of some other piece of code with the POWMAN system, for example, generated C code / CPU instructions involved in controlling [...]

Yes, that is exactly what I think too. CircuitPython is a complex program and I am trying to hunt down what is happening. The problem is that "something else" is not showing up in LAST_SWCORE_PWRUP. This register is always the same, regardless of a random wakeup or a wakeup I actually triggered by toggling the wakeup GPIO.

bablokb, 2025-11-03:
Yes, the external debugger seems to cause problems, see the comment above by @eightycc https://github.com/adafruit/circuitpython/issues/9491#issuecomment-2952711328. I am not using it though. Regarding the other suggestions: I will put them on my list and test them.

bablokb, 2025-11-03:
I think I solved the problem. I added some delays (mainly to give the system time to flush out trace-messages). CP has two APIs for that: `mp_hal_delay_ms()` and `mp_hal_delay_us()`. The former runs additional code, the latter is just a busy-wait. After switching to the latter I don't see the random behavior anymore.

There is still some way to go to make this usable in the context of CP, but at least the basics are working now.

dhalbert, 2026-10-30:
Fixed by #11434.

dhalbert, 2026-11-02:
Claude wrote this:

Reopening: on a Metro RP2350, a `PinAlarm` does not wake the board from real deep sleep.

Tested with `main` at 9e926ddae8 (11.0.0-alpha.2-59), and with #11723:

- `code.py` calls `alarm.exit_and_deep_sleep_until_alarms()` with `PinAlarm(board.D5, value=False, pull=True)`. A0 (GPIO41) was also tried. Each boot logs the reset reason, `supervisor.runtime.usb_connected` and `alarm.wake_alarm` to `microcontroller.nvm`.
- The board was booted with no USB host (a charge-only cable, power-cycled), left for over 10 seconds so it entered real deep sleep, and then D5 was grounded.
- The board does not wake: no boot is logged until USB is reconnected.
- With USB connected (fake deep sleep), the same alarms wake it, and `alarm.wake_alarm.pin` is correct.
- The same procedure does wake a Feather RP2040 from real deep sleep (see #11721).

Not yet investigated. A `TimeAlarm` wake from real deep sleep has not been tried.

closed 2026-11-03 by dhalbert