adafruit/circuitpython #9496: Calling micrcontroller.reset on the Qualia (with 3.2" bar 820x320) causes reset into TinyUF2 Bootloader mode

tyeth, 2024-08-09:
### CircuitPython version

```python
9.1.1
```

### Code/REPL

```python
Adafruit CircuitPython 9.1.1 on 2024-07-22; Adafruit-Qualia-S3-RGB666 with ESP32S3
>>> 
>>> import microcontroller
>>> microcontroller.reset()
```

### Behavior

TFTS3BOOT drive appears instead of CIRCUITPY

bill88t, 2024-08-12:
Oh I know this. It's the double reset setting in tinyuf2.

```
// GPIO that implement 1-bit memory with RC components which hold the
// pin value long enough for double reset detection.
#define PIN_DOUBLE_RESET_RC   4
```

Furthermore, it seems the latest commit there changed that from IO5 to IO4.
https://github.com/adafruit/tinyuf2/commit/da3eaf9949ef33e484106115b16b849034559664

Defining this without the proper circuit leads to this exact behavior.
Suggestion: Transfer this issue to the TinyUF2 repo.

tyeth, 2024-08-12:
Curiously I'm pretty sure I've been running firmware since that change, but to be fair I've no idea the actual last version I ran before 9.x

bill88t, 2024-08-12:
> Curiously I'm pretty sure I've been running firmware since that change, but to be fair I've no idea the actual last version I ran before 9.x

@tyeth The CircuitPython version has nothing to do with this. It's just the bug doesn't happen every time (it's a 80% from what I can tell).

CircuitPython resets correctly, but then the TinyUF2 bootloader performs the double reset check and (in most cases) succeeds, sending you to TinyUF2 instead of CircuitPython.

The TinyUF2 bootloader is also an updatable piece of firmware, you can view it's version from the info txt file in the TFTS3BOOT. (Maybe a wrong TinyUF2 version is installed? You might wanna post it.)

tyeth, 2024-08-12:
Ahhh yeah good points, morning brain wasn't thinking bootloaders.
Not sure which versions I'm running on each, but probably 0.18.2 (definitely on this problematic one), and half aware that the bootloaders on circuitpython.org/downloads are often older than latest tinyuf2 releases (most are on 0.18.2).

My INFO_UF2.txt:
```
TinyUF2 Bootloader 0.18.2 - tinyusb (0.15.0-566-gf1e006d09)
Model: Adafruit Qualia ESP32-S3 RGB666
Board-ID: ESP32S3-Qualia-revC
Date: Mar 22 2024
Flash Size: 0x00200000 bytes
```

bill88t, 2024-08-12:
The TinyUF2 version is recent and correct from the looks of it, so there isn't anything I can do to help you on from here.
I do not have the board to physically test and evaluate against the schematic.

tannewt, 2024-08-21:
I'm unable to reproduce this. I can do `microcontroller.reset()` just fine. Does it happen with the latest main for you?

tannewt, 2024-08-29:
@tyeth Please retest this.

hathach, 2024-09-01:
Maybe this would help, https://github.com/adafruit/tinyuf2/pull/406/files check the reset reason and only check for double reset for EN/nRST pin

dhalbert, 2024-09-17:
I retested on 9.2.0-beta.0, and could not reproduce this with the Qualia. I tried both with and without a `settings.toml` that connected to the local wifi.

tyeth, 2024-09-19:
thanks all. running 9.2.0-beta.0, but cancelling the running example and calling microcontroller.reset caused the tfts3boot drive to show (still running tinyuf2 0.18.2 bootloader).
 Updating bootloader to 0.20.1 allows the microcontroller.reset to reliably work fine and show the circuitpy drive.