adafruit/circuitpython #9498: Qualia / DotClockFrameBuffer unexpectedly renders a few lines of pixels at once in the most bizarre way

tyeth, 2024-08-09:
### CircuitPython version

```python
9.1.1
```

### Code/REPL

```python
Adapted version of Liz's countdown clock for CircuitPython Day 2024, bitmap background + bitmap label that scrolls
```

### Behavior

Image renders about 5 lines at a time, but seemingly randomly out of order and not the whole line at once.
Probably made worse by my use of `display.rotation=90`

tyeth, 2024-08-09:
It's also incredibly slow once the bitmap label starts being scrolled (but I'd possibly expect/accept that)

Code:
[G.zip](https://github.com/user-attachments/files/16564691/G.zip)

Video:
 https://photos.app.goo.gl/DLvyEcP4XDUtz9xN8 
Still image part way through initial render:
[image]

tyeth, 2024-08-09:
Just realised I'm not using the color converter for the main tilegrid that the default example uses, like this:
```
tile_grid = displayio.TileGrid(bitmap, pixel_shader=displayio.ColorConverter(input_colorspace=displayio.Colorspace.RGB565))
```

somenice, 2024-08-17:
I experienced really slow updates on Qualia w 2.1 round display when using any display rotation. 
Even calling rotation=0 seemed to cause slow loading if I recall.

tannewt, 2024-08-19:
> I experienced really slow updates on Qualia w 2.1 round display when using any display rotation. Even calling rotation=0 seemed to cause slow loading if I recall.

Is this new? Loading from flash into PSRAM is going to be limited by the speed of their shared memory bus.

Rotation can be bad for loading since it may cause skipping through the file system to go column by column instead of row by row which wouldn't. This isn't new and I'd suggest rotating the asset to align with the display's rotation.