adafruit/circuitpython #9501: Consider using TRNG for RP2350

deshipu, 2024-08-09:
The RP2350 has a new peripheral for a true random generator, it could be nice to make use of it.

jepler, 2024-08-10:
Based on a quick perusal of the 2350 datasheet the 2040 compatible "rosc randombit" generator should work with the new chip but using the new hw module mother be faster or something.

jepler, 2024-08-10:
Better yet switch to pico rand from the sdk and delete my "crypto" code from circuitpthon. It will select the right hw mode for generating random bytes

jepler, 2024-08-28:
I would probably drop this from the 9.2.0 milestone. While I don't actually have any RP2350 hardware for testing, I believe the ROSC based RNG is currently working on both chips, so this is not urgent missing functionality.