adafruit/circuitpython #9503: File Copy Error and Corruption in CircuitPython Device (multisparkline.py, polygon.py, rect.py, roundrect.py, sparkline.py, triangle.py)

Axlfc, 2024-08-10:
Similar issue [here](https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/issues/93).

**Description:**

When attempting to copy the files `multisparkline.py`, `polygon.py`, `rect.py`, `roundrect.py`, `sparkline.py`, `triangle.py`  into my CircuitPython device (drive E: named CIRCUITPY), the file becomes corrupted with strange characters, and the following error message appears:

```
No puede copiar el archivo debido a un error inesperado. Si sigue recibiendo este error, puede usar el código de error para buscar ayuda para este problema.

Error 0x800701 B1: Se ha especificado un dispositivo que no existe.
```

**Steps to Reproduce:**

1. Connect the CircuitPython device to the computer.
2. Copy the files `multisparkline.py`, `polygon.py`, `rect.py`, `roundrect.py`, `sparkline.py`, `triangle.py` from the source directory to the CIRCUITPY drive at directory `E:\lib\adafruit_display_shapes`.
3. Observe the error message during the copy process.
4. After copying, open any of the folloring files `multisparkline.py`, `polygon.py`, `rect.py`, `roundrect.py`, `sparkline.py`, `triangle.py` on the CIRCUITPY drive.

**Actual Behavior:**

- Error message: `No puede copiar el archivo debido a un error inesperado. Si sigue recibiendo este error, puede usar el código de error para buscar ayuda para este problema. Error 0x800701 B1: Se ha especificado un dispositivo que no existe.`
- The resulting file on the CIRCUITPY drive contains strange characters and appears to be corrupted with NUL and other unexpected characters.

**Additional Information:**

- **Device Model:** M5Stack Cardputer
- **CircuitPython Version:** 9.1.0-beta.3 on 2024-05-22
- **Operating System:** Windows 10

**Content on the resultant files:**
multisparkline.py:
```
# SPDX-FileCopyrightText: 2020 Kevin Matocha
#
# SPDX-License-Identifier: MIT

"""
`multisparkline`
================================================================================

Various common shapes for use with displayio - Multiple Sparklines on one chart!

* Author(s): Kevin Matocha, Maciej Sokolowski

Implementation Notes
--------------------

**Software and Dependencies:**

* Adafruit CircuitPython firmware for the supported boards:
  https://github.com/adafruit/circuitpython/releases

"""

try:
    from typing import Optional, List, TypeVar

    T = TypeVar("T")
except ImportError:
    pass
import displayio
from adafruit_display_shapes.polygon import Polygon

class _CyclicBuffer:
    def __init__(self, size: int, init_value: T) -> None:
        self._buffer = [init_value] * size
        self._start = 0  # between 0 and size-1
        self._end = 0  # between 0 and 2*size-1

[... 654 lines cut ...]
        self,
        x0: int,
        y0: int,
        x1: int,
        y1: int,
        x2: int,
        y2: int,
    ) -> None:
        if y0 == y2:  # Handle awkward all-on-same-line case as its own thing
            a = x0
            b = x0
            if x1 < a:
                a = x1
            elif x1 > b:
                b = x1

            if x2 < a:
                a = x2
            elif x2 > b:
                b = x2
            self._line(a, y0, b, y0, self._FILL)
            é?À{7@î   	    ÿÿ      <ˆ  2TÍ«            9.1.0-beta.3                    circuitpython                   20:28:06        May 22 2024     da9dc379                        ¶T¦1˜„5l"‹~©NSÌq$~G8ŸY›™0D‡                                                                                __dir__ *   <module>    __call__    __class__   __delitem__ __enter__   __exit__    __getattr__ __getitem__ __hash__    __init__    __int__ __iter__    __len__ __main__    __module__  __name__    __new__ __next__    __qualname__    __repr__    __setitem__ __str__ ArithmeticError AssertionError  AttributeError  BaseException   EOFError    Ellipsis    GeneratorExit   ImportError IndentationError    IndexError  KeyError    KeyboardInterrupt   LookupError MemoryError NameError   NoneType    NotImplementedError OSError OverflowError   RuntimeError    StopIteration   SyntaxError SystemExit  TypeError   ValueError  ZeroDivisionError   abs any append  args    bool    builtins    bytearray   bytecode    callable    chr classmethod clear   close   const   copy    dict    divmod  endswith    eval    exec    extend  find    format  from_bytes  get getattr globals hasattr hash    id  insert  isalpha isdigit isinstance  islower isspace issubclass  isupper join    key len little  
```

Best regards

dhalbert, 2024-08-10:
Transferred issue to https://github.com/adafruit/circuitpython, because this issue is not related to the particular files.

There are various reasons the copies might be corrupted.
1. How are you copying the files? Drag and drop?
2. The CIRCUITPY drive may be corrupted due to previously unplugging or resetting the device before a write was complete. Try this in the REPL serial connection. It will erase and reformat CIRCUITPY. Then copy the files agan.
```py
import storage
storage.erase_filesystem()
```
3. The flash chip on the Cardputer may be defective. This is quite rare.
4. The original files may be corrupted. I'm guessing you looked for this already?
5. There my be utility programs (antivirus or disk utility, etc.) that are interfering with the copy.

Axlfc, 2024-08-10:
Hello and so many thanks for the quick response.

>     1. How are you copying the files? Drag and drop?

Yes, I download manually the source code and copy/paste to the board's folder, I went file by file because I noticed when copying the whole folder the behavior happens too.

>     2. The CIRCUITPY drive may be corrupted due to previously unplugging or resetting the device before a write was complete. Try this in the REPL serial connection. It will erase and reformat CIRCUITPY. Then copy the files agan.
> ```python
> import storage
> storage.erase_filesystem()
> ```

I tried this line, as well as installing the circuitpython .bin file... I used the [bmorcelli's M5Stick-Launcher project](https://github.com/bmorcelli/M5Stick-Launcher) and downloaded the .bin file from [CircuitPython M5Stack Cardputer](https://circuitpython.org/board/m5stack_cardputer/)

>     3. The flash chip on the Cardputer may be defective. This is quite rare.

No idea, right now I'm currently trying to do this again using another USB port from my computer. It seems that using a non blue USB port it let me copy all the files without the error. I may have my blue USB port damaged.

>     4. The original files may be corrupted. I'm guessing you looked for this already?

Yes, the original files are okay.

>     5. There my be utility programs (antivirus or disk utility, etc.) that are interfering with the copy.

I tried to blocking antivirus and still hapening.

It seems that using another usb port from my computer that is not blue let me copy the files successfully and without any errors, I guess I solved the issue myself this way.

Axlfc, 2024-08-10:
It still happening with `switch_round.py` 🤐

Axlfc, 2024-08-10:
This is how the unit looks when I try to copy the files:
[image]

After a short time (or pressing Enter in the Cardputer) it mounts.
[image]

dhalbert, 2024-08-10:
Are you using a hub or a direct connection?

Please try a different cable as well.

Could you try CircuitPython 9.0.5 and see if it makes any difference?

Axlfc, 2024-08-10:
> Are you using a hub or a direct connection?

Direct connection to the pc.

> Please try a different cable as well.

I have another cable and it still happening...

> Could you try CircuitPython 9.0.5 and see if it makes any difference?

Sure I will try again, I'm resetting using the command you gave me earlier and try the whole process again and write it here in few minutes.

```
import storage
storage.erase_filesystem()
```

Axlfc, 2024-08-10:
Using CircuitPython 9.0.5:

[image]

This files are copied without problems.

Now I try to copy the mentioned files.

[image]

This has given no errors so far, now I go with the folder on my next step:

[image]

ALL FILES COPIED SUCCESSFULLY USING CircuitPython 9.0.5

Axlfc, 2024-08-10:
I also have been messing putting a git repository in this folder... That may have affected this in some way... after i copied a .git folder into the file system the error appeared again...

dhalbert, 2024-08-10:
So 9.0.5 is not better when the .git tree is added?

If you have another computer to try that's rather different (e.g. Mac, RPi, etc.) that would be a useful experiment.

Axlfc, 2024-08-10:
Sorry I do not have a Mac nor RapsberryPi, but I can try tomorrow to replicate whole file copying and file editing process in my dual boot ubuntu GNU/Linux (it's the same pc I run Windows 10) and see what happens.

---

It blocked again like this
[image]
trying to edit the `code.py` file...

😰

It's only happening randomly at some times... But fortunately the last changes I made it worked.

But for now it's accomplished copying the library files inside the filesystem and they are not showing corrupt.

Well I guess I won't be using .git folder in the Cardputer running CircuitPython 9.0.5 nor the beta version, I will keep running this version for now, but we accomplished to copy the libraries, **very much a success**. 😁

Thank you very much for your help!!

dhalbert, 2024-09-17:
This sounds like it might be #9472. I am closing this in favor of that, but will add additional details as needed. Thanks!

closed 2024-09-17 by dhalbert

dhalbert, 2024-09-17:
We think this may be fixed in 9.1.4 and 9.2.0-beta.0. The reporter of #9472 was unable to reproduce with 9.1.4. Try again if you can with a newer version.

dapsvi, 2025-03-08:
Not fixed, I have the exact same issue when trying to move some files from windows to circuitpython. I have tested with the version 9.1.4 and 9.2.4. I also tried to move the files from my SD card to the filesystem (I am trying to put [Beryllium]() on my cardputer) using this code for `code.py`, yet it stops working at the very same moment as when doing it directly from my computer, that is, when copying `/Beryllium/lib/ftp_server.mpy` :
```py

# SPDX-FileCopyrightText: 2017 Limor Fried for Adafruit Industries
#
# SPDX-License-Identifier: MIT

import os
import sdcardio
import board
import busio
import digitalio
import storage

storage.remount("/", readonly=False)
# Use any pin that is not taken by SPI
cs = board.SD_CS

# Set up SPI and mount the SD card filesystem at /sd
spi = busio.SPI(board.SD_SCK, board.SD_MOSI, board.SD_MISO)
sdcard = sdcardio.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

# -----------------------------------------------------------------------------
# Helper function: ensure_parent_dir
#
# Given a file path, this function makes sure that the parent directory
# exists by recursively creating any missing directories.
# For example, if file_path is "/subdir1/subdir2/file.txt", it ensures that
# both "/subdir1" and "/subdir1/subdir2" exist.
# -----------------------------------------------------------------------------
def ensure_parent_dir(file_path):
    # Get the parent directory (everything before the last slash)
    parent = "/".join(file_path.split("/")[:-1])
    if parent == "" or parent == "/":
        return  # The root always exists.
    try:
        os.stat(parent)
    except OSError:
        # Parent doesn't exist. Recursively create its parent first.
        ensure_parent_dir(parent)
        os.mkdir(parent)

# -----------------------------------------------------------------------------
# Helper function: copy_file
#
# This function copies a single file from 'src' to 'dst' in 512-byte chunks.
# It first ensures that the destination's parent directory exists.
# -----------------------------------------------------------------------------
def copy_file(src, dst):
    ensure_parent_dir(dst)
    with open(src, "rb") as in_file:
        with open(dst, "wb") as out_file:
            while True:
                buf = in_file.read(512)  # Read in 512-byte chunks.
                if not buf:
                    break
                out_file.write(buf)

# -----------------------------------------------------------------------------
# Helper function: copy_recursive
#
# This function checks whether the current source path is a directory.
# - If it is a directory, it makes sure the destination directory exists
#   (creating it if necessary) and then recursively copies all its contents.
# - If it is a file, it calls copy_file to perform the copy.
#
# Note: We use a bitmask (0x4000) on os.stat(src)[0] to check for directories.
# -----------------------------------------------------------------------------
def copy_recursive(src, dst):
    st = os.stat(src)
    if st[0] & 0x4000:
        # Source is a directory.
        try:
            os.mkdir(dst)
        except OSError:
            # The destination directory might already exist.
            pass
        # Recursively copy each file/directory inside the current source.
        for filename in os.listdir(src):
            src_path = src + "/" + filename
            dst_path = dst + "/" + filename
            copy_recursive(src_path, dst_path)
    else:
        # Source is a file; copy it.
        copy_file(src, dst)

# -----------------------------------------------------------------------------
# Main Process
#
# Here we copy everything from the /sd/Beryllium-files directory into the root (/).
# Each file or directory under /sd/Beryllium-files is copied so that, for example,
# a file at "/sd/Beryllium-files/foo.txt" ends up as "/foo.txt" on the root.
# -----------------------------------------------------------------------------
print("Copying contents of /sd/Beryllium-files to /")
copy_recursive("/sd/Beryllium-files", "/")
print("Copy completed!")
```