adafruit/circuitpython #9507: please support XIAO ESP32S3

marcobrianza, 2024-08-11:
I see that most XIAO are already supported by circuitpython but XIAO ESP32S3 is missing. in there a reason for this?

https://wiki.seeedstudio.com/xiao_esp32s3_getting_started/

deshipu, 2024-08-11:
The pull request is here. It's waiting for a USB VID/PID from Seeed.

https://github.com/adafruit/circuitpython/pull/9118

marcobrianza, 2024-08-11:
If I understand correctly the board identifies itself with standard USB VID/PID instead of specific ones
I think that lolin boards do the same but are suported.
What is the blocking issue in having a shared VID/PID?

dhalbert, 2024-08-11:
> What is the blocking issue in having a shared VID/PID?

The reasons for having unique VID/PIDS are discussed here: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython/frequently-asked-questions#faq-3130480

tannewt, 2024-08-12:
If a manufacture won't grant a USB PID, you can also request a PID from [pid.codes](https://github.com/pidcodes/pidcodes.github.com) as a third-party adding open source software on a closed source board.

djairjr, 2024-08-16:
Ok guys, there is some issue with the firmware generation process. Both firmwares generate a read-only CIRCUITPY. But REPL works, and the pinout are correct. There is something wrong with TinyUSB? Tested on my local machine with dual boot (Ubuntu/Windows) and a Amazon virtual machine (also Ubuntu).

See if this firmwares works for you, ok?

https://github.com/djairjr/Seeed_Xiao_ESPS3_Sense_Circuitpython

dhalbert, 2024-08-16:
@djairjr Let's discuss in #9528.

marcobrianza, 2024-08-20:
Hi I have written an email to techsupport@seeed.io asking for VID PID and below is the replay I got
I think this helps the official support ;-) 
...

Hi Marco,

Thank you for reaching out. Here are the VID and PID for the XIAO ESP32-S3:

VID: 0x2886
PID: 0x8056
These details should assist with the process. If you have any more questions or need further assistance, feel free to ask.

Regards!

Frank
Seeed Technical Support Team

dhalbert, 2024-12-13:
Done by #9530. The Sense is the base board with an add-on board. The same build can be used for both.

closed 2024-12-13 by dhalbert