adafruit/circuitpython #9517: RP2350 Audio crashes

todbot, 2024-08-13:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0-alpha.2350-6-gf064354648-dirty on 2024-08-13; SparkFun Pro Micro RP2350 with rp2350a
```

### Code/REPL

```python
import board, audiocore, audiopwmio
audio = audiopwmio.PWMAudioOut(board.D0)  # or any GPIO
wave = audiocore.WaveFile("laser2.wav")
audio.play(wave)
while True:
  pass
```

### Behavior

As soon as `audio.play()` is run, board resets.

### Description

Playing a `RawSample` from RAM works, e.g. this does not crash:
```py
import board, audiocore, audiopwmio, array
audio = audiopwmio.PWMAudioOut(board.D0) 
sample = audiocore.RawSample( array.array("h", [0] * 32) )
audio.play(sample, loop=True)
```

### Additional information

On SparkFun Pro Micro RP2350, happens with both latest Pico2 firmware (no PSRAM) on Amazon S3 and https://github.com/adafruit/circuitpython/pull/9515 firmware (PSRAM enabled).

tannewt, 2024-08-13:
I think this was caused by https://github.com/raspberrypi/pico-sdk/issues/1817

todbot, 2024-08-13:
Confirmed that #9520 fixes this issue.

KRAxxKakou, 2024-08-14:
I have no idea what's going on in my thing    I did not approve of this   I don't know what is this   Or who   I'm at work  @GitHub

KRAxxKakou, 2024-08-14:
- [ ] I[u see eyes**```

> - [ ] ```**]()🫥🫥🫥

PR #9520 (2024-08-14): Switch to pico-sdk style interrupt names.
CMSIS-style is broken in pico-sdk 2.0.0.

config_autogen is less used in SDK 2.0.0 so remove everything but the rename_exceptions header. It is guarded internally by `LIB_CMSIS_CORE` so we can leave it.

Fixes #9517
commit a579eab1 (2024-08-14): Switch to pico-sdk style interrupt names.

CMSIS-style is broken in pico-sdk 2.0.0.

Fixes #9517
closed 2024-08-14 by dhalbert