adafruit/circuitpython #9519: Metro RP2350 I2C pins swapped

todbot, 2024-08-13:
### CircuitPython version

```python
n/a, don't have the board yet
```

### Code/REPL

```python
import board; i2c = board.I2C()
```

### Behavior

What I expect one would see: 
```
>>> import board; i2c = board.I2C()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Invalid pins
```

### Description

In https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/boards/adafruit_metro_rp2350/mpconfigboard.h, it has:
```c
#define DEFAULT_I2C_BUS_SCL (&pin_GPIO24)
#define DEFAULT_I2C_BUS_SDA (&pin_GPIO25)
```
yet on Section 1.2.3, page 19 of the [RP2350 datasheet PDF](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf), it lists GPIO24 as I2C0 SDA and GPIO25 as I2C0 SCL:
[image]

tannewt, 2024-08-14:
We're revising the metro anyway for the 80-pin chip so we'll fix this then.

dhalbert, 2025-01-30:
Board was revised to use Rp2350B, so this is moot now.

closed 2025-01-30 by dhalbert