adafruit/circuitpython #9527: seeed xiao ble circuitpyhton 9.1.1 bluetooth, core code crashed hard whoops!

ozzloy, 2024-08-15:
### CircuitPython version

```python
adafruit-circuitpython-bundle-9.x-mpy-20240730
adafruit-circuitpython-seeed_xiao_esp32c6-en_US-9.1.1.bin
seeed xiao ble
```

### Code/REPL

```python
from adafruit_ble import BLERadio

radio = BLERadio()
print("scanning")
found = set()
for entry in radio.start_scan(timeout=60, minimum_rssi=-80):
    addr = entry.address
    if addr not in found:
        print(entry)
    found.add(addr)

print("scan done")
```

### Behavior

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
scanning
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x124
load:0x4086c410,len:0xbe0
load:0x4086e610,len:0x2360
entry 0x4086c410

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```
then after hitting reset and reconnecting serial console:
```

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
scanning
<Advertisement flags=<AdvertisingFlags general_discovery le_only > >
<Advertisement flags=<AdvertisingFlags  > >
scan done

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

i'm trying to scan using bluetooth.  i wouldn't be surprised if i needed to copy over another directory or something.

ozzloy, 2024-08-15:
got some help in discord, tried this sample code
https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/examples/ble_simpletest.py
got this output
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
scanning
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0x1f (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x124
load:0x4086c410,len:0xbe0
load:0x4086e610,len:0x2360
entry 0x4086c410

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

ozzloy, 2024-08-15:
after hitting reset and reconnecting serial, running the [same sample code](https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/main/examples/ble_simpletest.py),

i saw meaningful output:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
scanning
<Address 66:df:81:01:b7:62> <Advertisement flags=<AdvertisingFlags  > >
        Advertisement(data=b"\x0e\xff\x03\x37\x51\x10\x7a\x87\xdb\x36\xcd\xbc\x36\x54\x56\x02\x01\x18\x03\x03\xbe\xfe")

<Address cd:d0:37:da:bc:11> <Advertisement flags=<AdvertisingFlags general_discovery le_only > >
        Advertisement(data=b"\x02\x01\x06\x0d\x16\xed\xfe\x02\x00\x95\xd6\x76\xd8\xd9\x53\xfc\x90\x03\x03\xed\xfe")

<Address 4e:79:a7:6a:95:48> <Advertisement flags=<AdvertisingFlags  > >
        Advertisement(data=b"\x11\xff\x03\x37\x51\x10\x3a\x07\xa2\x37\xa7\x26\x52\x01\x93\x1a\xf5\x2b\x02\x01\x18\x03\x03\xbe\xfe")

<Address 47:31:ca:40:1d:4b> <Advertisement flags=<AdvertisingFlags  > >
        Advertisement(data=b"\x11\xff\x03\x37\x51\x10\x3a\x07\xa2\x37\xa7\x26\x52\x01\x93\x1a\xf5\x2b\x02\x01\x18\x03\x03\xbe\xfe")

<Address 54:56:4f:d5:c7:14> <Advertisement flags=<AdvertisingFlags  > >
        Advertisement(data=b"\x11\xff\x03\x37\x51\x10\x3a\x07\xa2\x37\xa7\x26\x52\x01\x93\x1a\xf5\x2b\x02\x01\x18\x03\x03\xbe\xfe")

<Address 69:83:9e:2b:37:04> <Advertisement flags=<AdvertisingFlags  > >
        Advertisement(data=b"\x11\xff\x03\x37\x71\x10\x3a\x07\xa2\x37\xa7\x26\x52\x01\x93\x1a\xf5\x2b\x02\x01\x18\x03\x03\xbe\xfe")
```

dhalbert, 2024-08-28:
Need to test on another board we have.

ozzloy, 2024-09-07:
let me know if you think of anything i could do to help out.

do you have recommendations on "getting started on bluetooth firmware troubleshooting" reading?  i have experience writing C and have written simple firmware for stepper motors and sonar sensors.  any pointers would be greatly appreciated.

dhalbert, 2024-09-08:
Could you try 9.2.0-alpha.2351 for this board?

ozzloy, 2024-09-08:
good news!  it has yet to crash!  thanks for the update!

### more details
with this code:
```python
from adafruit_ble import BLERadio
ble = BLERadio()
ble.name
```

i see this output:

> code.py output:
> CIRCUITPY2262ba

with this code:
```python
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService

ble = BLERadio()
uart = UARTService()
advertisement = ProvideServicesAdvertisement(uart)

ble.start_advertising(advertisement)
while True:
    # Normally other work would be done here after connecting.
	pass
```

i can see `CIRCUITPY2262ba` on my phone!  ~~but not from my computer.~~  i saw it on my computer after turning the bluetooth off and back on.

### i'll test more in 9ish hours
i'll have to try more stuff with it more after getting some sleep.

code pointers, ### please
if you happen to know where it is, can you point me to the part of https://github.com/adafruit/Adafruit_CircuitPython_Bundle that made the difference?  i would like to read it over to see what's involved.

thanks again!

tannewt, 2024-09-12:
> good news! it has yet to crash! thanks for the update!

Looks like it is fixed in the alpha with the IDF and BLE updates. Closing.

closed 2024-09-12 by tannewt