adafruit/circuitpython #9528: Filesystem READ ONLY on Circuitpython 9.1.1 and 9.2.0 builds for Seeed Xiao ESP32S3_Sense

djairjr, 2024-08-16:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.1 on 2024-08-16; Seeed Xiao ESP32-S3 Sense with ESP32S3
```

### Code/REPL

```python
>>> f = open('settings.toml', 'w')
f.write('CIRCUITPY_WIFI_SSID = "wifissid"\n')
f.write('CIRCUITPY_WIFI_PASSWORD = "wifipassword"\n')
f.write('CIRCUITPY_WEB_API_PASSWORD = "webpassword"\n')
f.close()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 30] Read-only filesystem
>>>
```

### Behavior

Can't save any file on my Circuitpython.
Circuitpython thumb drive wasn't mounted on WIndows (but ok in linux)

Compiled in local machine dual boot Ubuntu 22.04 and also Virtual Amazon Machine.

### Description

Try to compile in 9.1.1 and 9.2.0.

Same error;

djairjr, 2024-08-16:
Tested workaround: manually install tinyuf2 bootloader at https://github.com/adafruit/tinyuf2/releases and copy .uf2 files also generates a unusable readonly CIRCUITPY (not show on windows system).

dhalbert, 2024-08-16:
The read-only is by design. CIRCUITPY is read-only from your code unless you remount it as read/write in `boot.py`. It is read/write from the host computer that has it mounted as a drive. This is to prevent corruption of CIRCUITPY if both the host computer your code try to write to it at the same time.

This is not true for boards like C3, C6, and plain ESP32, which do not present a CIRCUITPY USB drive.

dhalbert, 2024-08-16:
The problem you describe with CIRCUITPY not appearing on your Windows machine is often due to third-party software that is interfering with the mounting of CIRCUITPY. See https://learn.adafruit.com/welcome-to-circuitpython/troubleshooting#circuitpy-drive-does-not-appear-or-disappears-quickly-3004181

djairjr, 2024-08-16:
No. It was not the problem.
I am using many other boards here. They all work.
When I am in linux, using fsck I got a message saying something about a dirty byte.
Again, in my Ubuntu boot, the CIRCUITPY is mounted as usual.
In Windows, only Serial port appears, but no Disk at all.

Thonny was able to access files in Circuitpython File system. But the Windows Explorer no. This happens only in windows.

dhalbert, 2024-08-16:
If you reformat the filesystem in the REPL, does it fix the problem?
```py
import storage
storage.erase_filesystem()
```
You can join us in discord https://adafru.it/discord, `#help-with-circuitpython`, if you would like some real-time help.

djairjr, 2024-08-16:
It's been a long time since I compiled Circuitpython directly from source. The last time was with Seeed's other card, the RP2040.
I have several boards here, including some ESP32C3, which are working perfectly using WebESPTool and the firmware available on your website.

I'm trying to make the Seeed Xiao ESP32 S3 Sense board compatible with Circuitpython, a project I had already started some time ago.

My system compiles the firmware properly, without any errors. And I can save the .bin firmware using WebEspTool and any other tool.

The problem is that the CIRCUITPY disk drive is only accessible from Linux. Even if I burn the firmware in Windows. The variation doesn't matter.

On Linux I can access it as a VFAT disk drive, write the files, install the libraries via circuit, everything is normal.

In Windows, the disk drive does not appear. Thonny can access via USB, I can read the files, format the drive, but I can't write any changed files.

I have already generated firmwares in versions 9.1.1 and 9.2.0 and the problem persists. I also used a tinyuf2 bootloader on the board. When I copy the uf2 firmware generated by circuitpython, I can't access the drive from Windows either.

djairjr, 2024-08-19:
I'll leave the observation here, in case anyone has the same type of problem. The folks at Seeed Studio still need to answer some questions regarding USB device registration. My Windows system was mounting the CIRCUITPY drive as read-only because it was not recognizing the drive as a drive. Probably due to a problem in the registry, which identified the USB device incorrectly.

I added values ​​from other boards (from ESPRESSIF itself) and managed to perform a first compilation. With the help of Danh, I managed to generate correct source files that, when compiled, generated a functional firmware, capable of running the camera and displaying the image on the XIAO Round Display.

This firmware could be [downloaded here](https://github.com/djairjr/Seeed_Xiao_ESPS3_Sense_Circuitpython), but still isn't Official Circuitpython, due this issue with USB/PID and USB/VID that need to be solved by Seeed Studio.

closed 2024-08-19 by djairjr