adafruit/circuitpython #9551: Adafruit ESP32-C6 Feather does not set NEOPIXEL_I2C_POWER high despite being defined in board.c

BlitzCityDIY, 2024-08-22:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.2 on 2024-08-22; Adafruit Feather ESP32-C6 4MB Flash No PSRAM with ESP32C6
```

### Code/REPL

```python
import board
from digitalio import DigitalInOut, Direction, Pull

pwr = DigitalInOut(board.NEOPIXEL_I2C_POWER)
pwr.direction = Direction.OUTPUT
print(pwr.value)
pwr.value = True
print(pwr.value)
```

### Behavior

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
False
True
```

### Description

the board definition for the C6 has the pin reset function in **board.c**:

```
bool espressif_board_reset_pin_number(gpio_num_t pin_number) {
    if (pin_number == 20) {
        // Turn on I2C power by default.
        gpio_set_direction(pin_number, GPIO_MODE_DEF_OUTPUT);
        gpio_set_level(pin_number, true);
        return true;
    }

    return false;
}
```

but for some reason does not seem to be working. i tested an esp32-s2 and esp32-s3 feather that have the same functions and they are setting the pin high on startup, so it seems to be isolated to the C6. i discovered this while running an i2c scan and continually getting:
`RuntimeError: No pull up found on SDA or SCL; check your wiring`
i had assumed incorrectly that the i2c pins were included in the pins that had been changed before 9.1.2 so i hadn't looked into it more closely.

### Additional information

In Arduino, the pin is automatically being set high so the pin number is correct (20).

dhalbert, 2024-08-22:
I reproduced this. It's mysterious. Debugging.

BlitzCityDIY, 2024-08-23:
@migs35323 try installing this build artifact bin file from https://github.com/adafruit/circuitpython/pull/9553: https://github.com/adafruit/circuitpython/actions/runs/10519459507/artifacts/1845423644

this fixes the power pin issue

migs35323, 2024-08-23:
> @migs35323 try installing this build artifact bin file from #9553: https://github.com/adafruit/circuitpython/actions/runs/10519459507/artifacts/1845423644
> 
> this fixes the power pin issue

yep, it fixed it apparently! breakout board is powered on automatically, not even needed to create the i2c

dhalbert, 2024-08-23:
Fixed in `9.1.x` (no release yet) by #9553. Will be merged into `main` soon.

PR #9553 (2024-08-23): correct board special pin setup
- Fixes #9551 

`espressif_board_reset_pin_number()` was being used on many boards to  set certain pins high or low on startup. However, it was using basic `gpio` functions without ensuring that the IOMUX for pin was set to GPIO. This meant the GPIO settings were not necessarily taking effect. This may have caused issues on other boards we didn't know about besides the Feather C6 from #9551.

I added a helper routine to ensure that `gpio_config()` was called before setting the pin level. `gpio_config()` does the IOMUX configuring.

A helper routine to set pulls  [... cut]
closed 2024-08-23 by dhalbert