adafruit/circuitpython #9565: ESP32-S3 CP9 PWM "RuntimeError: Internal Error" for certain parameters

anecdata, 2024-08-28:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.7 on 2023-10-19; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3

vs.

Adafruit CircuitPython 9.1.2 on 2024-08-22; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3
Adafruit CircuitPython 9.2.0-alpha.2351 on 2024-08-22; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3
```

### Code/REPL

```python
# Adafruit CircuitPython 8.2.7 on 2023-10-19; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3
>>> import board, pwmio
>>> a = pwmio.PWMOut(board.A0, frequency=1, duty_cycle=64)
>>> 

# Adafruit CircuitPython 9.2.0-alpha.2351 on 2024-08-22; Adafruit QT Py ESP32-S3 4MB Flash 2MB PSRAM with ESP32S3
>>> import board, pwmio
>>> a = pwmio.PWMOut(board.A0, frequency=1, duty_cycle=64)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Internal error
>>>
```

### Behavior

I wanted to PWM an LED for a low-key visual heartbeat, but recent releases for ESP32-S3 seem to have trouble with `frequency=1` or `frequency=2`.

All frequencies work fine on ESP32-S2 in CP9.

`frequency` >= 3 works on ESP32-S3 in CP9.

jepler, 2024-09-03:
It's difficult to report detailed error information from the constructor of a PWMOut object, because the constructor is designed to never throw an exception; instead, it can return one of a small number of generic failure reasons.

It was written this way so that the code could safely be called outside the VM, specifically so that LCD backlight objects could be created at board initialization time.

As for why it fails now and did not in the past, it is probably due to changes in esp-idf adding additional error checking or changing low level details of how the PWM peripheral is clocked. Do you know if in 8.2.7 `frequency=1` actually gave a 1Hz PWM signal?

anecdata, 2024-09-03:
> Do you know if in 8.2.7 frequency=1 actually gave a 1Hz PWM signal?

I didn't measure it precisely, just observed that an LED pulsed approximately once per second. I've used this in the past on other ports, and it works fine on ESP32-S2 in CP9.

tannewt, 2024-09-11:
Ok, this was an intentional change in the IDF: https://github.com/espressif/esp-idf/commit/8e3bc160bc7e31c59a48eb5d6d95ccf8aaff2612

CP 8 (IDF 4.4) is using REF_TICK to provide the clock for this. In IDF 5 this is no longer possible because of the commit. So, this is working as intended.

closed 2024-09-11 by tannewt

anecdata, 2024-09-12:
Thanks for tracking it down. It's disappointing since for years we've been able to have low-frequency PWM. I use it especially when I have a blocking activity that won't let me manually toggle the LED regularly.