adafruit/circuitpython #9569: boot loop on Hacktablet and Makerfabs matouch 7" S3 display boards

RetiredWizard, 2024-08-28:
### CircuitPython version

```python
CircuitPython 9.2.0-alpha.2351
```

### Code/REPL

```python
No code executed
```

### Behavior

> [13:24:07.365] Connected to /dev/ttyACM0
I (623) esp_image: segment 0: paddr=00010020 vaddr=3c150020 size=3acc0h (240832) map
I (660) esp_image: segment 1: paddr=0004ace8 vaddr=3fc9f200 size=05330h ( 21296) load
I (664) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=14718ch (1339788) map
I (869) esp_image: segment 3: paddr=001971b4 vaddr=3fca4530 size=00128h (   296) load
I (869) esp_image: segment 4: paddr=001972e4 vaddr=40374000 size=1b19ch (111004) load
I (896) esp_image: segment 5: paddr=001b2488 vaddr=600fe000 size=00100h (   256) load
I (897) esp_image: segment 6: paddr=001b2590 vaddr=600fe100 size=01008h (  4104) load
I (914) boot: Loaded app from partition at offset 0x10000
I (914) boot: Disabling RNG early entropy source...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378d11
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16dc
load:0x403c9700,len:0x4
load:0x403c9704,len:0x1054
load:0x403cc700,len:0x3090
entry 0x403c9a68
I (26) boot: ESP-IDF v5.1.1 2nd stage bootloader
I (26) boot: compile time Aug  9 2024 13:55:41
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.1
... etc...

### Description

After installing CircuitPython 9.2.0-alpha.2351 firmware on both the Hacktablet (espressif ESP32-S3 Dev Kit N8R8) and the Makerfabs Matouch 7" display boards a boot loop occurs. The boot loop doesn't occur with 9.1.2.

I've loaded the same firmware on the Adafruit ESP32S3 reverse TFT, Qualia S3 (without a display attached) and the ESP32-S3-GEEK display boards without seeing the same issue.

tannewt, 2024-08-28:
Can you reproduce this on an Adafruit or Espressif board? I don't have either of the ones that fail.

RetiredWizard, 2024-08-28:
I just tried the Matrix Portal S3 and it worked fine as well. I'm guessing it has to do with the 666 dotclock display init in board.c but I don't think I have any Adafruit/Espressif boards that initialize a dot clock display automatically.

RetiredWizard, 2024-08-28:
Another difference between the boards that are failing for me, is that I update the firmware by first flashing combined.bin and then copying the UF2. All the other S3 boards I've tested will bring up the boot loader drive via a button press sequence.  I don't have any idea if the boot loader could be a factor in this type of issue.

RetiredWizard, 2024-08-28:
I was just about to start hacking up the Makerfabs board.c file to try and identify the cause and discovered that the Makerfabs board doesn't actually init the display in board.c so that's not the source of the problem after all.

RetiredWizard, 2024-08-28:
Commenting out these two lines from mpconfigboard.h
```
#define CIRCUITPY_CONSOLE_UART_TX (&pin_GPIO43)
#define CIRCUITPY_CONSOLE_UART_RX (&pin_GPIO44)
```
and these lines from sdkconfig
```
CONFIG_ESP_CONSOLE_UART_CUSTOM=y
CONFIG_ESP_CONSOLE_UART=y
CONFIG_ESP_CONSOLE_UART_CUSTOM_NUM_0=y
CONFIG_ESP_CONSOLE_UART_NUM=0
CONFIG_ESP_CONSOLE_UART_TX_GPIO=43
CONFIG_ESP_CONSOLE_UART_RX_GPIO=44
CONFIG_ESP_CONSOLE_UART_BAUDRATE=115200
```
prevents the boot loop on the Makerfabs board. Is there a new way of configuring the UART console?

RetiredWizard, 2024-08-28:
It looks to me like in addition to the Hacktablet and Makerfabs TFT 7" boards, the Espressif ESP32S3_LCD_EV and possibly (I don't know if non-S3 boards are affected) the Lilygo twatch_2020_v3 may be affected by this issue. I don't have either of those devices to test with.

dhalbert, 2024-09-17:
From https://github.com/adafruit/circuitpython/pull/9583#issuecomment-2354071837:

> Adafruit CircuitPython 9.2.0-alpha.2351-37-g015a5cf922 on 2024-09-16; MakerFabs-ESP32-S3-Parallel-TFT-With-Touch-7inch with ESP32S3
> Adafruit CircuitPython 9.2.0-alpha.2351-37-g015a5cf922 on 2024-09-16; ESP32-S3-DevKitC-1-N8R8-with-HACKTABLET with ESP32S3

PR #9583 (2024-09-18): Remove ESP_CONSOLE_UART from sdkconfig for 666 S3 Display panels
This fixes #9569 boot loop for the Hacktablet and Makerfabs 7" TFT ESP32-S3 666 Dotclock display boards.

It looks to me like in addition to the Hacktablet and Makerfabs TFT 7" boards, the Espressif ESP32S3_LCD_EV and possibly (I don't know if non-S3 boards are affected) the Lilygo twatch_2020_v3 may be affected by this issue. I don't have either of those devices to test with but could add the same sdkconfig changes for those boards if that makes sense?