adafruit/circuitpython #9573: Board enters safe mode after disconnect from 1 of 2 BLE connections.

FlyingThings, 2024-08-29:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.1 on 2024-07-23; FeatherS3 with ESP32S3
Board ID:unexpectedmaker_feathers3
UID:0740D1DA1CCA
```

### Code/REPL

```python
import board
from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService
import neopixel
import time

def monitor():
    ble = BLERadio()
    #uart = UARTService()
    #advertisement = ProvideServicesAdvertisement(uart)
    pix = neopixel.NeoPixel(board.NEOPIXEL,10)
    tog = True
    last_time_tog = time.monotonic()
    
    while True:
        try:
            #connect to two uart devices.
            while ble.connected and any(UARTService in connection for connection in ble.connections) and len(ble.connections) == 2:
                                        

                try:
                    if right_paw_uart.in_waiting:
                        print(right_paw_uart.read(4))
                    if left_paw_uart.in_waiting:
                        print(left_paw_uart.read(4))
                    if time.monotonic()-last_time_tog>=1:
                        last_time_tog = time.monotonic()
                        print('here')
                        if tog:
                            pix.fill(0xff0000)
                            tog = not tog
                        else:
                            pix.fill(0)
                            tog = not tog
                except ConnectionError:
                    print('NOT CONNECTED')
                    break

            print("disconnected, scanning", len(ble.connections))
            try:
                for advertisement in ble.start_scan(ProvideServicesAdvertisement, timeout=1):
                    if UARTService not in advertisement.services:
                        continue
                    #ble.connect(advertisement)
                    if advertisement.complete_name == "RIGHTPAW":
                        right_paw = ble.connect(advertisement)
                        right_paw_uart = right_paw[UARTService]
                    if advertisement.complete_name == "LEFTPAW":
                        left_paw = ble.connect(advertisement)
                        left_paw_uart = left_paw[UARTService]
                    print(f"connected {advertisement.complete_name}")
                    break
            except ConnectionError:
                print('connection error:not connected')
            ble.stop_scan()
        except ConnectionError as er:
            print('IDK its a connection error:',er)
            time.sleep(1)
monitor()
```

### Behavior

Board will connect to 2 devices "LEFTPAW" and "RIGHTPAW" normally, and receive messages as expected. Restarting one of the connected devices once is usually OK, code will reconnect normally. After a number of restarts of a connected device (sometimes up to 3 restarts) this board will hang, and then finally crash to safe mode.
Here is the caught error immediately before safe mode:

Connection has been disconnected and can no longer be used. Create a new connection.

And the safe mode message:

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

### Additional information

I have similar code running without issue on a Feather NRF52840. The connected devices are Circuit Playground Bluefruit boards running CircuitPython.

tannewt, 2024-09-12:
Please retest with the absolute newest build.

FlyingThings, 2024-09-13:
Same issue as described above. Here is the error after it restarts into safe mode:

```
Adafruit CircuitPython 9.1.3-2-gfad755d5e7 on 2024-09-12; FeatherS3 with ESP32S3
>>>
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

I appreciate the help!

tannewt, 2024-09-13:
Thanks for the retest! It looks like you got the latest of the 9.1.x branch. Please try https://adafruit-circuit-python.s3.amazonaws.com/bin/unexpectedmaker_feathers3/en_US/adafruit-circuitpython-unexpectedmaker_feathers3-en_US-20240913-main-PR9619-64cb762.uf2 which is on the main branch and includes IDF updates.

dhalbert, 2024-09-17:
Could you try 9.2.0-beta.0?

FlyingThings, 2024-09-22:
9.1.x same behavior. Crashes when connected device disconnects.
```
Adafruit CircuitPython 9.2.0-alpha.2351-35-g64cb76213f on 2024-09-13; FeatherS3 with ESP32S3
>>>
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```
9.2.0 beta.0 has a little bit different behavior, can initially connect, and reconnect, but one a message is received it crashes.
```

Adafruit CircuitPython 9.2.0-beta.0 on 2024-09-17; FeatherS3 with ESP32S3
>>>
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
```

FlyingThings, 2024-10-12:
OK retested with just one connection. Still crashing once a message is sent from the peripheral.
```
Adafruit CircuitPython 9.2.0-beta.0 on 2024-09-17; FeatherS3 with ESP32S3
>>>
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.

```

also tried the latest beta and same behavior

```
Adafruit CircuitPython 9.2.0-beta.1-1-g55823ba3b0 on 2024-10-11; FeatherS3 with ESP32S3
>>>
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.

```

dhalbert, 2024-10-13:
Could you post or point to the code for the LEFTPAW and RIGHTPAW devices? Confirming that these are CPB's.

hartmutbohmer, 2024-12-17:
Same issue. Connecting and disconnecting loop with prior message exchange either alternating between two connections or one connection only. Tried different timings and enhanced double cleanup mechanism.

gkecskes78, 2025-02-12:
I observed entering into safe mode on nRF dongle with 9.2.4 when accidentally tried to establish a connection, even a connection already exists. Just use the codes below, hope it helps.

`###		Dongle 2
import _bleio
from adafruit_ble import BLERadio
from adafruit_ble.services.nordic import UARTService
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
conn=[]
addresses=[]

def conNext(timeout):
    device = 'de:4e:27:f0:bc:e6'    #   Enter your MAC address
    print("Looking for address " + device, end="")
    match = False
    while not match:
        tmp = ble.start_scan(ProvideServicesAdvertisement,timeout=0.5)
        for t in tmp:
            if device in str(t.address):
                print(" Found " + device + "  " + str(t.address)+" -> OK")
                addresses.append(t)
                match = True
                break
        ble.stop_scan()
    i=0
    for adv in addresses:
        print("Connecting to address " + str(adv.address),end="")
        match1 = False
        while not match1:
            if UARTService not in adv.services:
                continue
            try:
                cnt=ble.connect(adv)
                #cnt1=cnt[UARTService]      #   with or without this line the issue can be reproduced
                if match==True:
                    conn.append(cnt)
                    match1 = True
                    print(" => OK")
            except Exception as e:
                print(" => Exception #1: "+str(e))
        i+=1
    print("Disconnecting...")
    for c in conn:
        c.disconnect()

ble=BLERadio()
uart=UARTService()
while True:
    conNext(60)
    print("Waiting 5s")`

`###		Dongle 1
import board,busio,_bleio,pwmio
from adafruit_ble import BLERadio
from digitalio import DigitalInOut, Direction
from adafruit_ble.services.nordic import UARTService
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement

ble=BLERadio()
uart=UARTService()
advert=ProvideServicesAdvertisement(uart)
addr=advert.address
_bleio.adapter.enabled=True
LED1=pwmio.PWMOut(board.LED1)

print("Device MAC address: " + str(_bleio.adapter.address))
if not ble.advertising:
    print("Start advertising ...")
    ble.start_advertising(advert)

while True:
    if ble.connected:
        ble.stop_advertising()
        LED1.duty_cycle=64000
    if not ble.connected:
        if not ble.advertising:
            ble.start_advertising(advert)
        LED1.duty_cycle=65535

`

FlyingThings, 2025-02-22:
Apologies for the late response. Crashing on 9.2.4 when peripheral sends a message.

I want to add this: the number of connections the central shows is 1 even after the peripheral turns off / disconnects.

Here is the code for one of the paws circuitplayground bluefruit. Hopefully this is enough, as crashes occur with one device restart and message.
[code.txt](https://github.com/user-attachments/files/18918388/code.txt)