adafruit/circuitpython #9599: ESP32-C6 Feather crashes and goes into safe mode when connecting to USB web workflow

BlitzCityDIY, 2024-09-05:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.3 on 2024-08-29; Adafruit Feather ESP32-C6 4MB Flash No PSRAM with ESP32C6
```

### Code/REPL

```python
import time
import board
import digitalio

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

while True:
    led.value = True
    time.sleep(0.5)
    led.value = False
    time.sleep(0.5)
```

### Behavior

When running the blink code above, I try to initiate a connection with https://code-beta.circuitpython.org/ over USB. Once the board connects, the onboard LED stops blinking and I get this message in the serial console:

```
connected

soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Unable to allocate to the heap.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.
```

### Description

I came across this while testing fixes related to [this issue](https://github.com/circuitpython/web-editor/issues/238). Mainly that when the board is running code that has a blocking `time.sleep` delay in the loop it seems to also block connecting to the web editor.

When I connected to the code editor after freshly installing CircuitPython, I was able to connect right away (running the default `print("hello world")`. After saving the blink code and then trying to reconnect to the editor, I hit the safe mode error. I've been able to repeat the safe mode error consistently.

tannewt, 2024-09-11:
This works ok for me with the latest main. Can you reproduce it with an absolute newest build?

BlitzCityDIY, 2024-09-11:
i just tested with absolute newest (`Adafruit CircuitPython 9.2.0-alpha.2351-32-g28b593341c on 2024-09-11; Adafruit Feather ESP32-C6 4MB Flash No PSRAM with ESP32C6`) and had the same safe mode error occur on https://code-beta.circuitpython.org/

commit 03b077dd (2024-09-12): Add deinit() to _bleio.Characteristic

Espressif's Characteristics allocate memory that may be on the
supervisor heap. We need to free it when stopping BLE workflow.
Otherwise, we leak the memory and eventually safe mode.

Fixes #9599
closed 2024-09-13 by dhalbert