adafruit/circuitpython #9614: System firmware error when trying to make a BLE connection on ESP32-S3

izmishi, 2024-09-11:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0-alpha.2350-17-g422eb7eff2-dirty on 2024-08-18; Seeed Xiao ESP32-S3 Sense with ESP32S3
```

### Code/REPL

```python
import _bleio

# dummy address bytes here, but also happens with valid addresses
someAddress = _bleio.Address(b'\x00\x00\x00\x00\x00\x00', 2)
_bleio.adapter.connect(someAddress, timeout=10)
```

### Behavior

Running the code immediately results in an error.
```
Traceback (most recent call last):
  File "code.py", line 5, in <module>
_bleio.BluetoothError: Unknown system firmware error: 530
```

### Description

On a Seeed Xiao ESP32-S3, when attempting to initiate a bluetooth connection to an already-paired device with a `RANDOM_PRIVATE_RESOLVABLE` address, an unknown system firmware error (530) occurs.

I've found that the actual address bytes don't matter, only the type (it also happens for `RANDOM_PRIVATE_NON_RESOLVABLE`, but not `PUBLIC` or `RANDOM_STATIC`). 

### Additional information

I've done a little digging through the CircuitPython source code and I suspect that it's a NimBLE error, in which case it would be an [Invalid HCI Command Parameters error (0x212)](https://mynewt.apache.org/latest/network/ble_hs/ble_hs_return_codes.html#return-codes-hci).

It occurs on both [9.1.1 and 9.2.0 alpha](https://github.com/djairjr/Seeed_Xiao_ESPS3_Sense_Circuitpython).

Running the above code on a Seeed Xiao nRF52840 (CircuitPython 9.0.4) performs as expected (a 'failed to connect' error after 10 seconds):
```
Traceback (most recent call last):
  File "code.py", line 5, in <module>
_bleio.BluetoothError: Failed to connect: internal error
```

tannewt, 2024-09-12:
> I've done a little digging through the CircuitPython source code and I suspect that it's a NimBLE error, in which case it would be an [Invalid HCI Command Parameters error (0x212)](https://mynewt.apache.org/latest/network/ble_hs/ble_hs_return_codes.html#return-codes-hci).

Nice job digging! Maybe we aren't flagging the address correctly internally. Want to see if you can fix it? We're happy to help guide. It's probably not a high priority for us to work on ourselves.

izmishi, 2024-09-12:
Sure, I can give it a go!

I'll definitely need guidance though

tannewt, 2024-09-12:
Awesome! Have you gotten it building? There is a tutorial for building here: https://learn.adafruit.com/building-circuitpython

We can answer development questions [on Discord](https:/adafru.it/discord) in #circuitpython-dev.

izmishi, 2024-09-13:
I've managed to get it building

izmishi, 2024-09-15:
What's the next step towards fixing it?

tannewt, 2024-09-16:
You'll need to sort out why the error is being returned and change the input parameters so it succeeds.