adafruit/circuitpython #9623: Add master volume to Audioio

Ctrl32, 2024-09-14:
I know you can change the volume(gain) using the circuitpython mixer object. However, there are a few reasons that I think it should also be added to the audioio object: 
- Better preformance. I've tried to make a mp3 player on the Adafruit Pygamer but adding a mixer slows it down too much so the audio sounds poppy and choppy. Changing volume is really just multiplying all the samples by the volume. So one float multiplication built into the audioio module will probably be faster than having to create another object to do that for you.
 
- Master volume control, so you don't have to set the volume for each voice.

tannewt, 2024-09-16:
I don't really think we want this. To do it, we'd need most of what mixer is doing (the extra buffer and multiply). I don't think the object creation is what slowing things down. I suspect it's the multiplying of all of the samples. We could probably make mixer work better in single voice mode that you could use after a multi-voice mixer.

tannewt, 2024-09-23:
You don't. That's my mistake. I still think this is better done as an optimization on Mixer though.

gamblor21, 2024-09-23:
I have been spending a lot of time deep-diving into the audio code lately and I'm not sure if moving it from Mixer to anywhere else will see much of a speedup. As is the Mixer code basically only does what you describe, multiple each sample by a value. I'm not even sure there is room for much optimization in there.

The other issue is the audioio code is port specific so would have to be done for each port.

I do not have a PyGamer so cannot easily try it out unfortunately.