adafruit/circuitpython #9624: Touchio - slow updates when touching multiple touch-pins

BennettB123, 2024-09-14:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.2 on 2024-08-22; Waveshare ESP32-S3-Zero with ESP32S3
```

### Code/REPL

```python
from touchio import TouchIn
import board
import time

pins = []
pins.append(TouchIn(board.D13))
pins.append(TouchIn(board.D12))
pins.append(TouchIn(board.D11))
pins.append(TouchIn(board.D10))
pins.append(TouchIn(board.D9))
pins.append(TouchIn(board.D8))
pins.append(TouchIn(board.D7))

while True:
    i = 0
    for pin in pins:
        print(i, "raw_value: ", pin.raw_value)
        i += 1
    print()
    time.sleep(.01)
```

### Behavior

The example code prints out the raw_values of all touch pins.  When 0 or 1 pins are touched, the values update extremely fast (updating every loop iteration). However, whenever 2 or more pins are touch simultaneously, All of the other pin's raw_values update extremely slowly (in some cases taking several seconds to update). 

### Additional information

My goal is to be able to accurately/quickly handle simultaneous touch inputs.

tannewt, 2024-09-16:
@dhalbert did you mean to self-assign this?

dhalbert, 2024-09-16:
> @dhalbert did you mean to self-assign this?

Yes, I discussed it with @BennettB123 in discord and plan to test this in more detail.

dhalbert, 2024-09-17:
@BennettB123 I tested this with an Adafruit Metro ESP32-S3, running CircuitPython 9.2.0-beta.0. I connected some ~4-inch jumpers with alligator clips at one end to D8 through D11. Touching two alligator cips with one hand and two with another produced raw values of 65535, as expected, but the loop was still running very quickly. I also tried clipping coins with the alligator clips, and that made no difference.

I am suspecting something about your setup, but I'm not sure what it is. We did talk about whether there was coupling between the wires, but I twisted all the wires together and it made no difference

BennettB123, 2024-09-17:
@dhalbert Thanks for testing!  I went ahead and upgraded to 9.2.0-beta.0 to rule out differences there and I'm still getting the same behavior.  One thing I may not have made clear is that the loop still runs very quickly when I'm touching several pins.  However, the actual raw_values don't update every loop iteration like they do when no pins are touched.

If you touch 3 of your alligator clips and leave the 4th alone, do you notice any slowdowns in how often the raw_value of the 4th pin updates?  When I do this scenario, the 4th pin's raw_value only updates once every second (rather than every loop iteration when no pins are touched)

dhalbert, 2024-09-18:
Ah, I did misunderstand you about the loop timing. However, I am seeing quick response when touching three and then adding a fourth. I will try some more experiments tomorrow.