adafruit/circuitpython #9626: espressif: BLE prevents true deepsleep, even when not enabled (?)

Sola85, 2024-09-14:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.3 on 2024-08-30; ESP32-S3-DevKitM-1-N8 with ESP32S3
and latest
```

### Code/REPL

```python
import alarm
import microcontroller as uc
uc.cpu.frequency = 240_000_000 # change

alarm.exit_and_deep_sleep_until_alarms()
```

### Behavior

The title is my current guess about what is going on. I am currently debugging a custom board with a ESP32-S3-Mini-1-N8 (8MB flash) which does not want to enter deep sleep and instead draws ~ 40mA, which is obviously way too high. (Note that USB is detached and the board reset before the current measurement.)

### Description

- Clue 1: Sleep Current consumption depends on CPU frequency: ~15mA@40MHz, ~40mA@240MHz. This fact makes me conclude that its not an issue with the board but rather a software issue.
- Clue 2: `Waveshare ESP32-S3-Zero with ESP32S3` (4MB flash) is not affected.
- Clue 3: If I change [this line](https://github.com/adafruit/circuitpython/blob/main/main.c#L669) in `main.c` to `if (true || awoke_from_true_deep_sleep || !supervisor_workflow_active()) {` the board sleeps correctly with the expected sleep current of a few microamps.
- Clue 4: I added the line 
`mp_printf(MP_PYTHON_PRINTER, "tud_ready(): %d, websocket_connected(): %d, ble_serial_connected(): %d", tud_ready(), websocket_connected(), ble_serial_connected());` to the top of [`supervisor_workflow_active()`](https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/workflow.c#L68), which prints: 
`"tud_ready(): 1, websocket_connected(): 0, ble_serial_connected(): 1"`

Hence I arrive at the conclusion that BLE is preventing deepsleep. This also explains why the Waveshare ESP32-S3-Zero is not affected, since it doesn't have BLE enabled at build time. 

Note that I have not enabled anything regarding BLE to reproduce this. `settings.toml` is empty and not even `adafruit_ble` is installed. The above code is the only file on the board after a fresh install of CircuitPython.

tannewt, 2024-09-16:
Can you make a PR for Clue 3 please? That makes it easier to see changes.

Is the issue that `ble_serial_connected` is true when it shouldn't be?

Thanks for looking into this!

Sola85, 2024-09-16:
> Can you make a PR for Clue 3 please?

Yes, will do tomorrow. But like I said, that was just a temporary change for debugging, by making CP never fake deep sleep, even when it thinks that a workflow is currently active.

> Is the issue that ble_serial_connected is true when it shouldn't be?

Yes that's my current guess. If I'm not mistaken this means that deepsleep is currently broken on all 8MB esp boards (specifically: all boards that have BLE enabled)

bablokb, 2024-09-17:
> Yes that's my current guess. If I'm not mistaken this means that deepsleep is currently broken on all 8MB esp boards (specifically: all boards that have BLE enabled)

There is #9464 that seems to address this.

Sola85, 2024-09-17:
> Can you make a PR for Clue 3 please?

Here is a PR to compare the changes: https://github.com/Sola85/circuitpython/pull/2. Removing `ble_serial_connected()` from `supervisor_workflow_active()` makes deep sleep work again (For course this is not a proper fix, just a debugging step).

dhalbert, 2024-10-03:
I have a fix - will be doing a PR soon.

dhalbert, 2024-10-04:
@Sola85 @dy0x Would you be able to test an artifact from here? https://github.com/adafruit/circuitpython/actions/runs/11169629161?pr=9679

dy0x, 2024-10-04:
> @Sola85 @dy0x Would you be able to test an artifact from here? https://github.com/adafruit/circuitpython/actions/runs/11169629161?pr=9679

This seems to have fixed the issue on my boards.

dhalbert, 2024-10-04:
> This seems to have fixed the issue on my boards.

Great, thank you for testing.

Sola85, 2024-10-04:
@dhalbert Thanks for working on this! 

Unfortunately I can only try out the fix next week. But given the problem description, I'm pretty confident that @dy0x and me experienced the same issue, so if it's fixed for him it's probably fixed for me.
If not, I'll let you know next week.

PR #9679 (2024-10-04): Correct `ble_serial_connected()` when BLE workflow was never started
- Fixes #9626.
- Fixes #9464 (already closed in favor of #9626).

`ble_serial_connected()` was checking `_tx_packet_buffer.conn_handle` as an indicator of whether BLE was connected. But that value was not initialized when BLE hadn't started up at all, and so `ble_serial_connected()` was returning `true` incorrectly in that case. The check should have also included a check for deinited. I added that and refactored it into a common-hal routine.

I asked the issue authors to test. It fixed the problem for at least one of them: h [... cut]