adafruit/circuitpython #9630: RP2040 - PIO sideset for pin direction

jchawks, 2024-09-15:
Currently,  the RP2040 PIO support covers all the hardware configuration bits _except_  the state machine
exec control for SIDE_PINDIR:   this is defined in the hardware_structs header pio.h,  for PIO_SMx_EXECCTRL : SIDE_PINDIR.
The bit next to it, SIDE_EN, _is_ supported:
The line parsing in adafruit_pioasm.py only checks for 'opt', and not 'pindirs':
```...
elif line.startswith(".side_set"):
                sideset_count = int(line.split()[1], 0)
                sideset_enable = "opt" in line
```
And the corresponding StateMachine() kwargs mapping would need to be added; the support would be analogous
to the SIDE_EN arg, which is now documented as:
- sideset_enable _(bool)_   True when the top sideset bit is to enable. This should be used with the “`.side_set # opt`” directive
(https://docs.circuitpython.org/en/latest/shared-bindings/rp2pio/index.html)

The idea being to support the PIO line
`.side_set 1 opt pindirs`.

The rest works great; very easy to use. Many thanks for the good work.

jchawks, 2024-09-16:
Thanks - I'll add the request this over there as well, linking back to here.

jchawks, 2024-09-16:
> WARNING
>
> **This message has originated from an External Source. Please use proper judgment and caution when opening attachments, clicking links, or responding to this email.**
>
> would you like to take a stab at adding it?

Sure:   I found a workaround for this project, but would like to do so for upcoming designs.
I'm a github noob (lifelong member of the 'other' team (SVN) 😉 ) but will do my homework on how this works.
Will have something in a few days...
Thanks!

Joe Curren

Principal Development Engineer

Amphenol All Sensors Corporation

[cid:12e27e3c-fb1a-425e-90fd-bb6d96e33948] ***@***.******@***.***>

[cid:44032606-8be7-4323-931e-8b4b321b1081] (408) 225-4314 ext. 206

[cid:3fad2755-6615-45a3-b424-441261466540] 16035 Vineyard Boulevard,
      Morgan Hill, CA 95037 USA

[cid:20878fb9-13c5-48e1-b6a2-516bdb97687d] www.allsensors.c<https://hubs.ly/H0PMZJ80>om

[cid:5c53789d-b5c3-44e8-87a6-8df2318cf844]

[cid:f15dbc28-1480-48ca-b052-429500f2dfac] [cid:54cfec54-6d46-4b5a-b4a0-51b889043638]  [cid:cbb940f0-be41-4c3e-b3ab-c8364197e0aa]

This e-mail is confidential. If you are not the intended recipient, you must not disclose or use the information contained in it. If you have received this e-mail in error, please tell us immediately by return e-mail and delete the document. No recipient may use the information in this e-mail in violation of any civil or criminal statute. We disclaim all liability for any unauthorized uses of this e-mail or its contents, and accept no liability or responsibility for any damage caused by any virus transmitted with this e-mail.

tannewt, 2024-09-17:
This is a good git tutorial: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

PR #9849 (2024-12-02): rp2pio: Add `sideset_pindirs` parameter to StateMachine constructor
This allows use of PIO programs that need to control pin *direction* from the side set portion of the instruction.

Closes #9630 

Will need an associated update to rp2pio to enable the assembler directive
closed 2024-12-02 by tannewt