adafruit/circuitpython #9642: adafruit_datetime.datetime.fromisoformat() doesn't support datetimes ending with "Z" representing UTC

skjdghsdjgsdj, 2024-09-18:
### CircuitPython version

```python
9.1.4
```

### Code/REPL

```python
import adafruit_datetime
print(adafruit_datetime.datetime.fromisoformat("2024-09-18T19:33:00Z"))
```

### Behavior

A `ValueError` is raised with message "Invalid isoformat string: '2024-09-18T19:33:00Z'"

### Description

[ISO 8601](https://en.wikipedia.org/wiki/ISO_8601) says:
"If the time is in [UTC](https://en.wikipedia.org/wiki/Coordinated_Universal_Time), add a Z directly after the time without a space. Z is the zone designator for the zero UTC offset. "09:30 UTC" is therefore represented as "09:30Z" or "T0930Z". "14:45:15 UTC" would be "14:45:15Z" or "T144515Z"."

I'm calling a REST API that sometimes returns datetimes this way. `adafruit_datetime.datetime.fromisoformat()` doesn't seem to support them and raises a `ValueError`.

RetiredWizard, 2024-09-18:
I think this is probably a duplicate of https://github.com/adafruit/Adafruit_CircuitPython_datetime/issues/22

skjdghsdjgsdj, 2024-09-18:
I think you're right.

closed 2024-09-18 by skjdghsdjgsdj