adafruit/circuitpython #9651: Add LDO Voltage Regulator module (LDOPWR?)

RetiredWizard, 2024-09-23:
Espressif is playing around on the ESP32-P4 with support for UHS-1 SD cards that use 1.8v signal voltages. Currently they have a couple of [recommendations](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32p4/api-reference/peripherals/sdmmc_host.html#configuring-voltage-level) on how to deal with the new voltage. One option is to just design the hardware with your intended voltage feeding the SD_VSS pin. A second option is to utilize some [LDO power supply features](https://docs.espressif.com/projects/esp-idf/en/stable/esp32p4/api-reference/peripherals/ldo_regulator.html) of the SOC itself. It turns out that on the Function-EV dev board, Espressif connected one of the ESP32-P4 LDO lines to the SD Card which means you should be able to control the voltage via software. As it stands, when you run the current version of CircuitPython, the dev board appears to be supplying about 1v to SD card pins which means that neither the standard SDIOIO or SPI modules will successfully work with the boards SD card slot.I have been able to build a version of the SDIOIO module that turns on the LDO power channel connected to the SD card which brings the voltage up to 3.3 volts and allows the SDIOIO module to properly mount an SD card.

While the SDIOIO module could be modified to work on the Function-EV board using either compile time board parameters or settings.toml runtime parameters (ths SPI or SDCARDIO module would also need updates in order for the Function-EV board to fully support the SD card as designed), I'm thinking the correct way to deal with multiple SDMMC interface voltages would be to provide a CircuitPython core LDOPWR library for controlling the LDO channels. That way the Python developer could decide how best to power the connected device based on their requirements.

tannewt, 2024-09-23:
You could write a Python level driver that uses `memorymap`: https://docs.circuitpython.org/en/latest/shared-bindings/memorymap/index.html

bill88t, 2024-09-23:
I think a python driver would be better, since we could have it frozen and the user could override the version if needed.
I'll look into it once I get a P4 if it's not done till then.

RetiredWizard, 2024-09-23:
I'll take a look at `memorymap` but at first glance, figuring out how to interface the LDO via memory manipulations would take me deeper into the IDF than I ever expected to go :grin:

The issue is sort of board related as it depends on how the chip is wired up to an SD card slot (or other SDIO peripheral) so maybe before deep diving the IDF, I'll see if I can create an object in board/board.c, maybe board.LDOn, similar to how board.DISPLAY is created.

RetiredWizard, 2024-09-23:
Well after a quick look at the board/board.c thought I had, I realized that's not really any different than my original suggestion to create a core device/library. I'll spend a bit of time poking at the `memorymap` option but may come back around to a core module eventually, unless @bill88t get's to it first.

arturo182, 2024-09-23:
In the IDF the LDO code is here:
https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/ldo/esp_ldo_regulator.c
and
https://github.com/espressif/esp-idf/blob/46acfdce969f03c02b001fe4d24fa9e98f6adc5e/components/hal/esp32p4/include/hal/ldo_ll.h
and 
https://github.com/espressif/esp-idf/blob/master/components/soc/esp32p4/include/soc/pmu_struct.h#L857
and
https://github.com/espressif/esp-idf/blob/46acfdce969f03c02b001fe4d24fa9e98f6adc5e/components/soc/esp32p4/include/soc/reg_base.h#L134

In case you were looking 😁

tannewt, 2024-09-24:
> I'll take a look at `memorymap` but at first glance, figuring out how to interface the LDO via memory manipulations would take me deeper into the IDF than I ever expected to go 😁

Ya true! If you want to wrap the IDF then you can add another ESP-specific module (like `espulp`). I don't think we want to abstract it for all ports yet.