adafruit/circuitpython #9660: Remove remaining direct uses of `mp_obj_property_t`

jepler, 2024-09-25:
We prefer MP_PROPERTY_GETTER and MP_PROPERTY_GETSET, as these can save flash space on some ports and are also more self explanatory

There's not currently a macro for a property that supports deletion. If there's some rare property that supports deletion, we should introduce and use MP_PROPERTY_GETSETDEL for it (albeit maybe with a better name)

jepler, 2024-09-25:
```c
ports/raspberrypi/bindings/rp2pio/StateMachine.c:const mp_obj_property_t rp2pio_statemachine_writing_obj = {
ports/raspberrypi/bindings/rp2pio/StateMachine.c:const mp_obj_property_t rp2pio_statemachine_pending_obj = {
ports/raspberrypi/bindings/rp2pio/StateMachine.c:const mp_obj_property_t rp2pio_statemachine_txstall_obj = {
py/objproperty.c:extern const mp_obj_property_t __property_getter_start, __property_getter_end, __property_getset_start, __property_getset_end;
py/objproperty.h:#define MP_PROPERTY_GETTER(P, G) const mp_obj_property_t P = {.base.type = &mp_type_property, .proxy = {G, MP_ROM_NONE, MP_ROM_NONE}}
py/objproperty.h:#define MP_PROPERTY_GETSET(P, G, S) const mp_obj_property_t P = {.base.type = &mp_type_property, .proxy = {G, S, MP_ROM_NONE}}
shared-bindings/displayio/OnDiskBitmap.c:const mp_obj_property_t displayio_ondiskbitmap_pixel_shader_obj = {
shared-bindings/keypad/__init__.c:const mp_obj_property_t keypad_generic_key_count_obj = {
shared-bindings/keypad/__init__.c:const mp_obj_property_t keypad_generic_events_obj = {
shared-bindings/keypad/__init__.h:extern const mp_obj_property_t keypad_generic_events_obj;
shared-bindings/keypad/__init__.h:extern const mp_obj_property_t keypad_generic_key_count_obj;
shared-bindings/vectorio/Rectangle.c:const mp_obj_property_t vectorio_rectangle_width_obj = {
shared-bindings/vectorio/Rectangle.c:const mp_obj_property_t vectorio_rectangle_height_obj = {
shared-bindings/vectorio/Rectangle.c:const mp_obj_property_t vectorio_rectangle_color_index_obj = {
```
These are the ones I found via grep in a recent revision of circuitpython.

diamant3, 2024-10-03:
Hello, may I take this one? thanks!

dhalbert, 2025-03-17:
See if completely done by #9684.

eightycc, 2025-03-17:
All cases were resolved by #9684. Closing issue.

closed 2025-03-17 by eightycc