adafruit/circuitpython #9663: Support for M5Stack CoreS3 SE

scarolan, 2024-09-25:
The CoreS3 SE is basically a stripped down version of this board:

https://circuitpython.org/board/m5stack_cores3/

Compared to the M5CoreS3, the M5CoreS3 SE does not feature a camera (GC0308), proximity sensor (LCR-553ALS-WA), IMU (BMI270), or magnetic compass (BMM150).

https://docs.m5stack.com/en/core/M5CoreS3%20SE

The CoreS3 firmware .bin loads and runs.  I will take a crack at removing the camera, prox sensor, and compass.

tannewt, 2024-09-26:
Non-Adafruit boards are usually community supported. There are instructions for adding a new board here: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

scarolan, 2025-03-18:
Closing out this issue due to lack of interest.

closed 2025-03-18 by scarolan