adafruit/circuitpython #9669: BLE HID issues on esp32s3

lukasmwerner, 2024-09-28:
### CircuitPython version

```python
Adafruit CircuitPython 9.2.0-beta.0-16-gec5cca6983 on 2024-09-27; TinyS3 with ESP32S3
```

### Code/REPL

```python
# SPDX-FileCopyrightText: 2020 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""
This example acts as a keyboard to peer devices.
"""

# import board
import sys
import time

from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS

import adafruit_ble
from adafruit_ble.advertising import Advertisement
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.standard.hid import HIDService
from adafruit_ble.services.standard.device_info import DeviceInfoService

# Use default HID descriptor
hid = HIDService()
device_info = DeviceInfoService(
    software_revision=adafruit_ble.__version__, manufacturer="Adafruit Industries"
)
advertisement = ProvideServicesAdvertisement(hid)
advertisement.appearance = 961
scan_response = Advertisement()

ble = adafruit_ble.BLERadio()
if ble.connected:
    for c in ble.connections:
        c.disconnect()

print("advertising")
ble.start_advertising(advertisement, scan_response)

k = Keyboard(hid.devices)
kl = KeyboardLayoutUS(k)
while True:
    while not ble.connected:
        pass
    print("Start typing:")
    while ble.connected:
        c = sys.stdin.read(1)
        sys.stdout.write(c)
        kl.write(c)
        # print("sleeping")
        time.sleep(0.1)
    ble.start_advertising(advertisement)
```

### Behavior

```python
Traceback (most recent call last):
  File "code.py", line 23, in <module>
  File "adafruit_ble/services/standard/hid.py", line 352, in __init__
  File "adafruit_ble/services/__init__.py", line 81, in __init__
  File "adafruit_ble/characteristics/__init__.py", line 164, in __set__
  File "adafruit_ble/characteristics/__init__.py", line 124, in _ensure_bound
  File "adafruit_ble/characteristics/__init__.py", line 149, in __bind_locally
ValueError: Value length > max_length
```

dhalbert, 2024-09-28:
Does not fail in 9.2.0-alpha.2351, so the change is recent.

lukasmwerner, 2024-09-28:
Is there a way I can download a 9.2.0-alpha build?

dhalbert, 2024-09-28:
Try https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/unexpectedmaker_tinys3/en_US/, and pick the release with PR9597. #9597 fixed some things that prevent BLE HID from working. I was just seeing if it was the same error.

lukasmwerner, 2024-09-28:
Cool looks like it is working there! Thank you so much for your help @dhalbert. Looking forward to the 9.2 release

dhalbert, 2024-09-29:
I'm reopening this because the latest builds, past #9597, cause the error you found. So there was a regression we need to fix.

dhalbert, 2024-09-30:
I did a bisect and 03b077dd912d9fd7faf9fb167fb7d8f48af300ab is the first "bad" commit.

closed 2024-10-02 by tannewt

ThomasAtBBTF, 2025-07-16:
Just for your info:
Only slightly modified code works fine under CP 10.0.0.-beta.0

```

import sys
import time
import supervisor
import adafruit_ble
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.standard.hid import HIDService
from adafruit_ble.services.standard.device_info import DeviceInfoService
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS

# Create HID and Device Info services
hid = HIDService()
device_info = DeviceInfoService(
    software_revision=adafruit_ble.__version__, manufacturer="Adafruit Industries"
)

# Advertise HID service
advertisement = ProvideServicesAdvertisement(hid)
advertisement.appearance = 961  # Keyboard appearance

ble = adafruit_ble.BLERadio()
print("disconnecting...")
for conn in ble.connections:
    conn.disconnect()

print("Advertising BLE HID keyboard...")
ble.start_advertising(advertisement)

# Create keyboard instances
kbd = Keyboard(hid.devices)
kl = KeyboardLayoutUS(kbd)

keyavail = False
# Send keystrokes from REPL input
while True:
    while not ble.connected:
        keyavail = supervisor.runtime.serial_bytes_available
        if keyavail:
            c = sys.stdin.read(1)
        pass
    print("Connected. Type to send keystrokes:")
    while True:
        keyavail = supervisor.runtime.serial_bytes_available
        if keyavail:
            c = sys.stdin.read(1)
        else:
            break
    while ble.connected:
        keyavail = supervisor.runtime.serial_bytes_available
        if keyavail:
            c = sys.stdin.read(1)
            if c == "x":
                break
            elif c == "?":
                for connection in ble.connections:
                    print("Connected to:", connection)
                    print(dir(connection))
                    # Optional: Check more connection details
                    #print("MTU size:", connection.mtu_size)
                    #print("Paired:", connection.paired)
            else:
                sys.stdout.write(c)
                kl.write(c)
                time.sleep(0.01)
    # Revert to advertising
    print("disconnecting...")
    for conn in ble.connections:
        conn.disconnect()
        while conn.connected:
            pass
    time.sleep(1)
    print("Advertising BLE HID keyboard...")
    ble.start_advertising(advertisement)

```

Modifications:
- using serial_bytes_available for not get "stuck" in sys.stdin.read(1) when no character is available.
"x" for disconnecting
"?" for information about the connections 

@dhalbert