adafruit/circuitpython #9674: Specifying alternate address pins in adafruit_matrixportal.matrix.Matrix crashes MatrixPortal M4

danielmader, 2024-09-30:
### CircuitPython version

```python
Adafruit CircuitPython 9.1.4 on 2024-09-17; Adafruit Matrix Portal M4 with samd51j19
```

### Code/REPL

```python
from adafruit_matrixportal.matrix import Matrix

# default pinout:
# R1 - G1
# B1 - GND
# R2 - G2
# B2 - B
# A - D
# C - E
# CLK - LAT
# /OE - E
matrix = Matrix(width=64, height=32)

# alternate pinout:
# R1 - G1
# B1 - GND
# R2 - G2
# B2 - E
# A - B
# C - D
# CLK - LAT
# /OE - GND
addr_pins = [board.MTX_ADDRA, board.MTX_ADDRD, board.MTX_ADDRC, board.MTX_ADDRE, board.MTX_ADDRB] 
matrix = Matrix(width=64, height=32, alt_addr_pins=addr_pins)   
display = matrix.display
```

### Behavior

The used RGB matrix from WaveShare has a different pinout. Changing the relevant pins using a list for alt_addr_pins results in a hard crash:
```
Adafruit CircuitPython 9.1.4 on 2024-09-17; Adafruit Matrix Portal M4 with samd51j19
>>> 
soft reboot

Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Hard fault: memory access or instruction error.
Please file an issue with your program at github.com/adafruit/circuitpython/issues.
Press reset to exit safe mode.
```

dhalbert, 2024-10-04:
@danielmader There are actually two problems going on here:
1. You are specifying a matrix of height 32, but you're specifying 5 address lines. That is one too many. See the documentation here about how many address lines are used: https://docs.circuitpython.org/en/latest/shared-bindings/rgbmatrix/index.html#rgbmatrix.RGBMatrix
2. A ValueError is thrown to tell you about that problem. However, when the VM restarts, there is a hard memory error, and, it goes into safe mode. The memory error is because `RGBMatrix` does not clean up sufficiently before the `ValueErro` is thrown.

So we need to fix problem 2, and you should try one less address line, due to problem 1.

danielmader, 2024-10-06:
@dhalbert Thank you for giving this issue attention!

I understand issue 1. However, I am still clueless how I can specify the different pinouts from the male plug of the MatrixPortal M4 board to the female socket of the WaveShare RGB matrix. I thought, the only way to do so is to pass the full list of address lines in the order of the pins.

[image]

dhalbert, 2024-10-06:
>I thought, the only way to do so is to pass the full list of address lines in the order of the pins.

You should only list as many address pins as are needed -- in this case, four pins. The alt address pins are just substituted for the ones that would be chosen by the library. With an Adafruit matrix the pins used would be `[board.MTX_ADDRA, board.MTX_ADDRB, board.MTX_ADDRC, board.MTX_ADDRD]`. That is the default chosen by `adafruit_matrixportal` based on the given parameters.

The screenshot you gave above for the Waveshare display is _not_ the one I found: https://www.waveshare.com/rgb-matrix-p2.5-64x32.htm.

[image]

The one you show does not have a `D` line, but the screenshot I am showing above does. Assuming mine is correct:

It looks like the Waveshare board has the `B` on Adafruit's`D` line, and the `E` line on Adafruit's `B` line. The `D` line is Adafruit's `E` line. So try `[board.MTX_ADDRA, board.MTX_ADDRD, board.MTX_ADDRC, board.MTX_ADDRB]`.

danielmader, 2024-10-06:
Thanks again, and yes, you were fully right about me posting a wrong pinout. I've taken it for this post only from an older version of the wiki without checking, please excuse.

Unfortunately, your suggestion still yields a broken display, at least when I use the higher-level class `adafruit_matrixportal.matrix.Matrix` instead of your suggested 
`rgbmatrix.RGBMatrix`:
```
from adafruit_matrixportal.matrix import Matrix
[...]
addr_pins = [
    board.MTX_ADDRA, 
    board.MTX_ADDRD, 
    board.MTX_ADDRC,
    board.MTX_ADDRB
]
matrix = Matrix(width=64, height=32, alt_addr_pins=addr_pins)
display = matrix.display
```
I've posted this as a question to the Adafruit forum, too. Maybe the discussion should be continued there: [https://forums.adafruit.com/viewtopic.php?t=213775]

dhalbert, 2024-10-06:
> Maybe the discussion should be continued there: https://forums.adafruit.com/viewtopic.php?t=213775

I"ll follow up there for the pin order.

dhalbert, 2024-10-08:
The core issue here to address is that exception failures in the `RGBMatrix` constructor do not clean up properly, and cause crashes. Moving this forward to 9.x.x.

dhalbert, 2024-10-09:
It turns out this diagram has several errors:
[image]
I am going to get that fixed!

The schematic is correct, of course:
[image]

The pin-out of the Waveshare panel is the **same** as the Adafruit panels. Pinouts of our panels are discussed here: https://learn.adafruit.com/32x16-32x32-rgb-led-matrix/new-wiring. Note you have to look for the 64x32 ones.

So you don't need to shuffle the address pins. I think these problems may be power-related, and will discuss that in the forums post.

jepler, 2024-11-27:
I want to look into the safe mode reset in particular; I don't know about all the rest.

dhalbert, 2024-12-16:
Reopening due to #9891 and reversion of #9865.

PR #9865 (2024-12-05): rgbmatrix: Avoid leaving an incompletely configured display bus
I did not reproduce the exact problem reported; however, I reproduced that a failure to construct an RGBMatrix object would leave CircuitPython in an inconsistent state with a display bus registered but not working:

```py
>>> import rgbmatrix, board; rgbmatrix.RGBMatrix(width=64, bit_depth=1, rgb_pins=[], addr_pins=[], clock_pin=board.A0, latch_pin=board.A1, output_enable_pin=board.A2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: The length of rgb_pins must be 6, 12, 18, 24, or 30
>>> impor [... cut]
commit ed5c0bd4 (2024-12-05): rgbmatrix: Avoid leaving an incompletely configured display bus

I did not reproduce the exact problem reported; however, I reproduced
that a failure to construct an RGBMatrix object would leave CircuitPython
in an inconsistent state with a display bus registered but not working:

```py
>>> import rgbmatrix, board; rgbmatrix.RGBMatrix(width=64, bit_depth=1, rgb_pins=[], addr_pins=[], clock_pin=board.A0, latch_pin=board.A1, output_enable_pin=board.A2)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: The length of rgb_pins must be 6, 12, 18, 24, or 30
>>> impo [... cut]
closed 2026-10-02 by tannewt