adafruit/circuitpython #9681: rotaryio library doesn't work on RP2350

D3Fq, 2024-10-03:
### CircuitPython version

```python
CircuitPython 9.2.0-beta.0 on 2024-09-17; Raspberry Pi Pico 2 with rp2350a
CircuitPython 9.2.0-beta.0-15-gca25c602a8 on 2024-09-27; Raspberry Pi Pico 2 with rp2350a
CircuitPython 9.2.0-beta.0-16-gec5cca6983 on 2024-09-27; Raspberry Pi Pico 2 with rp2350a
```

### Code/REPL

```python
import rotaryio
import board

encoder = rotaryio.IncrementalEncoder(board.GP15, board.GP14)
last_position = None
while True:
    position = encoder.position
    if last_position is None or position != last_position:
        print(position)
    last_position = position
```

### Behavior

Terminal returns 0 value once and doesn't increment another during encoder rotation.

### Description

I tested this code on 3 different Circuitpyton versions of v9.2.0-beta.0 on RP2350 with the same result. I also checked voltages on Pico2 inputs by oscilloscope and HIGH/LOW states seems to look fine.

This code run on RP2040 with CircuitPython 9.2.0-beta.0 works correctly. 

A bug shouldn't be related to RP2350 Errata E9 issue, because rotaryio use pull-ups on inputs.

jepler, 2024-10-03:
If you can test with a build artifact from #9682 (should be ready in an hour or less), it'd be appreciated! I had also encountered the bug & fixed it, but apparently neglected to make a PR. Thanks for the report!

D3Fq, 2024-10-04:
> If you can test with a build artifact from #9682 (should be ready in an hour or less), it'd be appreciated! I had also encountered the bug & fixed it, but apparently neglected to make a PR. Thanks for the report!

Yes, I've just tested 9.2.0-beta.0-23-g11d4bce481 and can confirm it's work fine on RP2350. 

@jepler @dhalbert Many thanks!

jepler, 2024-10-05:
You're welcome, and thanks again for the report.

PR #9682 (2024-10-04): Work around an sdk bug
This was already reported & fixed upstream, but the fix will not be in a released SDK until 2.0.1. It's easy enough to use our own workaround.

Testing: without this bug fix, rotaryencoder did not work on a pico 2

Note: the cut&paste code was reindented by uncrustify, giving it a slightly different style than the upstream sdk code.

Closes: #9681 

I thought I had PRd this two weeks ago, but apparently I didn't. The problem was independently rediscovered by @d3fq, thanks for the report.